gtxaspec/wz_mini_hacks #289: Using wz_mini_hacks on a device which has the "official" Wyze RTSP beta firmware fails to start the RTSP server

KevinKadow, 2022-08-31:
In configuring a batch of camera we failed to realize that some of our V3 cameras already were modified previously with the RTSP firmware.

In these instances, the hack seemed to work successfully (webserver for example started okay) but instead of starting the RTSP service on port 8554, the original Wyze service on port 554 will be running.

As an initial mitigation, it would be useful for the wz_mini_hacks startup scripts to detect the Wyze RTSP firmware (for example, by looking for /system/bin/liveMediaServer installed and running) and provide a warning in the logs and also via the webserver (at least in diagnostics, if not on the main page).

Currently the only indication in diagnostics is the line "Firmware Version: 4.61.0.3" in the output.
[image]

gtxaspec, 2022-08-31:
I will add a warning in the logs, thanks.

tbrausch, 2022-08-31:
Quick question - should the wz_mini_hack work on Wyze firmware 4.61.0.3 (Wyze RTSP) if I disable RTSP for the camera in the Wyze App?  I am hoping to continue running the Wyze RTSP firmware but slowly migrate to wz_mini_hack.  

I flashed one camera back to the main formeare and was able to get wx_mini_hack working for the low resolution stream, but the high resolution stream consistently exhibits tearing at the bottom of the display.

gtxaspec, 2022-09-02:
fixed in https://github.com/gtxaspec/wz_mini_hacks/commit/4cb55cbb1e0f1bbd4ed064c0b3143665a57aa2ac

gtxaspec, 2022-09-02:
> Quick question - should the wz_mini_hack work on Wyze firmware 4.61.0.3 (Wyze RTSP) if I disable RTSP for the camera in the Wyze App? I am hoping to continue running the Wyze RTSP firmware but slowly migrate to wz_mini_hack.
> 
> I flashed one camera back to the main firmware and was able to get wx_mini_hack working for the low resolution stream, but the high resolution stream consistently exhibits tearing at the bottom of the display.

it should.

closed 2022-09-02 by gtxaspec

virmaior, 2022-09-02:
I will try to add this to the diagnostics page. Mostly everything on that page is culled from gtxaspec helping someone with a problem, so you get to be the origin on this one!

baudneo, 2022-09-03:
> Quick question - should the wz_mini_hack work on Wyze firmware 4.61.0.3 (Wyze RTSP) if I disable RTSP for the camera in the Wyze App?  I am hoping to continue running the Wyze RTSP firmware but [...]

What are you using to view the stream that is tearing? I had same issue and it turned out to be CUDA hwdecoder via ffmpeg.

tbrausch, 2022-09-04:
I mostly use the Frigate Card (in Home Assistant) with webrtc to view the
stream, but the tearing also happens in VLC simply running a network
stream.  After seeing another post referring to this, I
set ENABLE_LOCAL_DNS and ENABLE_FSCK_ON_BOOT to true which seemed to
significantly reduce the number of times the tearing occurs, but it still
happens  4-5 times per minute or so.  My views now use 640x360 which never
fails and I use the 1080P for recording only.