gtxaspec/wz_mini_hacks #293: DNS Issues Still

guarddog13, 2022-09-04:
Ok so I've been trying to troubleshoot a couple of things with my cameras and noticed something in my adguard. I have 3 cameras that eventually time out their scripts at least every other day. They need reboots. These 3 cameras also drop frames a lot when using the high res stream on AgentDVR. When checking Adguard all 3 of these cameras top the DNS inquires of my entire network. I have DNSMasq turned on. I have 4 other cameras that make calls periodically as expected. I can't figure out why these 3 are doing this. Any help would be greatly appreaciated.

guarddog13, 2022-09-04:
The DNS inquiries and dropped frames/scripts timing out seem to go hand in hand. I can reboot the cameras until the dropped frames stop. When the feeds quit dropping frames the excessive DNS inquries stop. Are the cameras trying to talk to Wyze and not able to? Is there a way around this? Why is communication to Wyze failing if that's the issue?

maxfield-allison, 2022-09-26:
what are the firmware versions? have you tried disabling adguard temporarily?