gtxaspec/wz_mini_hacks #305: PAN V1 question.....

ragenhe, 2022-09-20:
I have a Pan V1 that will not flash the firmware manually.

I have a V2 and a V3 that will and is running wz_mini_hacks, so I know the process well enough to remove user error as the cause.

I have tried different SD cards to ensure it wasn't corruption.

It wasn't bought when they first came out, so batch wise I don't think it was in the beginning.

Can anyone suggest another way?  Or has anyone overcome this issue?

Thanks!

virmaior, 2022-09-21:
just to confirm  you ran `compile_image.sh pan` to produce the image?

another possibility is that at least for Wyze v2s (I don't own a v2 or panv1 so am depending on memory reading about a similar issue), there have been different bootloaders where others have mentioned finding rare cameras that won't do the normal load. I think there was a process described about a month ago where the bootloader could be overwritten resulting in a simplified install process for v2s.

ragenhe, 2022-09-21:
Yes, I ran it with the pan present.

Let me find that and read through it.

Thank you!
Ragen

ragenhe, 2022-09-21:
Even the downloaded special uboot demo bin cannot be flashed.

For the record, right after I bought the camera, I tried to flash the wyze rtsp firmware and it wouldn't work either.

So for this camera, it appears I'm just stuck with stock firmware.

closed 2022-09-22 by ragenhe