gtxaspec/wz_mini_hacks #314: Selfhosted Mode failing to patch iCamera

dgrzjohn, 2022-09-28:
The iCamera patch is failing to apply on my Wyze v3 cam which causes selfhosted mode to still have connectivity issues.  I think this is the relevant part from the wz_init.log:

`====> Calling patch_out_code_test_enable

Original at 0x7dfcc
00000000: cbf0 110c                                ....
Patched
00000000: 0000 0000                                ....
iCamera patch failed.
Reverting iCamera patch.
Removed.
`
Firmware version 4.36.9.139

Let me know if I can supply anything else, I've been waiting for this!

gtxaspec, 2022-09-28:
confirmed,

```
====> Calling patch_out_code_test_enable

Original at 0x7dfcc
00000000: cbf0 110c                                ....
Patched
00000000: 0000 0000                                ....
iCamera patch failed.
Reverting iCamera patch.
Removed.
```

gtxaspec, 2022-09-28:
@dgrzjohn 
in the file `/opt/wz_mini/usr/bin/patch_icamera.sh`
comment out line #239 

`#set -e`
and then reboot and see if it works.

dgrzjohn, 2022-09-28:
That seems to have allowed the patch to complete successfully.   Thank you!

```

Patching done.
816a42f5f35dd41ad9aae19163e98c49  iCamera
Installed.
iCamera patch applied successfully.

```

kohrar, 2022-09-28:
Thanks for reporting this @dgrzjohn. 

PR #315 should fix this issue.

gtxaspec, 2022-09-29:
#315 merged!

dgrzjohn, 2022-09-29:
Working after updating the firmware.  Thanks again!