gtxaspec/wz_mini_hacks #324: Changes to FPS in RTSP do not get applied correctly

tbrausch, 2022-10-02:
When I make changes to the RTSP settings in the wz_mini.conf file they never seem to have any effect. For example, if I set the RTSP_HI_RES_FPS="" and view the camera in VLC the media info shows 25 FPS as expected. However if I change it to RTSP_HI_RES_FPS="20" the VLC app still shows 25 FPS. When I change other settings beside RTSP it seems to work.  So am I missing something obvious?  

A few days later I tried to set the FPS to 30 and magically it worked.  However, now I cannot change it back.

tbrausch, 2022-10-02:
Wow, thought this hack was not supposed to change the camera hardware/firmware!  I had to factory reset the device to get the FPS back to 20, although this did not help the HIRES stream - still stability and tearing issues.

gtxaspec, 2022-10-02:
In order to push the camera beyond 20fps, we have to modify the config files on the camera.  This is one of the only situations where the camera config is modified.  

I will look into your issue of the FPS not returning.

tbrausch, 2022-10-03:
Thanks, things now make a little more sense.

tbrausch, 2022-10-03:
Could I ask one more question?  Several times I have installed wz_mini_hacks on a Wyze V3 camera which is running the Wyze RTSP beta firmware and tried to get the wz_mini_hacks RTSP to work, but to no avail.  Should the RTSP streams work with the 4.61.0.1 firmware? (I tried with the Wyze RTSP both enabled and disabled).  I have read here that wz_mini should work, but not specifically if the RTSP portion does.

Thanks again.

kohrar, 2022-10-05:
@tbrausch There's likely some compatibility issues between the libcallback hack and firmware 4.61.0.1 (RTSP beta). I do recall having the same issues you did where the RTSP stream from wz_mini_hacks doesn't work on that particular version.
The RTSP stream from the beta firmware that you enable via the Wyze app should still work, however.

tbrausch, 2022-11-27:
> @tbrausch There's likely some compatibility issues between the libcallback hack and firmware 4.61.0.1 (RTSP beta). I do recall having the same issues you did where the RTSP stream from wz_mini_hacks [...]

It would be great if the wyze_mini_hacks RTSP worked on Version 4.61.0.1 (RTSP Beta).  I would like to use the Wyze RTSP for 1080P and the wyze_mini_hacks RTSP for 360P (because the wyze_mini_hacks RTSP exhibits a lot of tearing and smearing in the 1080P stream)

MomosX, 2022-11-28:
I also tried to lower the FPS on the RTSP stream to anything below 20.  It does not work.  Using only the high res stream, but tried also in dual stream configuration.  

Any solution to this please ?