gtxaspec/wz_mini_hacks #329: changed timezone str to EST now not booting

rickgitdone, 2022-10-06:
am on latest 4.36.9.139 ,,, used the web gui camera config portion to change the timezone string to EDT ...

Could this brick the camera? won't boot fully now with sdcard in or out? red led comes on and nothing further ...

I have also reapplied the 4.36.9.139 and same... If anyone has any ideas LMK...

gtxaspec, 2022-10-06:
can you check the logs folder on the sd card? you can upload them here for us to find out whats going on.

virmaior, 2022-10-06:
@rickgitdone that sucks. 

I did warn that I was unable to test this so far ... 
I'll work on fixing the code; unfortunately, I can't help with the damage.

gtxaspec, 2022-10-06:
@virmaior not to worry, i can help @rickgitdone fix it.  stand by for a fix

rickgitdone, 2022-10-06:
@gtxaspec 
 
tried adding this to the wz_mini/etc/rc.d/S14nightdrop and rebooted a few times

> if ! [ -f /opt/onetimerun ] ; then
>     rm -f /configs/.user_config
>     cp /configs/.user_config.old /configs/.user_config
>     touch /opt/onetimerun
>     sync
> fi

it appears now my backup copy of the .user_config must also have been corrupted since after I ran thru the RTSP firmware upgrade then back to 4.36.9.139 and it came back  ...I boldly ran the camera update script again not realizing the peril... meh...

is it possible I generate a .user_config from scratch and copy that during boot overwriting what's been corrupted? or copy the corrupt  .user_config over to my sdcard during boot so I can edit out the bad parts?

gtxaspec, 2022-10-06:
try
```
if ! [ -f /opt/onetimerun ] ; then
    rm -f /configs/.user_config
    cp /configs/.user_config.1 /configs/.user_config
    touch /opt/onetimerun
    sync
fi
```

rickgitdone, 2022-10-06:
OK .. I am in.. here is my /configs/.user_config from it now below... still looks corrupt .. I went thru each version and found .3 looks clean pasted to bottom ... if I copy .3 ver over the current do I need to preserver the p2pid= and r_encr= section at the top? or just edit out all the quoted sections?

Thanks so far really appreciate the assistance here

[NET]="[NET]"
bindOk=1
p2pid=3D3DHKNBRHG6K36U111A
r_encr=TmllonsHUsxNyhJ+

[SETTING]="[SETTING]"
AMALevel="2"
drawBoxSwitch="2"
res=1
MMALevel="2"
recordType="3"
AST="1"
nightVision="2"
NIGHT_LED_flag="2"
watermark_flag="2"
indicator="2"
bitRate=120
telntSwitch=2
timezone=-4

.user_config.3

[NET]
bindOk=1
p2pid=3D3DHKNBRHG6K36U111A
r_encr=FjHyLccS0xPrXJEd

[SETTING]
AMALevel=2
drawBoxSwitch=2
res=1
MMALevel=2
recordType=3
AST=1
nightVision=2
NIGHT_LED_flag=2
watermark_flag=2
indicator=2
bitRate=120
telntSwitch=2
timezone=-4

[CAMERA_INFO]
TZ_STR=

gtxaspec, 2022-10-06:
use `.user_config.3`

rickgitdone, 2022-10-06:
@gtxaspec  ... back in business ... thx again.

I did the following while active w/o incident and commented out the other edits in wz_mini/etc/rc.d/S14nightdrop then rebooted

rm -f /configs/.user_config && cp /configs/.user_config.3 /configs/.user_config

gtxaspec, 2022-10-06:
glad to hear!

virmaior, 2022-10-06:
glad to see it got fixed. 

https://github.com/gtxaspec/wz_mini_hacks/pull/330 should make it so that this doesn't happen.
I replaced the technique that wrote a completely new wz_mini.conf file (and thus borked the format for this file) with one that first makes a copy, only replaces altered single key=value pair, and then switches file names.
also prettied up the interface just a tad.

gtxaspec, 2023-07-19:
editing anything is now blocked by https://github.com/gtxaspec/wz_mini_hacks/pull/558 due to safety concerns.

closed 2023-07-19 by gtxaspec