gtxaspec/wz_mini_hacks #340: No audio in tinycam app due to audio codec used in RTSP stream

mega3st, 2022-10-10:
When trying to activate audio in the tinycam app, the following error message appears

"Unsupported codec. Switch camera audio codec to Linear PCM, G.711 A-law (PCMA), G7.11 μ-law (PCMU), AAC, AMR-NB, G.726 (32kbps)"

Is it possible to change the audio codec by adjusting a file?

If not, would it be possible to create an option for this in the future, or maybe even think about changing the default codec in the interest of maximum compatibility?

virmaior, 2022-10-11:
This is a continuing issue that arises from a limitation of what we have:

https://github.com/gtxaspec/wz_mini_hacks/discussions/93
https://github.com/gtxaspec/wz_mini_hacks/discussions/197

basically requires redoing the audio using ffmpeg as the native encoder  v4l2rtspserver doesn't have the ability to do this.

mega3st, 2022-10-11:
Sorry, I wasn’t aware of that - the tinyCam Monitor app seems to be a bit picky about audio.

Assuming that the developer of the app will not provide a solution any time soon, the only alternative would be to switch to a more versatile RTSP server like [rtsp-simple-server](https://github.com/aler9/rtsp-simple-server), right?

And since this is certainly easier said than done, I should probably better resign myself to the fact that the combination of wz_mini_hacks and tinyCam just doesn't work :(

@gtxaspec love this project anyway, even if my use case as a baby monitor on the smartphone without sound unfortunately doesn’t work. By the way, @virmaior thanks for your work with the web server and @kohrar it's great that with self hosted mode and the possibility to link a local NTP server finally a privacy friendly solution is available!!!

kohrar, 2022-10-11:
@mega3st that sounds about right. I've used rtsp-simple-server in the past with good results. Just enable RTSP on the Wyze cam and then add a new source to the rtsp-simple-server config like this:
```
camera:
    source: "rtsp://admin:wyzecam@wyze-cam-ip:8554/unicast"
    sourceProtocol: tcp
```

You can then try pointing your tinyCam app to the rtsp-simple-server instance and see if that works.

mega3st, 2022-10-12:
@kohrar with RTSP firmware 4.61.0.1, I had directly both video and audio in the tinyCam app. However, there were often longer freezers (despite full internet access).

Since the stream is the same, I can't imagine that another hop through a separate rtsp-simple-server changes anything.

Version 4.36.9.139 in combination with the official Wyze app runs smoothly but but isn’t my favorite solution.

Do you know if it would be theoretically possible to encode the raw audio stream on the camera (e.g. via ffmpeg) to AAC and replace the v4l2rtspserver inside the wz_mini_hacks with the rtsp-simple-server? Running a separate rtsp-simple-server instance seems a bit overkill to me.

mrlt8, 2022-10-12:
@kohrar I don't think rebroadcasting the stream will fix it. The problem is that the camera is using an audio format that is not compatible with RTSP (s16le) which needs to be re-encoded. Wyze seems to be going back and forth with the audio codec on different firmware so that's why we just re-encode any incompatible format to AAC on the bridge.

@mega3st Why not just connect to the wyze cams directly? Doesn't tinycam support stock firmware cams?

mega3st, 2022-10-12:
@mrlt8 the intended solution was to operate the camera locally without internet access, which is why I became aware of this project in the first place.

In addition to the unstable integration via RTSP in combination with the RTSP firmware, tinyCam indeed seems to support a cloud connection. However, this would mean that the camera would have to stay online and the latency is also higher.

mrlt8, 2022-10-12:
Not necessarily. I believe tinycam is using the same TUTK sdk as the wyze app, so it will always attempt to connect locally if on the same LAN as the cameras (that's why the app will sometimes show the cam as offline but still be able to connect).

clee604, 2022-10-12:
I use simple-rtsp-server to proxy and recode the rtsp stream to get audio in TinyCam.

Proxy:
```
  rtsp1:
    source: rtsp://xxx:xxx@xxx.xxx.x.xxx:xxxx/unicast
```
Reencode audio
`runOnDemand: ffmpeg -rtsp_transport udp -i rtsp://localhost:8554/rtsp1 -c:v copy -c:a aac -f rtsp rtsp://localhost:$RTSP_PORT/$RTSP_PATH`

I keep the two lines separate so I can still access the original feed via proxy for VLC clients to save on CPU usage when encoding isn't necessary. 

Disclaimer, the proxy and reencoded rtsp feeds above are not secured.

mega3st, 2022-10-12:
@mrlt8 I just tried to connect with tinyCam directly to a local V3 without internet access, but either without initial connection to the Wyze server also no local connection comes up or the self hosted mode of the wz_mini_hacks prevents it...

@clee604 running a separate rtsp server for my simple use case seems unnecessarily complex. Unfortunately the releases of the rtsp-simple-server do currently not contain a MIPS version, otherwise it would have been worth a shot to run it directly on the camera.

So far the combination of RTSP firmware with wz_mini_hancks for self hosted mode and local NTP server comes closest to my desired solution. The only drawback is a high audio latency of about 4 seconds in tinyCam.

skilau, 2022-10-13:
As I was pondering what it would take to do this in v4l2rtspserver...

I see that they have a pending PR to add support for converting the audio stream to MP3:
https://github.com/mpromonet/v4l2rtspserver/pull/247

Granted, it looks like it has been sitting there for a few months open/not merged, but this would suggest that this problem may go away in the future.

gtxaspec, 2023-07-12:
this should be fixed soon with the new go2rtc implementation...stay tuned.