gtxaspec/wz_mini_hacks #343: Stuck in a boot loop?

scottjl, 2022-10-14:
v3 cam, v4.36.9.139 firmware. 16G FAT formatted card. copied over contents of SD_ROOT as instructed. Camera powers on, I can hear mechanical clicking, speaker click, 4 led's flash very quickly, whole cycle repeats endlessly.

any advice? thanks!

virmaior, 2022-10-15:
1. does the camera boot normally if the SD card is not in it?
2. make sure there's no v3_demo.bin in the root directory?
3. what are the contents of wz_mini.conf ?

scottjl, 2022-10-15:
1. yes. or if the card is in and empty
2. card is freshly formatted and empty when i copy the git files over
3. doesn't seem to matter, i've tried with stock config file, as well as my lightly edited config file. (set user and password for rtsp and enabled it).

virmaior, 2022-10-15:
1. are there any log files being generated under `/wz_mini/log` ? that might identify where things are going wrong.
2. presumably, you're not trying defaulting to USB direct or Ethernet over USB?
3. the directory structure looks like:

````
/factory_t31_ZMC6tiIDQN
/wz_mini/
/wz_mini/bin/
/wz_mini/lib/
...
```
right?

we can eliminate:
* Wrong format on SD card. IF the card were formatted ExFAT instead of FAT32, it simply would have no effect. So we can cross that off the list.
* corruption of wz_mini.conf by using the web editor.
* problem with wz_mini.conf because you said you've tried with it unedited

The only other ideas that immediately spring to mind are
* the SD card itself is dying.
* new bootloader that breaks wz_mini ?

If none of that helps, we'll have to wait for smarter people to appear and come up with ideas!

scottjl, 2022-10-15:
yep.. this is what is confusing me too. the card is certainly fat32. i've actually tried two different cards, the first one was 1g and i thought maybe it's just somehow too small, so i tried a 8g card. the file structure was correct, i simply did a copy from SD_ROOT with `cp -rv * /Volumes/Unititled` (I'm doing this from a mac). i edited wz_mini.conf with vim, which will respect the file format, and as i said i also tried just the stock wz_mini.conf because i thought somehow my changes weren't valid. there is nothing generated in /logs. i'm guessing the kernel isn't even getting that far to start writing to log files.

thinking it was my unit somehow i did manually re-flash it with a fresh download of the v4.36.9.139 firmware from wyze's site. unit works just fine, still verifies that it's running the correct firmware. it just won't boot this hack. i haven't tried any other hacks yet.

my purpose was to enable rtsp on the unit, i've since loaded up wyze-bridge, which i know isn't the same thing, but it's working for my purposes. would be nice to see if i can get this to work just for fun though.

endertable, 2022-10-15:
Have you ever had the hack on this cam before?
How new is this cam?
Does /wz_mini/tmp/.storage/rcS  exist?
  If it does not, create it and enter the following:

```
#!/bin/sh
touch /opt/WTFRCS
```

After camera reboots, check contents of SD card and see if it created the file "WTFRCS" on the root.

if the file  /wz_mini/tmp/.storage/rcS DOES exist, exit it and under the #!/bin/sh line at top add the following:

`touch /opt/IAMRCS`

If either file is created, then there is hope.
If no file is created then as @virmaior stated:
  Bad card   (solution: try new/different card)
  New bootloader  (solution:  try WyzeHacks to access telnet to run a dd and md5sum on the BL)

virmaior, 2022-10-18:
WyzeHacks probably won't work on that recent of a firmware since 
1. new default password is unknown
2. they removed telnet ?

in that case, downgrade to a lower firmware version that still has the old password. I've been using 4.36.2.5 but that's pretty old (prior to the recent handiwork of kohrar higher versions couldn't really do offline running)

scottjl, 2022-10-18:
@endertable 

> After camera reboots, check contents of SD card and see if it created the file "WTFRCS" on the root.

Nope, file doesn't exist.

@virmaior 
From the main page of this project: 

> Features:
> • Works on ANY firmware release up to 4.36.9.139

virmaior, 2022-10-18:
Err... WyzeHacks != wz_mini
https://github.com/HclX/wyzehacks is an older hack that worked using a different mechanism. It can't provide SSH. It enables telnet but there's no telnet on 4.36.9.139 so you would need to go to a lower version in order to see if we can learn anything about the bootloader.

endertable, 2022-10-20:
If you are going to use WyzeHacks, create an Upgrade.tar file by doing the following:

`mkdir Upgrade`

In this Upgrade directory have the following:

busybox             (one that includes telnetd, can even use the wz_mini one)
upgraderun.sh   (this is the script that is called by assis for updates)
shadow              (a shadow password file with the encrypted password of your choice)

shadow (with password 'ismart12') can look like:

root:rJ0FHsG0ZbyZo:10933:0:99999:7:::

upgraderun.sh can look something like this:

```
#/bin/sh

mount -o bind  /tmp/Upgrade/shadow /etc/shadow
/tmp/Upgrade/busybox telnetd &

mkdir /media/mmc/TEST 2>/dev/null
if [ -d /media/mmc/TEST ]
then
  dd if=/dev/mtd0 of=/media/mmc/TEST/mtd.img
  md5sum /media/mmc/TEST/mtd.img > /media/mmc/TEST/mtd.md5sum
fi

sleep 300
reboot
```

Be in parent directory of the Upgrade directory and tar it up with something like:

`tar --sort=name --owner=root:0 --group=root:0 --mtime='1970-01-01' --dereference  -cf ./Upgrade.tar  Upgrade`

Substitute the Upgrade.tar file in WyzeHacks as your update file.  You can put even more files as needed in this Upgrade.tar but assis will feed the watchdog for only so long, either 6, 9, or 15 minutes before camera reboots, so don't get too elaborate.  If you don't want to telnet it and just want to see if it creates the mtd0 files then lower or comment out the sleep command.  

The above is to use WyzeHacks mechanism to trigger the update, but will NOT install WyzeHacks or anything else, upon reboot cam will be as it was before.  I tested this on my v4.36.9.139 and it worked.

scottjl, 2022-10-20:
sorry, but this is a bit more than i was planning on doing for this project. i had hoped there was a quick fix for my problem and there isn't.

closed 2022-10-20 by scottjl

thurinus, 2022-11-20:
I experienced this exact symptom, and managed to work around it by doing the formatting and copying of the SD_ROOT files on my MacBook instead of my Windows machine. Am not sure what the underlying cause is... does BSD somehow copy files differently?

This isn't a proper solution, but figured I'd leave a paper trail for anyone who stumbled on this thread trying to debug the same problem. Maybe someone who knows more about this can chime in. Hope it doesn't re-open the issue.

Jugrnot, 2023-03-14:
> I experienced this exact symptom, and managed to work around it by doing the formatting and copying of the SD_ROOT files on my MacBook instead of my Windows machine. Am not sure what the underlying [...]

Unfortunately I too found using my windows workstation to copy the files was the issue.  Fired up my linux laptop and was able to copy them successfully.

blounolito, 2024-11-09:
1 1/2 year after, I encountered the same issue while trying to format, and copy files from Windows.
I then tried to use a Debian VM to do both those steps, and it's working !
I'm not sure why, may it be the presence of "System Volume Information" that make the initial setup loop crash and restart ?
I'm pretty confident that editing the authorized_keys on Windows was not the issue, as I always make sur to use the correct CR LR for Unix.