gtxaspec/wz_mini_hacks #346: Request: Ability to set SSID/pass in conf file

misterjmw, 2022-10-15:
As far as I am aware, the only need to attach a device to Wyze services is to set the SSID credentials.
 
Can a wz_mini add compatibility for a wpa_supplicant.conf file or similar? it would be great to not need to start up a new camera without having to communicate with Wyze at all.

virmaior, 2022-10-18:
wifi ssid is at `/configs/.wifissid`
wifi password is at `/configs/.wifipasswd`

so that means with reboot we can setup this up to be changeable. Of course if you change either to something wrong, you're probably going to lose access unless you're using USB-ethernet or direct.

misterjmw, 2022-10-19:
> for your second part, there's now a part of the web interface that can manipulate wyze values (/cgi-bin/cam.cgi).
I've tried to improve it but a previous version borked a camera for another user which gtxaspec helped them restore.
right now, it requires a reboot for any changes, but for a couple of them we can probably do it without rebooting if we pass them to cmd irled on or cmd irled off directly.
> 
> this is on my wishlist / todo list, just have to fit into full-time life.

I fully understand. Regarding those settings; Here's an example of my use case:
I have a 3d printer in a shed with a Wyze v2 pointed at it.  
I want it to stay in IR mode only however if running 'isolated' none of those controls are exposed. 
I had looked at the web page, and it seemed that most of the settings were still unavailable (e.g. disabling the wyze logo).

As for the wifi portion; I was unaware of these settings. I looked for a (more common) wpa_supplicant.conf or related settings. Likewise the documentation didn't seem to have that information. 
I will test this when I get a free minute or 15. Thank you.

If it was unclear from my ticket; My idea for this was 'simple'. I want to be able to shove a microsd card in a camera and have it usable. I wanted to avoid doing the dance between wyze, firewall, mobile app, QR code, etc.  The closer I can get to running everything through web UI or Blue Iris, the better.

virmaior, 2022-10-19:
I think the ideal way to do this is if we can pre-empt the configuration file on the camera with one on the SD card. If so, then it could work with the slap it in method.  I haven't delved into the mechanics of that myself but a few of the other contributors especially the maintainer would be knowledgeable about whether that's possible.

misterjmw, 2022-10-20:
> I think the ideal way to do this is if we can pre-empt the configuration file on the camera with one on the SD card. If so, then it could work with the slap it in method. I haven't delved into the [...]

I as well am not familiar with how the Wyze internals/configs work. 
That said; I am willing to test or provide information as much as possible.

kohrar, 2022-10-20:
> As far as I am aware, the only need to attach a device to Wyze services is to set the SSID credentials.
> Can a wz_mini add compatibility for a wpa_supplicant.conf file or similar? it would be great to not need to start up a new camera without having to communicate with Wyze at all.

I added a feature a few weeks ago that lets you set up the camera without the Wyze App or the cloud at all.  You only need to have the QR code containing your WiFi information. I've detailed how this works on my own notes at https://leo.leung.xyz/wiki/Wyze_Cam#Setup_a_Wyze_Cam_without_the_Wyze_App_or_internet.

Essentially, with a Wyze cam running wz_mini_hacks which is not already configured with any WiFi settings, you can press the SETUP button and present it with a QR code to set up WiFi. 

If you want to test this on an already-configured camera, you will have to press the SETUP button to clear the WiFi settings, then power cycle the camera. This is because the feature I added only watches for a QR code being used at boot time only if WiFi isn't configured.

baudev, 2022-10-21:
@kohrar Thanks for your work, I'll try this quickly
Your blog page is a treasure!

Which solution did you choose to disconnect your camera from Wyze servers?

I was thinking about:

- Putting a raspberry between my camera and my router and prevent any communication from the camera with the outside world
- Using iptables to filter outgoing communications (block all except RTSP). Is it enough ?

kohrar, 2022-10-22:
That solution sounds fine to me.

I've opted to put them on my wireless network with the network gateway and DNS servers set to something invalid to effectively stop it from being able to talk to the internet. I've previously used OpenWRT on a Raspberry Pi to set up a completely separate network with a port forwarding to each camera, but I've since just placed them all on my local LAN to simplify everything.

baudev, 2022-10-22:
Thanks for your answer @kohrar. I used iptables and it works great 🎉 

To provide feedback on your isolated mode:

- it is important to follow the order of the steps explained [here](https://leo.leung.xyz/wiki/Wyze_Cam#Setup_a_Wyze_Cam_without_the_Wyze_App_or_internet)
(indeed on a camera already configured, I had a lot of problems. Maybe it should be mentioned in the README that the camera must be reset? )
- It is important to add `indicator=2` in `/configs/.user_config`, otherwise the blue led flashes indefinitely

misterjmw, 2022-10-24:
> I added a feature a few weeks ago that lets you set up the camera without the Wyze App or the cloud at all. You only need to have the QR code containing your WiFi information. I've detailed how this [...]

1. That's great information to have. I had decoded the QR code out of curiosity but gone no further than that.
2. It still feels like an band-aid solution. Why go through the legwork of generating a code, resetting a camera, etc. If I can drop a wpa-supplicant file on the microsd and update it and away we go.

> wifi ssid is at `/configs/.wifissid`
> wifi password is at `/configs/.wifipasswd`

If the above works, then the QR code is a moot point. 

If I'm the only one asking, I guess it's not a super important feature, but I'd like to be able to flash the camera to the right firmware, then toss the microSD card with my camera settings, wifi settings and wz_mini on it, and have it boot up and go right to it.

misterjmw, 2022-10-24:
> wifi ssid is at `/configs/.wifissid`
> wifi password is at `/configs/.wifipasswd`

I have just tested this and indeed placing the ssid in the first, and the password in the second moved the v2 to a different wifi.
Can these files be created prior to the device being setup? 

I would also agree that @kohrar's wiki is fantastic. Personally it looks like adding the features noted at: https://leo.leung.xyz/wiki/Wyze_Cam#Firmware_settings to the Web UI would be the best step for removing any wyze app dependency. 

I'm going to kick around the code a bit and see if I cant come up with something useful. I am a very sloppy programmer, but I might be able to contribute something functional.

virmaior, 2022-10-25:
@misterjmw  we already have most of that functionality in `/cgi-bin/cam.cgi`
1. I might be missing quite a few settings though. I only have V3s so I don't know whether the code even works correctly on V2s.
2. as of right now, we can't inject it into a running Wyze camera -- we have to reboot to get our settings to go (not yet done but we should be able to (a) turn off/on the infrared without reboot and (b) turn off/MP4 writing without reboot).
3. What I want to know is if we can mount or /configs/ files and thus avoid affecting the camera itself or whether our bootloader hooks too late for that ([b]ash is not my strongest language by a long shot).

misterjmw, 2022-10-25:
> 1. I might be missing quite a few settings though. I only have V3s so I don't know whether the code even works correctly on V2s.
Regarding the V2; There were some oddities. I changed multiple variables then rebooted several times before I saw actual changes in the rtsp stream. Can't explain it, but it occurred. 

As for the parameters, the issue I noted (and a reason for opening this ticket) simply was that they were not pre-populated and I did not see documentation indicating I could do what I needed. 

It wasn't until @kohrar's post that I saw documentation stating I could change the items I needed. 
Using it as a guide; I edited the .parameters file to include the options I needed. Once added they became controllable via the web interface. My v2 was setup as a "Never touches Wyze"-camera, so from a "fresh install" the only options available were AASwitch, SASwitch and CASwitch. Likewise, from the GUI there is not a way to add more.

I looked into my V3. It appears the settings pre-populate to some degree, which did not occur with the V2. 
Spitballing; This also may have been because the V2 was never configured with the Wyze app so the parameters were never written, thus could not be changed in the UI.

virmaior, 2022-10-28:
@misterjmw 
I guess the top thing (at least from perspective) that I'd like figured out is if we can mount our own config file from the SD card so that the camera only sees that one. 

If so, I could much more easily test messing with the camera's config from the web interface. Right now, I'm wary to do things that edit those settings because I'm worried about bricking my V3s (though gtxaspec was able to help someone recover one for someone that my code bricked).

If we have that I can test adding lines directly from the web interface with little fear of permanent damage (always qualified by the reality that I may receive ire from my wife if code tinkering leads to loss of desired footage)

misterjmw, 2022-10-28:
> @misterjmw I guess the top thing (at least from perspective) that I'd like figured out is if we can mount our own config file from the SD card so that the camera only sees that one.
>
> If so, I could [...]

I am curious how the brick occurred, though that curiosity is 100% tangential.

I have a spare v3. but I do not know enough about the hack itself to (probably) be useful. e.g. can we copy the original to the sdcard, then symlink from the original path to the sdcard on boot? 

Happy to help if I can.

virmaior, 2022-10-30:
off to a good start building on https://github.com/gtxaspec/wz_mini_hacks/discussions/374

am thinking if we use `/opt/wz_mini/slapin` we could have it mount 
1. configuration file (`/configs/.parameters` in v2 / `/configs/.user_config` in v3).
2. `wifissid`
3. `wifipassword`

This would make it a lot easier to no-install slap-in and never have to configure anything.

My initial test works (was able to get set the indicator light to on using SD card's slapped user config).

misterjmw, 2023-11-10:
@virmaior 
I know it has been quite some time since I opened this request. Is this active? 
I ask because I come back to the exact same issue as when I opened the ticket. 

My camera/card failed, I'm replacing it and need to know if the camera can be started up without installing the Wyze app.
Looking at #374 it doesn't seem there's been resolution there either.
Thanks!

SolAZDev, 2024-10-28:
> 1. That's great information to have. I had decoded the QR code out of curiosity but gone no further than that.
>
>     2. It still feels like an band-aid solution. Why go through the legwork of [...]

Quite frankly, I just started on a brand new v3 and this comment directed to the Notes, really helped out. Following the steps got me to connecting without the app, and being able to ssh as the repo's wiki suggests. I think this should be added somewhere on the wiki. I now have a Wyze-less wyze camera. Many thanks!