gtxaspec/wz_mini_hacks #347: No audio recording on v2

petermg, 2022-10-17:
So I complied the demo.bin for v2 as per the instructions and everything worked but the audio. I was no longer able to hear or record any audio from the camera. I was using 
[4.9.9.1433](https://s3.us-west-2.amazonaws.com/download.wyzecam.com/firmware/v2/demo_4.9.9.1433.bin.zip) (September 27, 2022)
I tried this on two cameras. I didn't see anyone else mention this so I'm posted it here. 
I flashed the camera back to the stock firmware same version and the audio works again. No idea what's going on. It seems like this mod breaks audio recording on the latest v2 firmware as of this post.

gtxaspec, 2022-10-17:
anything past 4.9.8.1002 is not supported
you will have to downgrade to that firmware
i will update the documentation accordingly.

petermg, 2022-10-17:
Thanks! I just noticed that's the version on the script. Just finished
compiling that one and am about to flash it now 👍

petermg, 2022-10-17:
Also the URL in the script is no longer valid since they "officially" removed that version, but there is still a working link, it's here:
https://download.wyzecam.com/firmware/v2/demo_4.9.8.1002.zip

skilau, 2022-10-17:
@gtxaspec : Any idea what Wyze did to break it for us?

I don't use any v2's, but I am always keeping my eye out on this kind of stuff for v3, since I am highly invested now in using wz_mini_hacks for all my cameras here at home, and do not want Wyze to break things for me.

gtxaspec, 2022-10-17:
They statically compiled the liblocalsdk libraries into the binary.  We were using LD_PRELOAD to hook into the libraries.  Now we must modify the binary to make changes

noggl, 2023-05-29:
Does anyone know what the last version that works with audio for the Pan Cam V1 is? 4.9.8.1002 doesn't seem to be available (I think the url should have been https://download.wyzecam.com/firmware/pan/demo_4.9.8.1002.zip)

gtxaspec, 2023-07-20:
Looking into this now...

kanemari, 2023-11-01:
I just read that Wyze firmwares have a security hole in them and the details are going to be released shortly. Since this audio bug hasnt been fixed Ive been sitting on the old firmware but now that I know its likely hackable I'm going to have to upgrade. Is there anything that can be done to keep audio?

gtxaspec, 2023-11-01:
the details are already out, see https://github.com/gtxaspec/wz_mini_hacks/discussions/659