gtxaspec/wz_mini_hacks #357: V3 fails to start if conf files changed, requiring mini hacks to be reinstalled.

rocket59, 2022-10-20:
It looks like I may be the only one with this issue. 

If I edit any config file either through my desktop or directly on the camera vis SSH/nano (eg wz_mini.conf or .user_config) a restart of the camera kills everything (no RTSP or SSH access). The same happens if I make a change through the web interface and reboot from it. The only way to recover is to redo the hack from scratch. I have no idea why?

virmaior, 2022-10-20:
It sounds like wz_mini is not running at all on reboot.

This makes me wonder if there's something causing your card to get reformatted to ExFAT on reboot. 
1. are you using the same SD card every time?
2. when you open the SD card on a computer are the files still there?
3. is the card still FAT32 formatted when you check it in a computer?

(The bootloader can only understand FAT32 but that the fully booted camera can use EXFAT, so it's possible that it's changing the format of the card and thus breaking the hack).

rocket59, 2022-10-20:
1. Yes 32gb Sandisk Ultra (nothing special about it)
2. Yes 
3. Yes

A straight restart of the camera with no changes to any files doesn't cause any problems, so only when a change is made it fails to run.

virmaior, 2022-10-20:
I'm at a loss, but  I'm wondering if the SD card might be bad. It seems really weird that this would happen in this way at all since the hack works initially (you do hear the "wz_mini initializing" message and then get a reboot right?)

1. Does it write mp4 files to the card as normal?
2. can you write any sort of file to the card without it getting clobbered? as in just write something meaningless into the /wz_mini/ directory
3. There's some log file info if you run `/wz_mini/bin/gather_wz_logs.sh` . Any errors in it?

rocket59, 2022-10-20:
It seems that the card was at fault although it works fine in my pc. I swapped it out for an old samsung pro card and so far editing in SSH or using the web interface and rebooting works as it should!

closed 2022-10-20 by rocket59