gtxaspec/wz_mini_hacks #360: DNS lookups failing - resolv.conf "comment"

the-real-crayboy, 2022-10-20:
lines in /etc/resolv.conf are being "commented" with the interface name
eg:
```
# HUALAI DNS #
search local # wlan0
nameserver 192.168.1.9 # wlan0
```
which is breaking name resolution:
```
nslookup github.com
nslookup: bad address '192.168.1.9 # wlan0'
```
Removing the comments ( and waiting a minute ) fixes this.
rebooting breaks it again ( can't find the script that's doing this )

sideup66, 2022-10-20:
nslookup in busybox is broken. Its been a long standing issue....are you running the local DNSMASQ on your camera?

the-real-crayboy, 2022-10-20:
I know what needs to be done, but am not familiar enough with this environment to neatly implement.
/usr/share/udhcpc/default.script
is creating bad resolv.conf configuration ( bad for this implementation anyway )
A fix that should last until a reboot is to copy /usr/share/udhcpc/default.script to /tmp ( unfortunately /usr/share/udhcpc/is r/o )
and make these changes:
```
# diff /tmp/default.script /usr/share/udhcpc/default.script
--- /tmp/default.script
+++ /usr/share/udhcpc/default.script
@@ -52,8 +52,7 @@
 		# drop info from this interface
 		# resolv.conf may be a symlink to /tmp/, so take care
 		TMPFILE=$(mktemp)
-		#grep -vE "# $interface\$" $RESOLV_CONF > $TMPFILE
-		sed -e "/# $interface/ { N; d; }" $RESOLV_CONF > $TMPFILE
+		grep -vE "# $interface\$" $RESOLV_CONF > $TMPFILE
 		cat $TMPFILE > $RESOLV_CONF
 		rm -f $TMPFILE

@@ -65,11 +64,11 @@
 		fi

 		[ -n "$search_list" ] &&
-			echo -e "# $interface \nsearch $search_list" >> $RESOLV_CONF
+			echo "search $search_list # $interface" >> $RESOLV_CONF

 		for i in $dns ; do
 			echo adding dns $i
-			echo -e "# $interface \nnameserver $i" >> $RESOLV_CONF
+			echo "nameserver $i # $interface" >> $RESOLV_CONF
 		done
 		;;
 esac
```
the kill the running udhcpc, remove  the wlan0 lines from /etc/resolv.conf and run:
```
udhcpc -t 10 -i wlan0 -p /var/run/udhcpc.pid -b -s /tmp/default.script
```
I am sure there is a way to make this permanent - I just don't know how things get overlayed by wz_mini_hacks.
Will noodle on this - end of day, need a brain break.

the-real-crayboy, 2022-10-21:
[udhcpc.gz](https://github.com/gtxaspec/wz_mini_hacks/files/9840009/udhcpc.gz)
[fixedresolver.script.gz](https://github.com/gtxaspec/wz_mini_hacks/files/9840011/fixedresolver.script.gz)
Not elegant, but this fixes resolving, and survives a reboot
copy udhcpc to /opt/wz_mini/etc/rc.local.d
copy fixedresolver.script to /opt/wz_mini/f
either run  /opt/wz_mini/etc/rc.local.d/udhcpc or reboot
nslookups (should ) now work.

gtxaspec, 2022-10-25:
i'll look into integrating this fix into the main repo