gtxaspec/wz_mini_hacks #361: Wyze V3 Firmware  4.36.0.88

foureight84, 2022-10-21:
I loaded a 32GB FAT 32 SD with the content of "SD_ROOT", and inserted the SD card into the Wyze Cam V3 (firmware  4.36.0.88). I can hear the voice prompt that "WZ Mini initializing first boot. Extracting the first swap, please wait." I wait a good 15 minutes but I don't see the camera connect to WIFI and the LED light stays red. 

This camera has been paired with my Wyze account once before and I put it away so the firmware was never updated.

Should I upgrade to newer firmware for the hacks to work? I've tried turning on USB Direct (USB_ETH disabled) and enabled USB_ETH (USB_DIRECT disabled) but I don't see the camera either. How do I go about getting log files to see what's going on?

kohrar, 2022-10-21:
4.36.0.88 is pretty old. That's one of the first firmware for the v3. We're now at 4.36.9.139. Can you try upgrading it to something a bit newer and see if that helps?

You can find the official versions from Wyze Lab at:
- https://support.wyze.com/hc/en-us/articles/360024852172-Release-Notes-Firmware

If you need a particular version, I have some saved in my repo:
- https://github.com/kohrar/Wyze-Firmwares/tree/master/v3

foureight84, 2022-10-21:
Sweet! Thank you for that. Updated to 4.36.3.18. For some reason, the sd card method wouldn't work with demo_wcv3.bin or demo.bin.

I have the conf set DISABLE_FW_UPGRADE="true". On a curious note, it's always possible to downgrade firmware right? There hasn't been a lockdown firmware to prevent downgrades?

kohrar, 2022-10-21:
As far as I know, you should be able to downgrade using one of the older firmware files in the repo using the SD card method.

Just make sure that your SD card is FAT formatted and the firmware file is named demo_wcv3.bin. You also need to hold the SETUP button as you're powering on the camera.

foureight84, 2022-10-21:
> As far as I know, you should be able to downgrade using one of the older firmware files in the repo using the SD card method.
> 
> Just make sure that your SD card is FAT formatted and the firmware file is named demo_wcv3.bin. You also need to hold the SETUP button as you're powering on the camera.

Thank you! I had it on FAT32... no wonder.

closed 2022-10-21 by foureight84

kohrar, 2022-10-21:
FAT32 should work. 

It looks like there's a new firmware version out. Don't use 4.36.10.2700 as it isn't supported (yet?).

foureight84, 2022-10-21:
Yea. I am on 4.36.3.18 since that's what was on my wyze app and it's not the latest unsupported firmware. I don't need any of the new features since I'll be using the wyze v3 with frigate. I just needed the wz mini hacks for rtsp.

foureight84, 2022-11-04:
> FAT32 should work.
> 
> It looks like there's a new firmware version out. Don't use 4.36.10.2700 as it isn't supported (yet?).

This is quite strange. I am currently on 4.36.3.19 and which was updated using the Wyze App. The newer firmware does not show up on the app and it thinks this is the latest version. I assume it's because it's on a fetch cycle. 

I went back trying to update the firmware manually to 4.36.9.139, which is the latest that's still compatible with the wz_mini_hacks. However, no matter how many times I boot up the device with the SETUP button held down, it doesn't flash the bin file on the SD CARD and goes into setup mode.

The SD Card is 128MB in size (I've tried with a 32GB card as well). I formatted to FAT, then tried FAT32, then tried exFAT. demo_wcv3.bin is sitting on the card's root. I've even tried formatting the card via the Wyze App to no avail. Not sure what's going on.

kohrar, 2022-11-04:
Could you try doing smaller version jumps with your upgrades? Maybe try upgrading from 4.36.3.19 to 4.36.6.17 and see if that works.

foureight84, 2022-11-04:
> Could you try doing smaller version jumps with your upgrades? Maybe try upgrading from 4.36.3.19 to 4.36.6.17 and see if that works.

I've just tried that as well. Still the same issue. I'm seeing reports from several people on Reddit and Wyze forums that it doesn't seem to like certain SD Card brands. Though I've tried Sandisk, Samsung, PNY, and now generic.

foureight84, 2022-11-04:
It looks like they recommend class 10 sd cards which is the specs on their branded sd cards. Apparently, I don't have any class 10 sd cards on hand funny enough. I bought some and should arrive tomorrow. If that doesn't work, I will go as far as buying one of their branded sd card to try.

foureight84, 2022-11-06:
The Class 10 SD cards work. I guess that's a requirement as well for flashing. 32GB or less, FAT32, Class 10.