gtxaspec/wz_mini_hacks #368: WEB_CAM_ENABLE Issues

tbrausch, 2022-10-24:
I tried to setup one of my V3 Wyzecams as a web camera, but after setting WEB_CAM_ENABLE to true the camera would no longer boot up.  I also tried installing the SD card from scratch with the only change being WEB_CAM_ENABLE=true and it still would not boot up.

virmaior, 2022-10-25:
what do you mean by "boot up"?
If you `WEB_CAM_ENABLE` then the camera needs to be plugged into a computer with a working USB cable (supplied cable is power only) and nothing else about the camera should work at all (no ssh, no wyze app, no SD card recording, no control of IR)

tbrausch, 2022-10-25:
When I plug the camra into a USB port it does the following:
1) Initially turns on red LED
2) Red LED turns off folowed by two clicks
3) Blue LED turns on followed by two small pinging sounds
Is this the expected behavior?

tbrausch, 2022-10-25:
My bad initially - I had a bad USB cable.  Now the camera is recognized in the Windows Device Manager.  However, I have tried multiple applications and "Web Cam Test" websites and it still does not work.  The websites initially recognize and find the webcam, but they are not able to use it.  I also tried an older version of wz_mini_hacks just in case, but this did not work either.

virmaior, 2022-10-25:
unfortunately that's the limits of what I know about this particular use. We'll have to wait until someone more knowledgeable comes along for an answer.

tbrausch, 2022-10-25:
OK, thx!

johnp789, 2023-02-15:
I've been trying to get UVC webcam mode working here, but without success so far.  I have a Wyze Cam V3 with the current 4.36.9.139 firmware loaded.  With `WEB_CAM_ENABLE` set to true, I see it show up as `/dev/video2` and `/dev/video3`.  For example, `v4l2-ctl` looks OK: 
```shell
$ v4l2-ctl --device /dev/video2 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
        [1]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.200s (5.000 fps)
```
However, none of the capture applications I've tried can get any frames.  The system journal shows the following:
```

Feb 14 18:04:13 kernel: usb 1-2: new high-speed USB device number 6 using xhci_hcd
Feb 14 18:04:13 kernel: usb 1-2: New USB device found, idVendor=18d1, idProduct=0002, bcdDevice= 3.10
Feb 14 18:04:13 kernel: usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 14 18:04:13 kernel: usb 1-2: Product: HD USB Camera
Feb 14 18:04:13 kernel: usb 1-2: Manufacturer: Ingenic Semiconductor CO., LTD.
Feb 14 18:04:13 kernel: usb 1-2: SerialNumber: Ucamera001
Feb 14 18:04:13 kernel: usb 1-2: Found UVC 1.00 device HD USB Camera (18d1:0002)
Feb 14 18:04:13 kernel: usb 1-2: Failed to query (GET_INFO) UVC control 6 on unit 2: -32 (exp. 1).
Feb 14 18:04:13 kernel: usb 1-2: Failed to query (GET_INFO) UVC control 9 on unit 2: -32 (exp. 1).
Feb 14 18:04:14 kernel: usb 1-2: Failed to query (GET_INFO) UVC control 4 on unit 2: -32 (exp. 1).
Feb 14 18:04:14 kernel: usb 1-2: Failed to query (GET_INFO) UVC control 11 on unit 2: -32 (exp. 1).
Feb 14 18:04:14 kernel: input: HD USB Camera: HD USB Camera as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/input/input31
Feb 14 18:04:14 kernel: usb 1-2: Warning! Unlikely big volume range (=3072), cval->res is probably wrong.
Feb 14 18:04:14 kernel: usb 1-2: [6] FU [Mic Capture Volume] ch = 1, val = -1536/1536/1
Feb 14 18:04:14 mtp-probe[6640]: checking bus 1, device 6: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2"
Feb 14 18:04:14 mtp-probe[6640]: bus: 1, device: 6 was not an MTP device
Feb 14 18:04:14 mtp-probe[6693]: checking bus 1, device 6: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2"
Feb 14 18:04:14 mtp-probe[6693]: bus: 1, device: 6 was not an MTP device
Feb 14 18:04:14 kernel: usb 1-2: Failed to query (GET_DEF) UVC control 6 on unit 2: -32 (exp. 2).
Feb 14 19:01:35 kernel: usb 1-2: Failed to query (GET_DEF) UVC control 6 on unit 2: -32 (exp. 2).
Feb 14 19:01:36 kernel: usb 1-2: Failed to query (GET_DEF) UVC control 6 on unit 2: -32 (exp. 2).
Feb 14 19:05:42 kernel: usb 1-2: Failed to query (GET_DEF) UVC control 6 on unit 2: -32 (exp. 2).
Feb 14 19:07:05 kernel: usb 1-2: Failed to query (GET_DEF) UVC control 6 on unit 2: -32 (exp. 2).
Feb 14 19:07:21 kernel: usb 1-2: Failed to query (GET_DEF) UVC control 6 on unit 2: -32 (exp. 2).
Feb 14 19:11:16 kernel: uvcvideo 1-2:1.1: Failed to set UVC probe control : -110 (exp. 26).
Feb 14 19:12:39 kernel: usb 1-2: Failed to query (GET_DEF) UVC control 6 on unit 2: -110 (exp. 2).
Feb 14 19:12:46 kernel: usb 1-2: Failed to query (GET_DEF) UVC control 6 on unit 2: -110 (exp. 2).
Feb 14 19:15:06 kernel: uvcvideo 1-2:1.1: Failed to set UVC probe control : -110 (exp. 26).
```
For example, `ffmpeg` reports an input/output error after a few seconds: 
```shell
$ ffmpeg -f v4l2 -input_format mjpeg -video_size 640x360 -i /dev/video2 -t 5 -c:v copy output.mkv
ffmpeg version n5.1.2 Copyright (c) 2000-2022 the FFmpeg developers
  built with gcc 12.2.0 (GCC)
  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-amf --enable-avisynth --enable-cuda-llvm --enable-lto --enable-fontconfig --enable-gmp --enable-gnutls --enable-gpl --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libdav1d --enable-libdrm --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libiec61883 --enable-libjack --enable-libmfx --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librav1e --enable-librsvg --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxcb --enable-libxml2 --enable-libxvid --enable-libzimg --enable-nvdec --enable-nvenc --enable-opencl --enable-opengl --enable-shared --enable-version3 --enable-vulkan
  libavutil      57. 28.100 / 57. 28.100
  libavcodec     59. 37.100 / 59. 37.100
  libavformat    59. 27.100 / 59. 27.100
  libavdevice    59.  7.100 / 59.  7.100
  libavfilter     8. 44.100 /  8. 44.100
  libswscale      6.  7.100 /  6.  7.100
  libswresample   4.  7.100 /  4.  7.100
  libpostproc    56.  6.100 / 56.  6.100
/dev/video2: Input/output error
```
Are there any logs I could collect or troubleshooting steps I could try?

gtxaspec, 2023-07-12:
works here using webcamoid and other apps.