gtxaspec/wz_mini_hacks #371: High CPU usage in self hosted mode

sandstormkeshav, 2022-10-27:
When running the v3 in self hosted mode, I see iCamera.patched using 93% of the CPU. When I was not in self hosted mode, it was about 30%. Is there a reason for the greater usage or is this a bug?

baudev, 2022-10-27:
Same problem on my V3 cam with this mode

kohrar, 2022-10-27:
Yes, I notice this issue too. I haven't been able to quite work the cause, but I suspect my no-ops are causing one of the threads in iCamera to go into a tight loop until something times out.

The CPU usage does seem to taper down after a few minutes past the initial boot phase and it becomes stable after that. Does it for you?

sandstormkeshav, 2022-10-27:
> Yes, I notice this issue too. I haven't been able to quite work the cause, but I suspect my no-ops are causing one of the threads in iCamera to go into a tight loop until something times out.
>
> The [...]

Even after about 15 minutes of uptime I am still seeing about 90% CPU usage

baudev, 2022-10-27:
Yes, 20 minutes after I've 33%!
Seems widely acceptable, thanks

kohrar, 2022-10-27:
Thanks for the reports @sandstormkeshav and @baudev.  That seems to be in line with what I'm experiencing too.

I'll see what I can do about the iCamera patch to address this when I have some free time.

Rustymage, 2022-12-23:
I've noticed that the RTSP stream is a little jumpy in self hosted mode, moving in 2 seconds chunks mostly. I wonder if this is the issue.

I've moved out of self hosted mode for now as these cameras are recording so 2 seconds jumps is not wanted.

I'd be interesting to see if anyone else has seen this in relation to the above issue?

baudev, 2022-12-23:
Could you be more precise ? What do you mean by "2sec chunks" ?

I'm currently using one V3 in self host mode and I've no RTSP problem

Rustymage, 2022-12-23:
the time stamp will go 1, 2, 4, 6, 7, 8, 10... It's random but within a 10 second period of time, you will loose 3-4 seconds on the timestamp.

You can see tis visually too with skipping frames if there's obvious movement, otherwise it's hard to spot.

Removing self hosted mode resolves this.

Video is of issue.

https://user-images.githubusercontent.com/19926955/209306014-cebeead3-8d43-4d4e-bb1f-6036bb95d0db.MOV