gtxaspec/wz_mini_hacks #375: Blue Iris RTSP streams Jitter

sandstormkeshav, 2022-10-27:
I am running the V3 through ethernet but am experiencing frame drops when trying to record through Blue Iris. It works consistently for a few seconds but then the fps goes from 20 -> 5. This does not occur with the sub stream. I have ran iperf between the camera and the server and have confirmed than bandwidth is not the issue. 
Streams seem to playback fine with VLC but wanted to ask if anyone else here has any tips for Blue Iris.

Bob-AOL, 2022-12-16:
Old thread, but...

I currently have five v3s consistently running in the 180kB/s ~ 230kB/s neighborhood at 20~22fps daytime and 15fps at night on my BI server. Three are on UCTRONICS USB/ETH adapters and two are on their own wifi channel. No drops. No latency. Very smooth RTSP performance from all of them.

I've never used the sub stream for the Wyze Cams (or any 2MP cameras for that matter). It's pretty much intended for remote viewing with the Wyze app in 'SD' mode on low bandwidth networks.

Try killing the sub streams and just run with the unicast main stream for awhile and see if that helps.

gtxaspec, 2023-09-26:
please test the latest release with go2rtc rtsp server, it has solved many performance issues.
please reopen if issue still persists.

closed 2023-09-26 by gtxaspec