gtxaspec/wz_mini_hacks #392: Work with V3 attached to flood light

ragenhe, 2022-11-07:
It appears that running wmh on a V3 that has the flood light setup (accessory) will prevent the accessory from working correctly.

I don't care if wmh cannot manage the accessory, but it would be nice if it wouldn't block the accessory from working, if at all possible.

carzine, 2022-12-12:
I also have the same issue.

bberg115, 2023-02-10:
Appears to be working for me and I didn't setup the spotlight in the conf file. I was successful at changing the brightness and the sensitivity. I used some of the info [here ](https://github.com/gtxaspec/wz_mini_hacks/discussions/309)to make changes.

[image]

bberg115, 2023-02-20:
Stopped having access for some reason... 
[image]

gtxaspec, 2023-07-20:
Development continued in #172

closed 2023-07-20 by gtxaspec