gtxaspec/wz_mini_hacks #401: cameras rebooting

orizzle, 2022-11-13:
I don't think this is specific to wz_mini but in general I have a problem with my v3 cameras rebooting and have yet to find a good fix. I log into my cameras via ssh to check uptime and see that some reboot daily, others reboot every few days.  In a 24 hour period, some cameras will capture the full 24 hours of footage but on bad days, some only seem to capture about 23 1/2 hours. I thought about turning on self hosting mode, but I like the event detection. Anyone have any suggestions?

endertable, 2022-11-13:
There is a glitch with Wyze FW that only records 23 hours. I believe some cameras miss the first hour of the day?!?

As far as the rebooting, do you have the cron set up to reboot daily on some of the cameras?  Otherwise the only other real reason can see is that they have network/internet problems, so they reboot.  Do the ones that reboot daily have the same uptime or are they varied greatly?

orizzle, 2022-11-13:
It's not exactly 23 hours. Sometimes it's like 23:30 or 23:45.

no cron jobs to reboot daily. here are some of my cameras and their last boot time, as you can see there is a lot of variability. can i stop them from rebooting for network issues, to rule that out?

 Sun Nov 13 12:40:55 EST 2022
  Sun Nov 13 13:04:34 EST 2022
     Sat Nov 12 23:39:24 EST 2022
      Mon Nov  7 14:35:45 EST 2022

virmaior, 2022-11-14:
@orizzle  
1. what version of the firmware are you using?
2. are the cameras connected to the internet?

orizzle, 2022-11-14:
> @orizzle
> 
> 1. what version of the firmware are you using?
> 2. are the cameras connected to the internet?

4.36.9.139 is what's reported in the app. Although I did have a lower version installed before. After applying wz mini hacks, it started reporting the new version

One camera did seem to update itself automatically to 4.36.10.2700 but the hack is still working

Everything is connected to the Internet

virmaior, 2022-11-16:
> 4.36.10.2700 but the hack is still working
major parts of the hack don't work on that version.
What are you using the hack for?

One thing I noticed is that if I created a high load situation the camera would reboot on its own.

any chance of wifi router issues at those specific times?

orizzle, 2022-11-16:
> > 4.36.10.2700 but the hack is still working
> > major parts of the hack don't work on that version.
> > What are you using the hack for?
> 
> One thing I noticed is that if I created a high load situation the camera would reboot on its own.
> 
> any chance of wifi router issues at those specific times?

I don't see any difference in uptime with the 2700 vs 139. Using the hack mainly to sync videos with my Nas. I wrote a script that runs every 15 minutes to copy videos over the network. Doing it this way instead of writing directly to the Nas in case of outage, I'll have the videos on the sdcard as a backup.  Not sure if copying would give me a high enough load to reboot. 

As far as wifi issues, I guess it's possible. Some cameras are located far away from the router, one camera I even installed a wireless extender in my garage to boost the signal. But the signals are seemingly strong everywhere there are cameras now. 

Is there any way to prevent the cameras from rebooting after losing network, in order to test this?

virmaior, 2022-11-16:
ahh, many people want to use this for RTSP or other uses. That all breaks with 4.36.10.2700.

I don't think that causes a high load (I actually have a cron moving video from SD to RPi).

What might be happening is that for both 4.35.9 and 4.36.10 unless you set it to run without connecting to wyze, then the camera will start rebooting if it loses connection to the internet. If the wifi were dying, it would set this off.

Shouldn't have this issue if you use `ENABLE_SELFHOSTED_MODE="true"` but then wyze app won't work.

orizzle, 2022-11-16:
> ahh, many people want to use this for RTSP or other uses. That all breaks with 4.36.10.2700.
>
> I don't think that causes a high load (I actually have a cron moving video from SD to RPi).
>
> What might [...]

Any way to keep events working with self hosted?

virmaior, 2022-11-17:
I don't know because I don't use events...

At this point, I don't know what you can do other than troubleshoot it. If it were me, I would start by checking the camera logs to see if something specific is happening at the reboot times. Since you do have a NAS system in place, you could also try to use that to pinpoint the timing.

TimDowker, 2022-11-17:
Having the same issue.  Not sure what's happening.  Running 4.36.9.139 firmware.  When I run a ping on it - it will come up for about 7/8 responses then back offline again.

orizzle, 2022-11-18:
> I don't know because I don't use events...
>
> At this point, I don't know what you can do other than troubleshoot it. If it were me, I would start by checking the camera logs to see if something [...]

I found log and rebootlog directories but the log files are not text, perhaps encrypted somehow? Are there logs elsewhere?

virmaior, 2022-11-19:
try `/wz_mini/bin/gather_wz_logs.sh`

baudev, 2023-03-07:
Same problem here, camera is rebooting every 24hours with `ENABLE_SELFHOSTED_MODE="true"` enabled 
Version `4.36.9.139`

The logs at reboot do not seem very relevant... I will export them after 20 hours of uptime

basnijholt, 2024-02-18:
@baudev, I notice random reboots too. Did you even learn more?