gtxaspec/wz_mini_hacks #407: Did firmware 4.36.10.2864 (break/stop/prevent) most future development of wz_mini_hacks

capncybo, 2022-11-25:
@gtxaspec (Hope you are doing okay?)
I've noticed "most" of the  wz_mini_hacks development seems to have stagnated.
Is that because when WYZE released firmware 4.36.10.2864 (November 2, 2022)...
It's essentially broken the primary (RTSP) streaming functionality, moving forward?
I've noticed it seems to be a bit more work to roll-back firmware _(See in BOLD below_)
But I'm somewhat perplexed by their current status regarding their two notes afterwards.
Regardless I've set all my camera configurations NOT to upgrade.
But I'm obviously not sure how PUSHY/Aggressive WYZE will get with their future updates.
Perhaps you are just super busy elsewhere with life & I hope you are well, Take care.

WYZE Cam v3 Firmware:
4.36.10.2864 (November 2, 2022)

    Improved Internet connection code
    Improved live stream connectivity
    Improved IoT connectivity
    Fixed an issue that prevented the device from reconnecting after a router reboot
    Fixed an issue that caused devices to stop working after a firmware update
    Fixed a bug that caused an incorrect membership status after quickly switching membership
    Fixed a bug that caused Error 3001 when viewing motion and sound Events
    Security improvements
    **Due to the security improvement on this firmware, you will need to go through the factory reset process for your camera if you go back to firmware 4.36.9.139 or below**

Note: This firmware is being paused to investigate an issue with event recording.

Note: We are doing a gradual release for this firmware over two weeks. If you do not have this firmware version yet, don't worry! You'll receive it later.

sideup66, 2022-11-25:
Actually yeah, as it sits, the new firmware update effectively breaks ANY RTSP functionality we had. Im suspecting the way forward is possibly a patched version of the icamera bin file, maybe retrofitting it with an older version for usability of rtsp functions.... However im not sure if some of the other proposed fixes exist. Some of the network changes could be handy to possibly deprecate some fixes we have thought up over time. I personally have not upgraded as I heard many complaints about the new versions having some problems with capture and other functions. If we can get around the blockade currently in place (we still have root on the plain v3s, jury is out on the v3pro), you'll probably see development come back. Thing is, I think alot of people are holding off on this update due to the problems, and maybe waiting for the following one, which may be a big bugfix update and smooth things out.

virmaior, 2022-11-28:
I'm actually still back on 4.36.2.5. 
I've got a pull request in for some web improvements but gtxaspec hasn't moved on it.
In general, the current system is working for me and my use case ... but I've got one camera (the one I usually test new things on) which lost all of its /configs directory at some point after I upgraded to the Oct 13 wz_mini code.  It also apparently just happened last night as well ... So my first task is to figure out why / how the entire /configs directory's contents are getting clobbered.

endertable, 2022-11-28:
Yep , the current release is pretty stable so no real reason to keep doing nuance updates that can possibly break something else.  Take vimaior's issue above concerning his /configs directory.  That's challenging to troubleshoot especially if an isolated incident.  

Then there's the necessary questions for other any issues that come up: What FW are you on, what version of wyze_hacks_mini, any custom scripts running, online/offline, rtsp or no, etc, etc.

At this point I figure updates would come from:

A breakthrough on the V3 Pro
Any developer, or otherwise, that can provide hacked iCamera, or have an iCamera wrapper to work on the latest Wyze FW
Any developer, or otherwise, that can provide much requested working binaries, e.g. for simplified local web browser streaming mJpegStreamer, onvif, webrtc, etc.

Just my 2¢

capncybo, 2022-11-28:
@virmaior re: the camera update you received last night... Yikes that is somewhat UN-nerving.
Especially considering it may have put one(or more) of these cameras into a non-functioning state.
I'd really just like to hear #1) that @gtxaspec is actually still alive & OK.
But also #2) if this project has been shutdown via C&D or if @gtxaspec has simply decided to cease future development.
I've notice the Pull Requests just  sitting.
But in days gone bye... @gtxaspec would have likely added them soon after their submission.
Obviously his spare-time could already be maxed out elsewhere.
And if he's done with the project, it's his choice.
I just hope he's Healthy.

skilau, 2022-11-30:
@gtxaspec has responded to threads here and there, but yeah, no idea if he is taking a break, or is working through the possible ways around Wyze's changes.

I know for myself, I am sticking pat for now.

3 of my cameras are all mounted outside, it is winter, and I live in Minnesota.
I have no plans to change anything with them for awhile, as there is no way I want to be climbing around on a ladder in our ice and snow!
(PS: If anyone wanted to know, I had them out there last winter too, and they work perfectly fine in -20F weather!)

DAMAGEDatheCORE, 2022-11-30:
Sticking pat? 🤔

Tbh, if I were gtxaspec, I'd take a well-deserved break before diving into
trying to find workarounds for the latest firmwares.

gtxaspec, 2022-12-03:
Howdy all,

I'm still here, really busy in real life, still developing and investigating the new releases. Working on the floodlight cam at the moment.

Also trying to get a hold of the V3 Pro, but they are sold out everywhere.

dohlin, 2022-12-03:
> Howdy all,
> 
> I'm still here, really busy in real life, still developing and investigating the new releases. Working on the floodlight cam at the moment.
> 
> Also trying to get a hold of the V3 Pro, but they are sold out everywhere.

Great to hear!  I'm brand new to this repo, but after reading the recent concerns I took was worried.  Just wanted to say thank you so much for doing all of this; myself and many others undoubtedly so appreciate your hard work!

MomosX, 2022-12-05:
> Howdy all,
> 
> I'm still here, really busy in real life, still developing and investigating the new releases. Working on the floodlight cam at the moment.
> 
> Also trying to get a hold of the V3 Pro, but they are sold out everywhere.

Glad , very glad you are still with us :)  You made many of us dependent on you as we love this mod. For me the ETH connection is the best feature ever !!!!

digitalpixies, 2022-12-07:
> Glad , very glad you are still with us :) You made many of us dependent on you as we love this mod. For me the ETH connection is the best feature ever !!!!

Same, I rely on the USB gadget support to get the Wyze v3 connected via USB to my rpi. This frees up wifi bandwidth and spread out the frigate nvr workload