gtxaspec/wz_mini_hacks #408: Feature request: Support for Realtek 8153 gigabit ethernet driver

juichi, 2022-11-27:
Hi, thank you for the time and this project.  I recently purchased the "Cable Matters Micro USB to Ethernet Adapter Up to 480Mbps for Streaming Sticks Including Chromecast, Google Home Mini and More - Not Compatible with Roku Device" (https://www.amazon.com/dp/B07N2ZHFY9) to bring ethernet to my wyze cam.  I knew I was rolling the dice but at the time it was cheaper than the uctronics adapter.   As it turns out it has the r8153 chip, which there isn't a driver for.  Apparently you can use the r8152 driver for this chip, but it is beyond my kung fu.

Here is the output from lsusb.

`Bus 001 Device 002: ID 0bda:8153
Bus 001 Device 001: ID 1d6b:0002
`
The second device is the hub. 

thanks!

Bob-AOL, 2022-12-15:
You can manually specify the r8152 driver in the wz_mini.conf USB/ETH settings and disable the auto function to see if that may work. 

#####NETWORKING#####
CUSTOM_HOSTNAME="WCV3"

ENABLE_USB_ETH="true"
ENABLE_USB_ETH_MODULE_AUTODETECT="false"
ENABLE_USB_ETH_MODULE_MANUAL="r8152"

ENABLE_USB_DIRECT="false"
USB_DIRECT_MAC_ADDR="02:01:02:03:04:08"

ENABLE_USB_RNDIS="false"

baudneo, 2024-07-29:
Specifying r8152 doesn't work for my r8153 ethernet adapter. How hard would it be to bundle the 8153 driver? I'll take a poke around and see if its doable but, any advice is appreciated! TIA.