gtxaspec/wz_mini_hacks #420: Upgrade failing with read-only filesystem in the logs

chilicheech, 2022-12-04:
I believe I had the august 20 version. I ran the upgrade script, all seemed well until the v2 camera rebooted. Upon rebooting, after making some clicking sounds, it stays on with the light alternating between blue and yellow. So I pulled the sd card out and looked at the log file and this is what it's showing:
[upgrade_wz_mini.log](https://github.com/gtxaspec/wz_mini_hacks/files/10149045/upgrade_wz_mini.log)

Any ideas on how I can proceed?

Cheers.

akivacp, 2022-12-19:
Experiencing the same issue. Cannot use the upgrade script.

gtxaspec, 2023-07-12:
usually means corrupted sd card.  format the card, do a "zero" format, and try a fresh install.

closed 2023-07-12 by gtxaspec