gtxaspec/wz_mini_hacks #424: No DHCP address on Model T31 with UCTRONICS

Telain, 2022-12-08:
I got a https://amazon.com/dp/B0B9ZLVGJ4 (UCTRONICS Upgraded ... Micro USB to Ethernet/PoE Adapter) and before shipping it back for not working, I want to see if there's additional troubleshooting I should do.

I successfully get WIFI on wz_mini with the camera, but obviously can't access it when I try to enable the ethernet. I did see a MAC address on my switch for the ethernet adapter, but the logs just spam that there's no IP. I tried manually loading the cdc_ether driver, but didn't get anything in that case.

```
#####S03ethernet#####
Auto-Detect an Ethernet Driver and load it
Loading Realtek RTL8152 driver...
Manually load any other Ethernet Drivers if asked for
cat: can't open '/sys/class/net/eth0/address': No such file or directory
```

```
: IP Address not acquired yet, try again in 5 seconds
S12remoteaccessory: IP Address not acquired yet, try again in 5 seconds
K01network: IP Address not acquired yet, try again in 5 seconds

```

Telain, 2022-12-08:
Is there any way I can enable ethernet to test the module and get feedback without disabling the wifi?

Telain, 2022-12-09:
Okay, I've managed to manually load the device drivers by poking through the initialization.

```
KMOD_PATH="/opt/wz_mini/lib/modules/3.10.14__isvp_swan_1.0__"
insmod $KMOD_PATH/kernel/drivers/net/usb/usbnet.ko
modprobe r8152
ifconfig eth0 hw ether 00:15:d5:00:00:01
ifconfig eth0 up
udhcpc -i eth0
```

Now I have an Ethernet IP, so why isn't it working when it does it on its own?

Telain, 2022-12-09:
And now it's working after a reboot, but with the MAC I assigned before. How can I recover the proper MAC address, and can I do anything to help find why assign it properly the first time?

Bob-AOL, 2022-12-11:
Wish I had the answer because I have what is basically the same problem. But first, these adapters work great once they are properly initialized. Not exactly plug'n'play, but relatively easy to get up and running once you realize it isn't the camera and needs its own IP address for its own MAC address.

That said....

After successfully initializing four of these adapters (the 'bad' ones at that) I've encountered a similar issue to yours trying to set up two new V3s for USB ethernet/POE using one of the known good adapters -- which already has a reserved IP address in my router/dhcp server mapped to its MAC address.

After enabling USB ethernet while the camera is in wifi mode (with the adapter in place) and rebooting to initialize the dhcp process for obtaining  the adapter address, it simply hangs in the boot process and never grabs the address. The Wyze app and my monotoring app still show the wifi address as active (which it is), but wz_mini doesn't seem to have loaded as the camera is otherwise none-functional and non-reachable and requires a reboot without the SD card to again become accessible in wifi mode.

I'm back at it this morning, but this certainly seems like a new bug that needs attention.

virmaior, 2022-12-11:
all of my cameras are using usb direct so I don't actually know anything about the usb eth setup.

the two are similar, however, and there's supposed to be a wait loop that waits for the eth device to go up in 5 second increments or so...

@Telain  what do you mean by. `How can I recover the proper MAC address` ?  Do you mean the USB ethernet adapter's factory mac address?

Telain, 2022-12-11:
Yes, I didn't have any luck finding a command that worked on wz_mini_hacks, and was hoping to avoid having to move it to my tablet to see what MAC it gets.

virmaior, 2022-12-11:
@Telain  you should be able to just skip the  `ifconfig eth0 hw ether 00:15:d5:00:00:01 ` line and then it should use its default -- assuming it has one.

Telain, 2022-12-11:
I haven't reentered that command on reboots since, but it's still using the 0015 MAC. I'll just have to resort to my tablet to check it and issue the command again to set it to what it should be.

Bob-AOL, 2022-12-11:
The dhcp discovery of the camera should do all of this when rebooted from wifi mode with the adapter attached and _after_ the USB/Eth feature has been enabled in wz_mini.conf. Enabling USB/Eth must be done with the camera online in wifi mode. When rebooted as such with the adapter attached, the camera should then make the dhcp request which will assign an IP address to the adapter MAC address and disable wifi functionality. The Wyze app itself will show the MAC and IP addresses of the adapter when successfully established. I use Analiti as an additional verification tool which provides several other pieces of information after a successful implementation.

Bob-AOL, 2022-12-11:
I can't get wz_mini to load at all after enabling USB/ETH and rebooting -- which should initiate the dhcp request for the adapter and bring the cam up on the wired network. Need to shut down and pull the card to get it back online.

Interestingly, two hostnames seem to remain even after rebooting without the SD card -- WZ Mini and WZ Mini Web Tools.

This used to be easy. No wonder so many folks think their adapters are bad.

Bob-AOL, 2022-12-11:
I misspoke about the Wyze app showing the MAC address of the adapter. It shows the IP address of the adapter, but the wifi MAC address of the camera remains. The MAC address requires external identification.

Bob-AOL, 2022-12-12:
After successful initialization of the UCTRONICS adapter as displayed with both the Wyze app and Analiti analyzer showing their respective results. Note the original inactive MAC remains in the Wyze app (along with disabled wifi) but is properly identified as the Realtek hardware in the Analiti app. Both correctly display the IP address of the adapter with Analiti offering additional useful details to confirm a successful initialization of the adapter. Great little app if you haven't tried it. Much, much more than just another Android speed test.

Anyway, I'm still pissing up a rope with this and broke down and tried dusting off my Wireshark chops. Boy has it changed and boy am I rusty on top of it. Near as I can figure so far (and as seems evident with the endless 'try again in 5 seconds' loop), we're missing dhcp discovery after activating the USB/ETH feature and rebooting. May be worth digging into by someone with better chops. Aside from that, I'm also stuck on SSH authentication to try editing wz_mini.conf with Nano to eliminate the possibility that editing the file with the web script may itself be the problem. Lots of changes to wz_mini in general since my last batch of successful adapter fire-ups. 

POE USB ethernet for these cams is the reason most folks gravitate here. It's certainly why I recommend they do and why I've spent as much time as I have promoting the UCTRONICS adapter -- _because it works flawlessly when properly initialized and_ _**wz_mini cooperates!**_.

[image]

[image]

Bob-AOL, 2022-12-13:
@virmaior @gtxaspec

I think I can safely say that the web config editor is not applying changes to the wz_mini.conf file unless the camera is rebooted via the web-based reboot command which also seems to save the changes. Modifying the .conf file via the web config editor and rebooting the camera from the Wyze app (as has been my successful method in the past) simply results in the camera rebooting with the existing configuration being re-applied. Rebooting from the web interface saves the changes to the file as observed directly from the  removed SD card, but essentially just seems to kill wifi and never establishes wired connectivity or loads wz_mini at all.

And while I've been successfully able to ssh to the wz_mini.conf file to edit, I now find myself confronted with a read-only file system that will not allow me to save any edits to the file. Talk about frustration. 

Any chance this could be looked into? Initializing one of the UCTRONICS adapters used to be a simple process which now seems to no longer be the case.

Thanks for any time you may be able to spare for this. Something has clearly changed since my last round of

Bob-AOL, 2022-12-13:
Managed to get ssh working to edit wz_mini.conf live and then reboot to initialize dhcp discovery to bring up the adapter. The file edits are successful and wifi is disabled but the camera does not come back up on the adapter.

Since I now have editing capability via ssh, I'm not as concerned with the web tool issue as why the adapters are no longer being recognized even when going about everything in recommended Linux fashion.

Any assistance will be tremendously appreciated.

Bob-AOL, 2022-12-15:
The UCTRONICS comes right up on my laptop. I'm guessing something is amiss with the driver configuration that is not picking up where the wifi leaves off in wz_mini? The camera simply becomes un-reachable until pulling the SD card to re-activate wifi. It tries, but never even spits out a DHCP discovery packet after enabling USB/ETH. And FWIW, the time and date stamps on things are all over the map way ahead in time..., like tomorrow. Not sure how that may be affecting certain functions, but a conflict here and there (everywhere?) can't be good to keep anything in sync that may require it. 

'C'mon fellas. The UCTRONICS is the logical answer to the adapter question at every level if the initialization issue can get ironed out for a simple and dependable solution that just works..., which it does when wz_mini tells a dhcp server it's now on the network as an ethernet device in need of some new and well-deserved respect.

Forgive my frustration. I apologize if I'm coming off as unappreciative or contentious. What you do is magic to me and I'm only trying to help by my obviously limited means of throwin' whatever seems like it may stick at the wall trial and error process of elimination parts swapping slam the hood harder next time and maybe it'll actually start methodology...., for lack of a better description. That said, I have dusted off some command line cobwebs and Wireshark chops to also look at this from a previous life as a broadband network guy, so it's not all just hollering and headbanging. I'm trying contribute where I think I can.

Anyway, the UCTRONICS adapters are wonderful for the v3s and a perfect compliment to wz_mini when they work. This one works just fine...., just not with wz_mini on a v3.

[image]

Bob-AOL, 2022-12-15:
I'm not quite certain how, (I have an idea or two), but the initializations began to be successful again today. See #441

Bob-AOL, 2023-01-13:
A little late, but just to close out my "offerings" above, I ultimately traced my issues to the card reader in the particular camera under test which also seemed to create subtle and random corruption of the card itself. This made effective troubleshooting impossible resulting in chasing my tail with extreme frustration. 

Discovering the problem with the card reader and moving on to a different camera and SD card yielded the expected (and repeatable) results I've become accustomed to when initializing UCTRONICS adapters. 

Apologies for aspersions cast.

xconverge, 2023-01-25:
@Telain were you able to get up and going and stay running? I have reproduced a lot of your findings, however I haven't been able to get anything to stick past a reboot.

Bob-AOL, 2023-01-25:
All of my UCTRONICS devices are working flawlessly with my v3s. Start with a clean wz_mini.conf file. If you can successfully boot the camera from the SD card loading wz_mini, your adapter should be easily initialized by simply enabling usb and auto-detect live online via ssh and rebooting the camera for the changes to be applied. Use the Wyze app for convenience to learn the new IP address of the adapter once the reboot process completes. You may need to do a second restart from the app for the new address to appear.) Be patient. Cloud delay can take a couple of minutes to resolve. The camera's wi-fi IP address and MAC address are no longer relevant and are replaced on the network by that of the ethernet adapter (though the old MAC address will remain in the camera's app and the adapter's will need to be determined by other means.)  The camera is now a completely new device on the network.

xconverge, 2023-01-25:
@Bob-AOL thanks for trying to help, I have seen you post the same thing several times, and I am glad it works for you. There is something more detailed/advanced going on that I am trying to debug

wz_init.log shows bad stuff happening after the attempt to change to use the ethernet adapter. If I disable using the ethernet adapter, I can run through the commands from above that @Telain ran (after looking through the init scripts as well myself)

> KMOD_PATH="/opt/wz_mini/lib/modules/3.10.14__isvp_swan_1.0__"
> insmod $KMOD_PATH/kernel/drivers/net/usb/usbnet.ko
> modprobe r8152
> ifconfig eth0 hw ether 00:15:d5:00:00:01
> ifconfig eth0 up
> udhcpc -i eth0

and can then get an IP address, but I can't get anything to stick past a reboot and am getting tired of swapping the SD card back and forth haha

```
welcome to wz_init.sh
PID 1

 __          ________  __  __ _____ _   _ _____
 \ \        / |___  / |  \/  |_   _| \ | |_   _|
  \ \  /\  / /   / /  | \  / | | | |  \| | | |
   \ \/  \/ /   / /   | |\/| | | | | . ` | | |
    \  /\  /   / /__  | |  | |_| |_| |\  |_| |_
     \/  \/   /_____| |_|  |_|_____|_| \_|_____|
                  ______
                 |______|

#####S01bind#####
Replace stock busybox
Replace stock fstab
Replace stock inittab
Replace /etc/profile for local/ssh shells
Mount kernel modules in /lib
Replace system hostname
#####S02tmpfs#####
Mount wz_mini tmpfs
Create workspace directory
Mounting global tmpfs
#####S03busybox#####
Install busybox applets
#####S04model#####
WYZE_CAKP2JFUS detected
#####S05mount#####
mounting /system
Current Mounts:
rootfs on / type rootfs (rw)
devtmpfs on /dev type devtmpfs (rw,relatime,size=46572k,nr_inodes=11643,mode=755)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
/dev/mmcblk0p1 on /opt type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mtdblock2 on / type squashfs (ro,relatime)
/dev/mmcblk0p1 on /bin/busybox type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /etc/fstab type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /etc/inittab type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /etc/profile type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /lib/modules type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /etc/hostname type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
tmpfs on /opt/wz_mini/tmp type tmpfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw,relatime)
/dev/mtdblock3 on /system type squashfs (ro,relatime)
#####S06car#####
Checking for CAR FW
#####S07inject#####
Copy stock rcS
Add verbose debugging to rcS
Inject init.d scripts to rcS
Modify Global Paths in rcS
Copy factory app_init.sh
Replace factory app_init.sh path
Enable libcallback
set path for iCamera
[... 343 lines cut ...]
S12remoteaccessory: IP Address not acquired yet, try again in 5 seconds
ifconfig: wlan0: error fetching interface information: Device not found
K01network: IP Address not acquired yet, try again in 5 seconds
ifconfig: wlan0: error fetching interface information: Device not found
: IP Address not acquired yet, try again in 5 seconds
ifconfig: wlan0: error fetching interface information: Device not found
S12remoteaccessory: IP Address not acquired yet, try again in 5 seconds
ifconfig: wlan0: error fetching interface information: Device not found
K01network: IP Address not acquired yet, try again in 5 seconds
ifconfig: wlan0: error fetching interface information: Device not found
: IP Address not acquired yet, try again in 5 seconds
ifconfig: wlan0: error fetching interface information: Device not found
S12remoteaccessory: IP Address not acquired yet, try again in 5 seconds
ifconfig: wlan0: error fetching interface information: Device not found
K01network: IP Address not acquired yet, try again in 5 seconds
ifconfig: wlan0: error fetching interface information: Device not found
: IP Address not acquired yet, try again in 5 seconds
ifconfig: wlan0: error fetching interface information: Device not found
S12remoteaccessory: IP Address not acquired yet, try again in 5 seconds
ifconfig: wlan0: error fetching interface information: Device not found
K01network: IP Address not acquired yet, try again in 5 seconds
ifconfig: wlan0: error fetching interface information: Device not found
: IP Address not acquired yet, try again in 5 seconds

```

Telain, 2023-01-25:
I don't understand why but my camera has kept the MAC I manually assigned through reboots.

xconverge, 2023-01-25:
> I don't understand why but my camera has kept the MAC I manually assigned through reboots.

Interesting, then you just enabled the usb adapter flag and the the autodetect flag and it continues to work?

I plugged the adapter into a pi zero w and it came up instantly with an IP so I think the adapter is fine. I will play with the init scripts a bit to see if I can get things going in an order that persists

Telain, 2023-01-25:
Correct. After I gave it a MAC, I haven't had any problems enabling the USB ethernet with auto detect.

xconverge, 2023-01-25:
This logic changed a bit as part of a refactor and I wonder if we just need to spin for a cycle or two as HW comes up for certain HW. 

I will play around more and see

before: 
https://github.com/gtxaspec/wz_mini_hacks/pull/216/commits/0702bb997c9a909846487e95b63f197b02f9b723#diff-8e5ed56a2cba55e3c23991ace7360dc790d88910d1cbce06037b4eede963425dL349

after:
https://github.com/gtxaspec/wz_mini_hacks/pull/216/commits/0702bb997c9a909846487e95b63f197b02f9b723#diff-768d694019649b2ed800d1826268ce9f1bb26cce89d2303458188164fd2456f3R49

Bob-AOL, 2023-01-25:
@xconverge > am getting tired of swapping the SD card back and forth haha

From where to where and why? They do become less and less reliable over time -- especially in and out of cheesy card readers in cheesy little cameras!

May want to try a fresh card and a new and unmolested version of the distro in the most basic form possible just to cover those bases. May also want to fire up Wireshark and see what's happening on ports 67 and 68 as observed from the wi-fi IP address of the camera as you reboot to initialize discover for the adapter. This _must_ occur as a soft reboot with the adapter already attached to the camera and the network. A power cycle restart will simply disable the wi-fi driver if ETH has been already enabled in wz_mini.conf. Consequently a discover request will not occur nor an IP address be obtained for the adapter and the camera will not completely boot since it still thinks it needs an IP address for the hard coded MAC.  

And if I'm being repetitive, forgive me. The process works when implemented as designed on devices that don't otherwise have hardware or firmware issues of their own. Sometimes it's just a 'forest for the trees' kinda thing.

xconverge, 2023-01-25:
I need to swap the sd card to make changes and check logs when the swap from wifi to Ethernet fails because I am left with no means to ssh to it.

I understand Linux and networking enough to understand where to look and what to try. Telling people that it works is not helpful more than once or twice. I will report back any findings next time I look at it

Bob-AOL, 2023-01-25:
> I need to swap the sd card to make changes and check logs when the swap from wifi to Ethernet fails because I am left with no means to ssh to it.

There's your problem.

Bob-AOL, 2023-01-25:
Are you initially making your changes and rebooting online via ssh?

Bob-AOL, 2023-01-25:
Just trying to help, dude. You're missing something very, very simple or it _would_ work.

By all means keep us posted on your progress.

xconverge, 2023-01-25:
I get a dropout after a few minutes after boot ~5 minutes, has happened twice so far. I think I am going to chalk this up to...not worth it for now

Bob-AOL, 2023-01-25:
Is it possible the controller in the upgraded devices is no longer a Realtek? Their prefix is 00:e0:4c...

xconverge, 2023-01-25:
per the original post (which I am experiencing as well and why I posted this here)

I also get
```
lsusb
Bus 001 Device 002: ID 0bda:8152
Bus 001 Device 001: ID 1d6b:0002
```

so I am 90% sure it is a realtek 8152, and the fact that I can bring it up with the modprobe of 8152 tells me yes,..

bjs-pdx, 2023-01-25:
I've been trying to get one of these to work without success.  But here is some of what I have discovered.  I've connected the newer version of the adapter (UC-3AF-USBEthernet-01) and find that on an Ubuntu laptop and on a Raspberry PI nano it never comes up with the same MAC twice...always different.  The original version of the adapter without the -01 works well and I have it connected to my Pan V2.  Here is some output from my DHCP server showing the PI:

[image]

xconverge, 2023-01-26:
I boxed mine up and am sending it back. I will try the OTG route with a different adapter/chipset I already have on hand that is supported, will work better for me to run a usb cable instead of flat ethernet cable anyways, so I can't help anymore, sorry :(

bjs-pdx, 2023-01-29:
I now have 3 of the Uctronics adapters working.  Seems to be an issue with the newer "updated" version of the adapter.  I have the "non-updated" ones and all three came online without any issues.

Link to the ones I used:  https://a.co/hqQUut8

Non-working version:  https://a.co/4Ibxg25

bjs-pdx, 2023-02-13:
Well, I've tested various V2s and one Pan V1 and can say that the Uctronics adapter just doesn't work with any of them.  For the life of me, I can't figure out what changed.  The V2 and Pan V1 WERE working but now do not.  Tried several different adapter/camera combinations without success.

From my point of view, the only cameras to use with the wz_mini_hacks and the Uctronics adapter is the V3.  It works like a champ with minimal effort.  The Pan V2 is limited by the inability to reposition the camera vertically making it just a stationary camera, if you position it correctly before booting the hack firmware and assuming it never moves from the preferred position.

willkitto, 2023-02-14:
I seem to be having the same issues as @xconverge and @Telain using the same UCTRONICS "updated" adapters.
modprobe r8152  works to enable the interface, but with no mac address.

```
# ifconfig -a
eth0      Link encap:Ethernet  HWaddr 00:00:00:00:00:00

```

If I add a HWaddr manually with "ifconfig eth0 hw ether..."  then I can successfully enable and use the ethernet interface.  

But any time I attempt to use the startup scripts using the wz_mini.conf setting of ENABLE_USB_ETH="true" then I get nothing but failure. I have attempted similar startup script type work-arounds as in @xconverge 's post, adding the "ifconfig eth0 hw ether .."  to S03ethernet or before, but for some reason it always fails. 
I can't seem to trace through the startup scripts how the ENABLE_USB_ETH actually does the job of taking over for the wlan0 interface.  I was hoping I could just bypass the numerous if then else statements of the startup scripts and make my own version with the hardcoded Hwaddr added in, but the only place in the rc.d directory that ENABLE_USB_ETH seems to make any difference is the S03ethernet file.  S06networkalt would use it too if bonding is enabled.

Has anyone had any luck with this issue?

Are there any instructions anywhere that help explain how iCamera works so that one could restart it manually using the eth0 interface instead of the wlan0 interface in this weird case where the startup scripts fail because of this missing HWaddr bug?

S07bonding mentions Fooling iCamera in a few places so it seems like this may need to happen when just using eth0 as well, without bonding, but I can't seem to figure out where this takes place.

Bob-AOL, 2023-02-14:
There is nothing wrong with the script and nothing needs to be re-invented to initialize a good adapter. And the camera will not boot to eth0 unless the adapter has been successfully initialized on the network with its own MAC address at least once. The camera needs wi-fi to boot unless somehow told otherwise -- which is accomplished by dhcp discover when initializing the adapter from a wi-fi state live online with the adapter attached to the network and able to obtain an IP address for its MAC address.

Bob-AOL, 2023-02-14:
My guess is you have a bad adapter.

Bob-AOL, 2023-02-14:
@willkitto 

> Are there any instructions anywhere? 

This is how it works with a good adapter and no self-inflicted issues introduced otherwise. 

[UCTRONICS Adapter Setup.pdf](https://github.com/gtxaspec/wz_mini_hacks/files/10728243/UCTRONICS.Adapter.Setup.pdf)

bjs-pdx, 2023-02-18:
I have some speculation on why the "updated" Uctronics adapters may be a problem.  I no longer have one of these but I do have four of the "non-updated" ones working: two on V2s and two on V3s.

When I had the updated version I couldn't get it to work with the cameras...any of the cams.  However, the adapters worked with a Raspberry Pi Zero and my ubuntu laptop.  But they were both assigning random MACs for the cards.  @Bob-AOL suggested that this was due to security settings in the OS which uses random MACs to obscure the device.  But as noted by @willkitto earlier in the post probing the NIC shows no MAC or at least all zeroes.  On the non-updated versions I get a RealTek MAC.

My suspicion is that the updated ones don't have a burned in MAC and the cameras are unable to deal with that and assign a random one.  There is an earlier post that shows the updated ones will work if you assign your own MAC but it doesn't stick past reboot.

Anyway, just my thoughts on this.

bjs-pdx, 2023-02-18:
Further reading and research on the interwebs indicates the lack of MAC could be a manufacturing problem.  Does anyone have contacts with Uctronics who could pass along the findings and ask if the lack of MAC is by design or by accident???

Bob-AOL, 2023-02-18:
I've recently initiated dialogue with them and have received favorable responses regarding their interest in solving the issues experienced by Wyze Cam folks attempting to use their adapter. They don't deny that production issues have been a problem with the product, but aren't convinced the problems experienced by Wyze Cam hack applications don't also have other sources of initialization failures having nothing to do with their device. I agree completely as you well know and as exhibited here by your own experiences.

That said, they have a v3 on the way and instructions for implementing wz_mini_hacks along with a detailed explanation of how to most effectively initialize their device with a properly configured wz_mini SD card. I expect to remain in the loop of their testing and results from purely an experienced end-user perspective and will report accordingly.

The adapters have MAC addresses as Realtek Semiconductor devices. The only difference between the original "bad ones" and the "upgraded" ones is an id to gnd being set in the newer devices for recognition by certain applications that may require it which explains why some adapters may be "seen" by some systems and not others and obviously why one can be seen but not heard (MAC address established) if it is indeed bad or not properly initialized for its particular application -- like hanging off a hacked Wyze Cam that may have any number of inherent issues present in the hard coded firmware or ancillary hardware that prohibit wz_mini from even working as intended in the first place. 

Case in point is the bad adapter recently sent to me by SomebodySysop. It was recognized by my Windows laptop as a Realtek Semiconductor device but that was it. No dhcp discover packet sent, so no way for a MAC address to be identified let alone an IP address assigned. Just a bad adapter for whatever reason -- somewhere beyond the ethernet stage since it at least knew it was on a network and the controller spit out a basic controller ID. It just couldn't get past the transport layer to establish the control data required for going any further. At least that's my take on it. On the other hand, the programming itself could be keeping the MAC from being publicized...

It'll be interesting to see what UCTRONICS comes up with for a solution from their perspective. I will certainly pose any relevant questions beginning with the MAC issue as you've presented it. 
 
A good UCTRONICS adapter generally works with the Wyze devices listed in the README if the methods and considerations provided are followed -- with obvious exceptions for the earlier products and pitfalls everywhere for the adventurous who only read half the required info before embarking on a rewrite of hack to add what they missed to begin with.

The biggest problem with even a perfectly working adapter is the README indicating it should be 'automatically detected' with no explanation of how this is supposed to magically happen...., and it doesn't.

willkitto, 2023-02-19:
@bjs-pdx Thanks for adding your experiences and thoughts
@Bob-AOL Thanks for reaching out to UCTRONICS! I am very interested to hear what they come up with.

Just to reinforce my experience with these adapters: Default MAC address is all zeros. Using ifconfig to change the MAC address, I can make the adapter work on the network. I can use a static ip address or use `udhcpc -i eth0` to get an ipaddress from my dhcp server. All this will work.  I even created a custom startup script that loads the r8152 module and assigns a MAC address. This works successfully, so the system will boot up with a functioning eth0. But, it doesn't work with the wz_hack startup scripts.  If ENABLE_USB_ETH is set to "true" then it fails.  I don't know why. This is why I want to know more about how the startup scripts work so I can just force it to work with a custom startup script. 

I added a new variable to the wz_mini.conf file:
```
#Example
USB_ETH_MAC_ADDR="12:34:56:78:90:12" 
```

And here is my custom startup script that uses the above variable to assign the MAC address to the eth0.
'/opt/wz_mini/etc/rc.d/S02network_preload'

```
#!/bin/sh
### BEGIN INIT INFO
# Provides:
# Short-Description: USB Ethernet Support
# Description:       Preload USB Ethernet adapter stuff
### END INIT INFO

. /opt/wz_mini/etc/rc.common
. /opt/wz_mini/wz_mini.conf

case "$1" in
        start)

                echo "#####$(basename "$0")#####"

                if [[ "$USB_ETH_MAC_ADDR" != "" ]]; then
                        echo "Loading Realtek RTL8152 driver..."
                        insmod $KMOD_PATH/kernel/drivers/net/usb/usbnet.ko
                        modprobe r8152
                        sleep 0.3
                        ifconfig eth0 down
                        sleep 0.3
                        ifconfig eth0 hw ether $USB_ETH_MAC_ADDR
                        sleep 0.3
                        ifconfig eth0 up
                fi
                ;;
        *)
                echo "Usage: $0 {start}"
                exit 1
                ;;
esac
```

Reason for all the "sleep" commands: Running the `ifconfig eth0 hw ether` command immediately after the modprobe r8152 would often fail to assign the MAC address. Adding the sleep commands in-between seemed to help things succeed, presumably this gives enough time for the module to finish loading.

This all works fine when ENABLE_USB_ETH is "false."  But when turning it "true," something breaks for some reason. It seems like it should work, but some step is not functioning.  I can't find any reason in the startup scripts why setting ENABLE_USB_ETH to true would make such a drastic difference.

Bob-AOL, 2023-02-19:
Nice work there. With your permission I'll include it in my next correspondence with UCTRONICS. Same with any other contributors here. Lots information in this thread addressing the heart and soul of how the USB/ETH function is supposed to work. A simple interface toggle that re-triggers dhcp when wi-fi is lost and initializes a wired adapter in its place. 

I personally still see it as a network issue, not coding. A good adapter works without modification to wz_mini. A bad one doesn't, even though it may still work on a more fundamental platform like a Raspberry Pi or other platform capable of seeing and examining devices on more of a raw level -- similar to your approach above. 

As a network guy and not a developer, what I see is a device working at the physical level but not at the data level to even reach the control level. OSI layers 1, 2, & 3. 

The above exercise substitutes the Layer 2 and 3 activity which in turn enables device functionality at layer 1 -- until it is rebooted which removes it from the network, thus requiring new parameters be established for renewed functionality. The script is simply substituting for a non-existent DHCP process because it isn't being initialized when wz_mini is toggled into wired mode. Aside from the possible raw id packet, no discover is even sent. no request returned. no offer, no ack. Nothing.

It's not that the adapter doesn't have a MAC address. It doesn't exist at all as far as the network is concerned. Forcing it into functionality externally indicates to me a data path problem with the device, (or perhaps the process), but not the hack.

Regarding the "true"/"false" issue, wouldn't rebooting the camera to make the change wipe the temporary adapter settings and basically just get you back to square one?

bjs-pdx, 2023-02-19:
@willkitto If you are within your return window for Amazon I'd send back the adapter and get a different one.  This is the link to the ones I have purchased and have been able to get working:

https://a.co/3qrX2cL

Bob-AOL, 2023-02-19:
Ditto, though I have two "upgraded" versions which also work.

I think we're getting closer, fellas.

bjs-pdx, 2023-02-19:
So, with the updated ones, do they show a RealTek MAC or ???

Bob-AOL, 2023-02-19:
Yes.

bjs-pdx, 2023-02-19:
That would seem to differ from the ones with all zeroes or missing any MAC.

willkitto, 2023-02-19:
> Nice work there. With your permission I'll include it in my next correspondence with UCTRONICS. Same with any other contributors here. 

Of course

> It's not that the adapter doesn't have a MAC address. It doesn't exist at all as far as the network is concerned. Forcing it into functionality externally indicates to me a data path problem with the device, (or perhaps the process), but not the hack.
 
Not sure what you are talking about here. Once the adapter is working it is working.  Once the module is loaded and then a MAC address is set, it is then on the network.  The unknown to me is why the wz_hack scripts don't run with it after that point.  My assumption is that there are some other tools, outside of /opt/wz_mini/etc/rc.d that do something, and I just haven't found those functions yet.

> Regarding the "true"/"false" issue, wouldn't rebooting the camera to make the change wipe the temporary adapter settings and basically just get you back to square one?

My above script is in the etc/rc.d directory and therefore runs at boot every time. I named it S02 so that it runs before the S03ethernet script which just checks for which module to load and once loaded does something with its mac address with the following command:
`cat /sys/class/net/eth0/address | tr '[:lower:]' '[:upper:]' > /opt/wz_mini/tmp/eth0_mac`
I haven't found a reference to /opt/wz_mini_tmp/eth0_mac anywhere else, so I can't figure out why it does this.

Bob-AOL, 2023-02-19:
Maybe it finds the real MAC address of the hardware and detects an IP conflict between what it thinks are two different devices? Just thinking out loud now...

I'm sticking with a bad adapter not even being seen by the network until your workaround establishes its presence in place of a normal dhcp process.

Bob-AOL, 2023-02-20:
After wondering about the significance of the ID pin to gnd change as the improvement in the "upgraded" version of the adapter, I spent quite awhile reading about the evolution of USB 2.0, A and B and the mini and micro connectors and how the ID aspect may play into the difficulties being experienced with the newer adapters. 

First, the ID involved is a USB function, not a network function like a MAC address or product ID. Some may already know that. Some may not. I didn't. Whether or not the ID is established and how it is established can determine the functionality of the USB connection at either or both ends of the cable or termination. It is generally how OTG is determined and whether a device is recognized as a host, a client, a charger or a phone or a tablet and how it will be treated by the connection -- host or slave. 

My understanding is that a device intended for a permanent host condition would have a floating ID pin with no connection which would be contrary to the information provided about the "upgraded" adapter now having the ID pin to gnd for OTG functionality. This change was evidently intended for additional flexibility in Raspberry Pi uses and had little if anything to do with Wyze Cams. 

This could be the reason (or one of) for the recent difficulties with the "upgraded" adapters. It doesn't explain the two I ordered from that particular Amazon link working without a hitch unless they simply shipped older adapters (which would not surprise me at all) and I just got lucky for the _fifth_ time with the "bad" ones.. I certainly intend to inquire about this with hopes of receiving some clarification. Anyone with additional knowledge of how the ID pin may be at play here please chime in.

I do notice this evening that there are different delivery dates for each listing (overnight and Tuesday) so there may be a degree of discretion now applied toward distinguishing one from the other -- for better return tracking if nothing else. As recently as last week each version was available overnight to me in the north central Ohio area. I find it highly unlikely they would be located in different distribution centers and still both have immediate availability, but maybe so. The skeptic in me thinks they're probably just pulled from the same bin and dropped into a bag and out the door they go.

Anyway, another angle to contemplate.

Bob-AOL, 2023-02-21:
To sort of wrap this up..., if you don't see something similar from a dhcp capture when firing up a camera behind a UCTRONICS adapter, you either have a bad adapter, bad USB or network connectivity, no dhcp server, or your wz_mini_hacks component is not correctly implemented. Any of these conditions will result in a blank display from no discover packet even being sent. 

What you will see with the necessary connectivity, a good adapter, and wz_mini implemented in its basic configuration as intended is the result below when using Wireshark to capture the activity on ports 67 and 68 during a camera bootup with a UCTRONICS Realtek) adapter. 
 
I had to dust off my Wireshark chops and knock some cobwebs loose, but it doesn't get much more ground floor than this and takes less than 30 seconds to determine if an adapter is announcing itself to the network. If it is, you're halfway home. If it isn't, you'll at least know it isn't and can begin to investigate the reason. I usually start with, "Is it plugged in?" and go from there.

[image]

Bob-AOL, 2023-02-21:
BTW @willkitto, I forwarded your workaround to the UCTRONICS contact for consideration along with several other efforts toward fixing _their_ broken product. The response I received simply conveyed their disappointment in learning that the Wyze Cam they ordered will be delayed another 2 to 4 weeks and they'll keep me posted about any developments when it arrives. Ahem... 

The superficial pleasantries have evidently run their course and I respectfully responded with a request for the information I provided be escalated to an appropriate level for consideration and response. I'm afraid my optimism has waned considerably.

Bob-AOL, 2023-02-24:
@bjs-pdx  @willkitto and anyone else involved or interested; UCTRONICS responded this morning with their first real acknowledgement of what they seem to feel could be one of the main issues with the problematic adapters.

As suspected and illustrated throughout this thread, the MAC issue was a general focal point of our various efforts and angles of workaround and explanation. It would seem from today's response to the material I previously provided for their analysis that they agree and there may indeed be a manufacturing (Realtek) error of missing MAC addresses on many controllers. 

That's the bad news. The good news is, they also have a tool and method for burning in a legitimate MAC address on such adapters..., with caveats, of course. They've provided me with what appears to be a 'starter kit' of sorts to experiment on the bad adapter I received from Sysop. I'll report any progress and may well require some assistance as it moves along.

Props to UCTRONICS. Maybe all is not lost.

bjs-pdx, 2023-02-24:
Glad to see progress on this.  Uctronics adapters really seem to be the better product for this project than any of the others.  The new information certainly supports the conclusion that the adapters were missing MACs.

willkitto, 2023-02-24:
@Bob-AOL  Thanks for the update. This sounds like very promising news.

bjs-pdx, 2023-02-24:
Ask them if they'd like to send us each two of the adapters to test, once they're fixed....

;-)

Bob-AOL, 2023-02-25:
I'm happy to report the tool worked to access the controller of a bad adapter via USB which indeed revealed the lack of a MAC address. Programming the controller was simple and established a Realtek MAC address. Several other parameters were specified as well. The tool basically programs Realtek controllers on a GUI level vs. automated manufacturing programming.
Well beyond my pay grade and I just followed directions.

The adapter became fully functional on the network as analyzed with Wireshark and tested on a live network until plugged into a camera powered by a POE switch which killed it. The adapter became completely inoperable / unresponsive on both USB and ethernet interfaces. The LEDs illuminated but that was it.

My initial thought was whatever programming parameters were in place when burning in the MAC address may also have included controls for any protective features to mitigate POE conditions that were not appropriately set -- allowing power to areas it shouldn't go. 

Made sense to me at the time and I set it aside for the night. I couldn't help thinking about their ID pin to ground upgrade being somehow involved since it does determine host/client/slave powering (and direction) depending on other factors aside from the connection itself. 

Fast forward to this morning when it fired right up and worked perfectly again as previously tested with the laptop and analyzed with Wireshark. Hmmmm. Maybe I was onto something with the POE theory and there is some level of protection built into the device (capacitor?) for such conditions with some type of pre-determined duration before returning it to normal?
A slowly discharging capacitor in such a use would make sense and could be the explanation for the revival.

Time to test it with the camera again, but by powering with a 5V wall wart and unpowered ethernet connection this time.
Same end result, unfortunately. I'll check this evening and see if it fires back up again after a rest and go from there. 

We probably won't hear anything from UCTRONICS on the matter until Monday or Tuesday, but I can't say enough about their support and cooperation as of late. We're getting there.

[6 more replies]

closed 2023-07-12 by gtxaspec