gtxaspec/wz_mini_hacks #442: Bricked Cam?

Rustymage, 2022-12-15:
Hi,

I've been using the hack for a few weeks. I have gone to reboot today after changing a web GUI setting. Now the camera won't boot.

I addition, with the SD card removed, the red light is on from boot. It still isn't connecting or starting.

This is the second camera that has run the hack then died.

I understand hack doesn't modify the files but is there any reason why this could be occuring and any way to recover the cameras?

Thanks in advance.

brokendash, 2022-12-15:
I beleive I'm having a simmile issue, my situation differes slirghtly... I flashed to 4.36.2.5 and then setup the sdcard as required, having issues with the RTSP being jittery and also I could not get the server to deleiver the media via TCP so I downgraded back to the wyze rtsp beta firmware and turned off the RTSP service provided with the wz_mini and used the service in the wyze beta firmware. Everything was working great until randomly I noticed it was offline, and getting red light. The unit responds to a manual firmware reload but then goes to red light and seems unresponsive. I did howeve put console pins on the unit and discovered thart it was not dead and has console output and appears alive, if I remove and install an SDcard it see's it an the kernel displays the info for it etc..  What I do not have at the moment is a log of the serial UART session at this time and I will get one an post it as soon as can.

Rustymage, 2022-12-15:
I should add these cameras have the beta RTSP firmware on them.

Without an SD card, the LED is always lit red, when the WZ hack is inserted, the LED goes red then turns off, yet it never connects to the wifi.

I have no ability to poke at pins.

gtxaspec, 2022-12-16:
@Rustymage which GUI setting did you change? can you detail the steps you took before it stopped responding?

Rustymage, 2022-12-16:
> @Rustymage which GUI setting did you change? can you detail the steps you took before it stopped responding?

I changed the hostname from the default and tagged on the room name is was in after. I hit update then reboot.

The default was WCV3
The change was to WCV3_driveway

Rustymage, 2022-12-16:
In an effort to be more helpful, I made a fresh FAT32 64GB card, loaded the same SD_ROOT copy of hacks I had from the other week (shown to be working) and booted the camera.

Here is the root of the card, zipped up, after I allowed it some time to boot unsuccessfuly - https://we.tl/t-RnhoQJ1r68

Notably, the following might be helpful:

```
welcome to wz_init.sh
PID 1

 __          ________  __  __ _____ _   _ _____
 \ \        / |___  / |  \/  |_   _| \ | |_   _|
  \ \  /\  / /   / /  | \  / | | | |  \| | | |
   \ \/  \/ /   / /   | |\/| | | | | . ` | | |
    \  /\  /   / /__  | |  | |_| |_| |\  |_| |_
     \/  \/   /_____| |_|  |_|_____|_| \_|_____|
                  ______
                 |______|

basename: applet not found
##########
Replace stock busybox
Replace stock fstab
Replace stock inittab
Replace /etc/profile for local/ssh shells
Mount kernel modules in /lib
Replace system hostname
#####S02tmpfs#####
Mount wz_mini tmpfs
Create workspace directory
Mounting global tmpfs
#####S03busybox#####
Install busybox applets
#####S04model#####
cat: can't open '/opt/wz_mini/tmp/configs/.product_config': No such file or directory
cat: can't open '/opt/wz_mini/tmp/configs/.product_config': No such file or directory
 detected
#####S05mount#####
mounting /system
Current Mounts:
rootfs on / type rootfs (rw)
devtmpfs on /dev type devtmpfs (rw,relatime,size=46572k,nr_inodes=11643,mode=755)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
/dev/mmcblk0p1 on /opt type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mtdblock2 on / type squashfs (ro,relatime)
/dev/mmcblk0p1 on /bin/busybox type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /etc/fstab type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /etc/inittab type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /etc/profile type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /lib/modules type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /etc/hostname type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
tmpfs on /opt/wz_mini/tmp type tmpfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw,relatime)
/dev/mtdblock3 on /system type squashfs (ro,relatime)
#####S06car#####
Checking for CAR FW
#####S07inject#####
Copy stock rcS
Add verbose debugging to rcS
Inject init.d scripts to rcS
Modify Global Paths in rcS
Copy factory app_init.sh
Replace factory app_init.sh path
Enable libcallback
set path for iCamera
#####S08passwd#####
Replace stock password
#####S09firstboot#####
First boot, initializing
[... 89 lines cut ...]
Setting up swapspace version 1, size = 134213632 bytes
ifconfig: wlan0: error fetching interface information: Device not found
K01network: Network HW not ready yet, try again in 5 seconds
ifconfig: wlan0: error fetching interface information: Device not found
K01network: Network HW not ready yet, try again in 5 seconds
UUID=3812196d-f303-4304-822f-203080851b4f
Swap file found, enable
#####S03ethernet#####
USB Ethernet disabled
#####S04usbdirect#####
USB Direct disabled
USB RNDIS disabled
No alternate iface enabled
#####S10firmware#####
Firmwware updates enabled, monitor script running
S09coredump: wpa_supplicant not initialized yet, try again in 5 seconds
#####S14nightdrop#####
#####S17motor#####
Motor enabled
#####S18fps#####
Seting system FPS if greater than default
sh: 20: unknown operand
sh: 20: unknown operand
+ /opt/wz_mini/tmp/.storage/app_init.sh
```

Rustymage, 2022-12-17:
Found this - going to try to roll back/forward to see if that helps - https://github.com/gtxaspec/wz_mini_hacks/issues/361

Rustymage, 2022-12-17:
So I had three cameras that had 'bricked'. I mentiond two above. The third one I did not consider to be anything to do with the hacks SW, however I tried the reflash method above.

All cameras were on the RTSP SW before 'bricking' with hacks. Flashing to 4.36.9.139 recovered two of them. The third, will not respond to any FW I flash, even after it shows a purple light.

At least I've save 66% of my investment right?

Rustymage, 2022-12-17:
Here is the log from the last remaining 'bricked camera'

```
welcome to wz_init.sh
PID 1

 __          ________  __  __ _____ _   _ _____
 \ \        / |___  / |  \/  |_   _| \ | |_   _|
  \ \  /\  / /   / /  | \  / | | | |  \| | | |
   \ \/  \/ /   / /   | |\/| | | | | . ` | | |
    \  /\  /   / /__  | |  | |_| |_| |\  |_| |_
     \/  \/   /_____| |_|  |_|_____|_| \_|_____|
                  ______
                 |______|

#####S01bind#####
Replace stock busybox
Replace stock fstab
Replace stock inittab
Replace /etc/profile for local/ssh shells
Mount kernel modules in /lib
Replace system hostname
#####S02tmpfs#####
Mount wz_mini tmpfs
Create workspace directory
Mounting global tmpfs
#####S03busybox#####
Install busybox applets
#####S04model#####
cat: can't open '/opt/wz_mini/tmp/configs/.product_config': No such file or directory
cat: can't open '/opt/wz_mini/tmp/configs/.product_config': No such file or directory
 detected
#####S05mount#####
mounting /system
Current Mounts:
rootfs on / type rootfs (rw)
devtmpfs on /dev type devtmpfs (rw,relatime,size=46572k,nr_inodes=11643,mode=755)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
/dev/mmcblk0p1 on /opt type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mtdblock2 on / type squashfs (ro,relatime)
/dev/mmcblk0p1 on /bin/busybox type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /etc/fstab type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /etc/inittab type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /etc/profile type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /lib/modules type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /etc/hostname type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
tmpfs on /opt/wz_mini/tmp type tmpfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw,relatime)
/dev/mtdblock3 on /system type squashfs (ro,relatime)
#####S06car#####
Checking for CAR FW
#####S07inject#####
Copy stock rcS
Add verbose debugging to rcS
Inject init.d scripts to rcS
Modify Global Paths in rcS
Copy factory app_init.sh
Replace factory app_init.sh path
Enable libcallback
set path for iCamera
#####S08passwd#####
Replace stock password
#####S09firstboot#####
First boot already completed
#####S10kmod#####
Enable 8189fs
Enable atbm603x_wifi_sdio
#####S11configbackup#####
Factory config backup directory missing
[... 134 lines cut ...]
: IP Address not acquired yet, try again in 5 seconds
httpd enabled
S12remoteaccessory: IP Address not acquired yet, try again in 5 seconds
S14storemac: wpa_supplicant not initialized yet, try again in 5 seconds
S09coredump: wpa_supplicant not initialized yet, try again in 5 seconds
#####S13mp4write#####
mp4_write disabled
#####S21syslog#####
Waiting for WiFi settings from QR code...
: IP Address not acquired yet, try again in 5 seconds
S14storemac: wpa_supplicant not initialized yet, try again in 5 seconds
S12remoteaccessory: IP Address not acquired yet, try again in 5 seconds
S09coredump: wpa_supplicant not initialized yet, try again in 5 seconds
Waiting for WiFi settings from QR code...
: IP Address not acquired yet, try again in 5 seconds
S14storemac: wpa_supplicant not initialized yet, try again in 5 seconds
S12remoteaccessory: IP Address not acquired yet, try again in 5 seconds
S09coredump: wpa_supplicant not initialized yet, try again in 5 seconds
Waiting for WiFi settings from QR code...
: IP Address not acquired yet, try again in 5 seconds
S14storemac: wpa_supplicant not initialized yet, try again in 5 seconds
S12remoteaccessory: IP Address not acquired yet, try again in 5 seconds
S09coredump: wpa_supplicant not initialized yet, try again in 5 seconds
Waiting for WiFi settings from QR code...
```

Rustymage, 2022-12-17:
I have no idea now - scratching my head a little. Luckily I can return this camera but we're going to loose any ability for me to test any theories etc shortly.

Let me know if you want logs taking or scripts running. This camera was running well until I installed the hack and reboot via the web gui so something is amiss here.

:)

Rustymage, 2022-12-17:
I've enabled syslog - here it is:

```
May 13 20:22:36 syslogd started: BusyBox v1.35.0
May 13 20:22:36 iCamera: + echo 'inject callback library and run'
May 13 20:22:36 iCamera: inject callback library and run
May 13 20:22:36 iCamera: + export 'LD_LIBRARY_PATH=/thirdlib:/system/lib:/opt/wz_mini/lib'
May 13 20:22:36 iCamera: + '[' -f /opt/wz_mini/tmp/.T20 ]
May 13 20:22:36 iCamera: + LD_PRELOAD=libcallback.so:libtinyalsa.so.2.0.0 /system/bin/iCamera
May 13 20:22:36 timesync: failed to get ip address: Cannot assign requested address
May 13 20:22:36 hl_client: 
May 13 20:22:36 hl_client: ===========welcome to hl_client=============
May 13 20:22:36 hl_client: [Build date] May 13 2022 (16:22:21)
May 13 20:22:36 hl_client: ctx->msgqueueId = 0
May 13 20:22:36 hl_client: metrics_put_data_string
May 13 20:22:36 hl_client: ret = 0 0 105
May 13 20:22:36 assis: 
May 13 20:22:36 assis: ===========welcome to assis process=========
May 13 20:22:36 assis: [Build date] May 13 2022 (16:22:20)
May 13 20:22:36 assis: [shellcall.c:269]msgSndId:32769
May 13 20:22:36 assis: [shellcall.c:284]msgRcvId:65538
May 13 20:22:36 syslog: assisprocess.c,184:g_msgToAssistWdgEMsgId cread success :131076
May 13 20:22:36 assis: ctx->msgqueueId = 0
May 13 20:22:36 assis: metrics_put_data_string
May 13 20:22:36 assis: ret = 0 0 101
May 13 20:22:36 assis: dbg: turn on watchdog success!
May 13 20:22:36 syslog: assisprocess.c,208:recive_wdg_message_looper run 
May 13 20:22:37 sinker: don't have config file /etc/config/.sinker_config, taking default configuration
May 13 20:22:37 sinker: ctx->msgqueueId = 0
May 13 20:22:37 sinker: Init HttpRequest Instance failed
May 13 20:22:37 audiocardprocess: Creating message queue Successfully
May 13 20:22:37 audiocardprocess: Creating thread pool Successfully
May 13 20:22:37 audiocardprocess: Creating timer Successfully
May 13 20:22:37 audiocardprocess: timer_add watchdog is ok 0
May 13 20:22:37 audiocardprocess: [soundcard_watchdog_init] ok TimerId:0
May 13 20:22:37 audiocardprocess: Soundcard watchdog init Successfully
May 13 20:22:37 audiocardprocess: Signal init Successfully
May 13 20:22:37 iCamera: 
May 13 20:22:37 iCamera: ===========welcome to iCamera_app===========
May 13 20:22:37 iCamera: [Build date] May 13 2022 (16:22:28)
May 13 20:22:38 iCamera: [shellcall.c:269]msgSndId:32769
May 13 20:22:38 iCamera: [shellcall.c:284]msgRcvId:65538
May 13 20:22:38 assis: [shellcall.c:400](no.100000) system ret:[0]
May 13 20:22:38 iCamera: [init.c:1662]sysMonitor.sh up success !
May 13 20:22:38 assis: [/system/bin/sysMonitor.sh] start!
May 13 20:22:38 iCamera: [msg.c:88][msg]log: msg_queue_create() ok  MsgId:98307
May 13 20:22:38 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 1 cb
May 13 20:22:38 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 2 cb
May 13 20:22:38 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 3 cb
May 13 20:22:38 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 4 cb
May 13 20:22:38 assis: [shellcall.c:400](no.100000) system ret:[0]
May 13 20:22:38 iCamera: [init.c:1697]copy cacery file success !
May 13 20:22:38 iCamera: [timer.c:148]init complete.
May 13 20:22:38 iCamera: debug_init() dbgId:0 
May 13 20:22:38 iCamera: [msg.c:119][msg]log: assis_IPC_wdg_init() ok  MsgId:131076
May 13 20:22:38 iCamera: [watchdog.c:114]assis_IPC_wdg_init success
May 13 20:22:38 iCamera: [watchdog.c:148]watchdog_init() ok  TimerId:1
May 13 20:22:38 iCamera: system Build:May 13 2022 16:22:30
May 13 20:22:38 iCamera: 
May 13 20:22:38 iCamera: 
May 13 20:22:38 iCamera: [metricsserver.c:757]Metric server start
May 13 20:22:38 iCamera: [metricsserver.c:768]Metric msgQueueId create 0
May 13 20:22:38 iCamera: [metricsserver.c:789]Metric msgQueueId init finish 
May 13 20:22:38 iCamera: [metricsserver.c:851]Create worker thread
May 13 20:22:38 iCamera: [metricsserver.c:854]pthread_create ret = 0
May 13 20:22:38 iCamera: [metricssysinfo.c:300]Successfully scheduled Metrics CPU Info Task
May 13 20:22:38 iCamera: [metricssysinfo.c:309]Successfully scheduled Metrics Memory Info Task
May 13 20:22:38 iCamera: 
May 13 20:22:38 iCamera: ^[[1m<Reading> /configs/.product_config ^[[0m
May 13 20:22:38 iCamera: [product_config.c:249][paracfg_product_config_init] Not This File: /configs/.product_config
[... 10 lines cut ...]
May 13 20:22:46 iCamera: [watchdog.c:73][time:1652473366] feed success (wdgMsgId:131076 getMsgId:131076)
May 13 20:22:46 timesync: failed to get ip address: Cannot assign requested address
May 13 20:22:46 assis: WDG_CMD_FEED_DOG!!!!
May 13 20:22:47 sinker: Init HttpRequest Instance failed
May 13 20:22:48 timesync: failed to get ip address: Cannot assign requested address
May 13 20:22:50 timesync: failed to get ip address: Cannot assign requested address
May 13 20:22:52 sinker: Init HttpRequest Instance failed
May 13 20:22:52 timesync: failed to get ip address: Cannot assign requested address
May 13 20:22:54 timesync: failed to get ip address: Cannot assign requested address
May 13 20:22:56 timesync: failed to get ip address: Cannot assign requested address
May 13 20:22:57 sinker: Init HttpRequest Instance failed
May 13 20:22:58 timesync: failed to get ip address: Cannot assign requested address
May 13 20:23:00 timesync: failed to get ip address: Cannot assign requested address
May 13 20:23:01 audiocardprocess: [watchdog]dbg: soundcard_watchdog_snd_msg_feed [time:1652473381]
[... cut ...]
May 13 20:23:02 timesync: failed to get ip address: Cannot assign requested address
May 13 20:23:02 assis: WDG_CMD_FEED_DOG!!!!
May 13 20:23:04 timesync: failed to get ip address: Cannot assign requested address
May 13 20:23:06 timesync: failed to get ip address: Cannot assign requested address
May 13 20:23:07 sinker: Init HttpRequest Instance failed
May 13 20:23:08 timesync: failed to get ip address: Cannot assign requested address
May 13 20:23:10 timesync: failed to get ip address: Cannot assign requested address

```

Rustymage, 2022-12-17:
Confirmed - using the webugui reboot feature soft bricks them.

Rustymage, 2022-12-17:
I've sent the camera off for a return now.

closed 2022-12-17 by Rustymage

Bob-AOL, 2022-12-17:
I've also experienced it not writing out the file before saving and also wiping clean any previously edited and stored wz_mini.conf backups -- as in deleting the entire contents of the file(s).

That said, and depending on the edit(s) applied, if enabling the USB/ETH feature happens to be one of them, rebooting kills the wifi and dhcp kicks in to obtain an address for whatever ethernet hardware is in place -- provided something is in place that is recognizable by wz_mini and accepts an IP address. Otherwise, the  camera will not even spit out a discovery packet.

The above is probably the least understood aspect of getting the USB/POE?ETH feature to work -- especially among Windows users. And I'm one of em that learned the hard way.

Rustymage, 2022-12-17:
> However, pulling the SD card to reboot into whatever firmware is on the camera should still work is normal.

This doesn't work if it's soft bricked. On 2 out of 3, I managed to recover the cameras, however, for the third, it remained soft bricked.

I used the working SD card in the continually soft bricked camera and it won't correct the issue. It flashes to the other FW version, but it will not connect to the wifi - with or without the hack SD card installed.

Bob-AOL, 2022-12-17:
The syslog file would seem to possibly indicate an IP address conflict.  It's requesting one, just unable to assign the one requested.

May 13? time sync issue?

Rustymage, 2022-12-17:
> The syslog file would seem to possibly indicate an IP address conflict. It's requesting one, just unable to assign the one requested.
> 
> May 13? time sync issue?

I saw this but wsn't sure how to resolve it. The cam wasn't even reaching out to the access point so to my knwoledge never tried to connect.

Removing the SD card did not allow me to use the 'Setup' button in stock FW mode either so I had no way to give it WiFi credentials.

brokendash, 2022-12-18:
That was what was happening to me too when I pulled my SD card it would not
boot up normally it was behaving the same way where it would produce the
red light. But upon disassembling it and putting terminal headers on the
console pens it's stuck but it's still responsive if I insert an SD card
and remove it it actually sees it and prints info into the console. I too
was running the beta rtsp firmware from wyse.

Bob-AOL, 2022-12-18:
#361 is good advice from a voice of experience.

My money is on a bad SD card or card reader in that particular camera. If a new ((not just different) Class 10 FAT32 SD card with a fresh and up-to-date firmware version can't be flashed, the card reader itself may be the problem --  and/or the existing firmware has been corrupted beyond recovery for any number of reasons most likely introduced by the user somewhere along the line.

Been there. Done that...., and willing to admit it.

87435, 2023-07-21:
Just came across this - I have the same issue #564 
Were you able to recover the cameras?

virmaior, 2023-08-06:
I've just run into the same error
1. had a camera that appeared to not be saving to its SD card
2. SSHed
3. `reboot`
Then 
`ifconfig: wlan0: error fetching interface information: Device not found`

virmaior, 2023-08-06:
In my case, this is because at some point the /configs system got clobbered on this camera. It's been months since this happened but running `reboot` at just the wrong timing seems to have reclobbered it.

restored by copying it back in.

87435, 2023-08-06:
> In my case, this is because at some point the /configs system got clobbered on this camera. It's been months since this happened but running `reboot` at just the wrong timing seems to have reclobbered it.
> 
> restored by copying it back in.

Interesting! Can you expand on that please? @gtxaspec also mentioned a `.configs` or `config` folder in my other issue. What should it contain and where do you restore it from?

virmaior, 2023-08-06:
> Interesting! Can you expand on that please? @gtxaspec also mentioned a `.configs` or `config` folder in my other issue. What should it contain and where do you restore it from?

just posted at https://github.com/gtxaspec/wz_mini_hacks/issues/587#issuecomment-1666896848