gtxaspec/wz_mini_hacks #450: Feature Request: Add support for the CoreChips SR9700 Micro-usb Ethernet Adaptor

fearthis4, 2022-12-31:
First of all I just wanted to say thank you again for all the time and work you have done on this project it is amazing!

I recently purchased a few cheap micro-usb ethernet adaptors on Ebay: https://www.ebay.com/itm/125368589938? "Micro USB 2.0 5 Pin to Ethernet 10/100Mbps RJ45 Network LAN Adapter Converter F" to convert my cameras to hardwired ethernet on my wyze cam v3 along with a POE splitter adapter. I knew I was probably risking it as you never really know what chipset is included with these types of purchases but at the time it was cheaper than the uctronics adapters and fit into the solution I was going to try to workout. 

As it turns out it has the a CoreChips SR9700, which isn't currently supported. 
[image]
However this is a supported chipset in linux with the sr9700 driver.

Here is the output from dmesg:
[    0.985436] usb 1-1: device v0fe6 p9700 is not supported

Here is the output from lsusb:
Bus 001 Device 002: ID 0fe6:9700

Here is what I could find:
https://linux-hardware.org/index.php?id=usb:0fe6-9700
This appears to the be driver: https://github.com/torvalds/linux/blob/master/drivers/net/usb/sr9700.c

Thanks again in advance for your time and help, I know there are likely more pressing issues in the queue here so no urgency or criticality, I figured it might be long shot but I assume it could be useful for others out there that might be interested in a low cost adapter to use for their camera's..

gtxaspec, 2023-07-11:
i'll take a look to see if this driver is present in our kernel version.

gtxaspec, 2023-07-13:
SR9700 not included in kernel source, I'll see if I can backport the newer one to work with our kernel.

fearthis4, 2023-07-16:
Thank you again for all of the help!

adbot1337, 2024-05-14:
i have a particular project im working on... how would one install this driver in termux?

genericrobotperson, 2025-07-15:
Hello.  I have been experimenting a bit and found something that worked.  I am using the SR9900 one
so the vendor/product code is 0fe6:9900

I used the r8152 drivers