gtxaspec/wz_mini_hacks #451: Webserver interface no longer reachable after soft reboot?

rocket59, 2022-12-31:
So I was making some changes to the config via the web interface and after rebooting the camera from the reboot link in that section the camera came back online as expected but the interface is no longer reachable. I can get to top level menu (headed "wz_mini web server") but all the second level items can't be accessed any more, it requires a hard reboot (pulling power) for them to reappear.

virmaior, 2023-01-01:
interesting.

after a hard reboot everything seems dandy?

> So I was making some changes to the config

just to clarify that's to the wz_mini config or the camera config?

rocket59, 2023-01-01:
Yes, hard reboot solves the issue.

And also yes, I was making changes to the wz_mini config via the web interface editor. As I mentioned the camera (it's a V3 cam, I forgot to mention this in the OP) does actually start streaming.

virmaior, 2023-01-01:
trying to figure this out. The top level page is just HTML; the other pages are cgi (ash -- which is basically bash from 30 years ago).
1. if you change via the web config, then soft reboot, can you ssh into the camera still?
2. If you can ssh post soft-reboot, does the `wz_mini.conf` file reflect your changes?
3. what type of changes are you making? (rtsp? frame rate? usb ethernet?)

rocket59, 2023-01-01:
Thanks for the suggestions. I will try SSHing into the camera and report back (probably tomorrow) on point 2 of your post also.

I was making RTSP settings changes but I'll try some other settings at the same time.

virmaior, 2023-01-01:
I wrote most of the web-facing code, so I can work on at least that much.

The main camera I've been using for development (of 5) is kind of fritzing itself periodically which has lowed my involvement.

Bob-AOL, 2023-01-13:
This screen will never refresh as indicating it will in 90 seconds. Simply re-entering the camera IP address will restore the home page.

[image]

virmaior, 2023-01-13:
@Bob-AOL - this is fixed in the pull request - https://github.com/gtxaspec/wz_mini_hacks/pull/372 -- or at least I believe I fixed it for that run. 
I think in a previous version I had commented out the refresh line by mistake

Bob-AOL, 2023-01-13:
The screenshot above is from a reboot issued just several minutes ago using Chrome. I've never had the screen refresh on its own to the home screen URL after a reboot.

Bob-AOL, 2023-01-13:
It would seem like a different instruction for returning to the home screen would be the simplest solution!

Bob-AOL, 2023-01-13:
Another very important point would be when activating USB/ETH functionality via the web tool, saving the changes and rebooting, the USB adapter will then initiate its own dhcp discovery for obtaining a completely different IP address than the browser was previously resolving.