gtxaspec/wz_mini_hacks #455: Rtsp profiles, videofmtp and audio fmtp when using Janus WebRTC server and RTSP.

baudneo, 2023-01-10:
I opened an issue before about profile-level-id. I am just wondering if I should open an issue in atomcam repo?

Hi-res for v2 sends a blank video fmtp in it's DESCRIBE response. This causes janus to error because it can't parse fmtp. Janus allows overriding videofmtp and several other options but only after it receives the server options.

The hi/low res streams work in VLC and other software, this is a uanus only issue but, I would like to see if I can help to get it fixed.

The audio that is sent from v2 is in L16 44.1 kHz mono format. Janus works with L16 16 kHz and 48 kHz only at the moment so, I was wondering if there is a way to set audio format and also if it's possible to configure the video fmtp.

I am guessing it's an upstream issue. Perhaps I need to reconfigure and recompile things?

Any insight or advice would be welcome.

Thanks for your hard work

gtxaspec, 2023-07-12:
try the new go2rtc implementation.  it supports opus @48khz

baudneo, 2023-07-14:
Sweet, I've had go2rtc on the back burner. Planning to implement 2 way audio with lorex doorbells using zoneminder web UI.

I'll upgrade this weekend and see how it goes.

Thanks for your work!

baudneo, 2023-09-04:
Closed as go2rtc implementation is different from teh codebase which this issue is about.

closed 2023-09-04 by baudneo