gtxaspec/wz_mini_hacks #459: run-upgrade.sh caused boot loop on V3

baudneo, 2023-01-11:
Title says it all, ran the upgrade script and it never turned back on. Had to retrieve camera to now do autopsy on sdcard to see whats going on.

What files should I save from sdcard if its not corrupted? Isnt there some fiel that holds the key to wyze AWS stuff?

baudneo, 2023-01-11:
```
File verification successful!
Move staging directory to perform upgrade
Installing latest upgrade-run from repo
Launching latest upgrade-script
Exit old script
[root@WCV3:~]# Resume upgrade-run, latest version
check for old directory path
enable wifi drivers if disabled in config
Backup user config
Rebooting into UPGRADE MODE
client_loop: send disconnect: Broken pipe
```

Boot loop ever since

baudneo, 2023-01-11:
upgrade log shows the sdcard was in RO mode. How can I avoid that in the future?
```
rm: can't remove '/opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/www/cgi-bin/cam-nightVision.md': Read-only file system
rm: can't remove '/opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/www/cgi-bin/cam-night_cut_thr.md': Read-only file system
rm: can't remove '/opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/www/cgi-bin/cam-res.md': Read-only file system
rm: can't remove '/opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/www/cgi-bin/cam-timezone.md': Read-only file system
rm: can't remove '/opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/www/cgi-bin/cam-values.txt': Read-only file system
rm: can't remove '/opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/www/cgi-bin/cam-watermark_flag.md': Read-only file system
rm: can't remove '/opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/www/cgi-bin/cam.cgi': Read-only file system
rm: can't remove '/opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/www/cgi-bin/car.sh': Read-only file system
rm: can't remove '/opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/www/cgi-bin/config.cgi': Read-only file system
rm: can't remove '/opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/www/cgi-bin/config.css': Read-only file system
rm: can't remove '/opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/www/cgi-bin/diagnostics.cgi': Read-only file system
rm: can't remove '/opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/www/cgi-bin/jpeg.cgi': Read-only file system
rm: can't remove '/opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/www/cgi-bin/numerics.txt': Read-only file system
rm: can't remove '/opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/www/cgi-bin/shared.cgi': Read-only file system
rm: can't remove '/opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/www/config.js': Read-only file system
```

gtxaspec, 2023-07-11:
SD card in RO mode means corruption, could be bad card or bad blocks.  

Reformat the card, fill it with zeros, and try again.  

Zeroing the card is required before trying again is needed.

baudneo, 2023-07-11:
It was a weird issue, remounting with rw fixed it for me.

closed 2023-07-12 by gtxaspec

87435, 2023-07-13:
> It was a weird issue, remounting with rw fixed it for me.

Can you explain how you did it? same issue with `read-only file system`

baudneo, 2023-07-14:
I cant remember but I found the command somewhere in this repo. In discussions or issues. A simple comment that shows how to remount the file system RW.

baudneo, 2023-07-14:
Found it: `mount -o remount,rw /dev/mmcblk0p1 /opt`