gtxaspec/wz_mini_hacks #460: pan v3 work in progress

gtxaspec, 2023-01-11:
partial bootlog:

```
U-Boot SPL 2013.07 (Jun 28 2022 - 07:05:47)
Timer init
CLK stop
PLL init
pll_init:366
pll_cfg.pdiv = 10, pll_cfg.h2div = 5, pll_cfg.h0div = 5, pll_cfg.cdiv = 1, pll_cfg.l2div = 2
nf=116 nr = 1 od0 = 1 od1 = 2
cppcr is 07405100
CPM_CPAPCR 0740510d
nf=100 nr = 1 od0 = 1 od1 = 2
cppcr is 06405100
CPM_CPMPCR 0640510d
nf=100 nr = 1 od0 = 1 od1 = 2
cppcr is 06405100
CPM_CPVPCR 0640510d
cppcr 0x9a7b5510
apll_freq 1392000000 
mpll_freq 1200000000 
vpll_freq = 1200000000
ddr sel mpll, cpu ����������1@V��������?|�����������<�<���8@x����?���� � L�A�`����������0�
U-Boot SPL 2013.07 (Jun 28 2022 - 07:05:47)
Timer init
CLK stop
PLL init
pll_init:366
pll_cfg.pdiv = 10, pll_cfg.h2div = 5, pll_cfg.h0div = 5, pll_cfg.cdiv = 1, pll_cfg.l2div = 2
nf=116 nr = 1 od0 = 1 od1 = 2
cppcr is 07405100
CPM_CPAPCR 0740510d
nf=100 nr = 1 od0 = 1 od1 = 2
cppcr is 06405100
CPM_CPMPCR 0640510d
nf=100 nr = 1 od0 = 1 od1 = 2
cppcr is 06405100
CPM_CPVPCR 0640510d
cppcr 0x9a7b5510
apll_freq 1392000000 
mpll_freq 1200000000 
vpll_freq = 1200000000
ddr sel mpll, cpu sel apll
ddrfreq 600000000
cclk  1392000000
l2clk 696000000
h0clk 240000000
h2clk 240000000
pclk  120000000
CLK init
SDRAM init
sdram init start
ddr_inno_phy_init ..!
phy reg = 0x00000007, CL = 0x00000007
ddr_inno_phy_init ..! 11:  00000004
ddr_inno_phy_init ..! 22:  00000006
ddr_inno_phy_init ..! 33:  00000006
REG_DDR_LMR: 00000210
REG_DDR_LMR: 00000310
REG_DDR_LMR: 00000110
REG_DDR_LMR, MR0: 00f73011
T31_0x5: 00000007
T31_0x15: 0000000c
T31_0x4: 00000000
T31_0x14: 00000002
INNO_TRAINING_CTRL 1: 00000000
INNO_TRAINING_CTRL 2: 000000a1
T31_cc: 00000003
INNO_TRAINING_CTRL 3: 000000a0
T31_118: 0000003c
[... 316 lines cut ...]
line_num:1
[FC] not found tfcard or usb store
[FC] Test.tar no exist
[FC] In [user] mode!
kernel.core_pattern = |/system/bin/ucoredmp_collector.sh --pid %p --signal %s --name %e --time %t --output-dir /media/mmc/cores
kernel.core_pipe_limit = 1
net.unix.max_dgram_qlen = 128
[    7.316364] name : i2c0 nr : 0
[    7.405391] i2c i2c-0: i2c_jz_irq 441, I2C transfer error, ABORT interrupt
[    7.412521] i2c i2c-0: --I2C txabrt:
[    7.416292] i2c i2c-0: --I2C TXABRT[0]=I2C_TXABRT_ABRT_7B_ADDR_NOACK
[    7.422859] error: sensor_read,287 ret = -5
[    7.427202]  sensor_read: addr=0x3107 value = 0x0
[    7.432052] err sensor read addr = 0x3107, value = 0x0
[    7.525700]  sensor_read: addr=0xf0 value = 0x20
[    7.530950]  sensor_read: addr=0xf1 value = 0x53
[    7.535770] info: success sensor find : gc2053
[    7.540387] misc sinfo_release
[    7.746239] gc2053 chip found @ 0x37 (i2c0) version H20211111a
[    8.455212] codec_set_device: set device: MIC...
[    8.735292] codec_set_device: set device: speaker...
[    8.746937] SPEAKER CTL MODE3 !
[   12.538999] [motor_speed1078] speed:900, level:15
[   12.543854] motor_ops_reset960
```

gtxaspec, 2023-01-11:
how to open:

the cube with the lens has a black plastic cover, peel this off with a blade, knife or something sharp.  Go slow so all the adhesive stays on the cover.
[image]
Remove the cover, then take the 4 screws off.  Pull the front cover off. Pry it out just like on the v3.
[image]
Serial ports are here:
[image]
Square is ground, Middle is TX, edge is RX.

gtxaspec, 2023-01-11:
Camera Board photos
[image]
[image]
[image]
[image]

capncybo, 2023-01-11:
And I had thought that perhaps you'd given up the tinker'n... Good Luck & Happy New Year !!!

endertable, 2023-01-12:
"let's break it open and see."

You weren't kidding.  😭

dsanner, 2023-01-12:
Since the wyze pan v3 is a rebranded Atom Swing camera ...is it possible to flash this on it to gain RTSP and SSH?  https://github.com/mnakada/atomcam_tools

endertable, 2023-01-12:
I think WYZE updated the firmware on it and made it more secure.  I am going to guess this is going to use the same FW iteration as the Cam Pro version, which has not been compromised yet, or not that I know.

virmaior, 2023-01-13:
@dsanner I think it's more accurate to say they are both white labels from the same Chinese manufacturer. The AtomCams tend to deviate a bit less from the white label product provided.

mastan30, 2023-03-07:
@gtxaspec any luck with getting the pan v3 rooted?

monkollek, 2023-03-10:
> I think WYZE updated the firmware on it and made it more secure. I am going to guess this is going to use the same FW iteration as the Cam Pro version, which has not been compromised yet, or not that I know.

@gtxaspec @virmaior When you have a moment, can you please elaborate on what you need help on? 
If the above holds true then using `factory_t31_ZMC6tiIDQN` on the SD card will not work?

monkollek, 2023-03-12:
@gtxaspec or anyone else. Do you have a version of the U-boot logs with SD card in? As I'm guessing the SD card was not in with the example you showed above: `SD card is not insert`

I'm not sure if it looks for things on the SD card if is inserted in but just curious.

KyloRilo, 2023-06-10:
Hey everyone, does anyone have an idea of the progress on this? I was looking to set up a livestream of the pan v3 for a local nonprofit using rtsp

TheCableGuy99, 2023-06-15:
Same here, bought the Wyze Cam Pan v3 and looking for RTSP hopefully, i'm guessing there's no progress on this thought because no one appears to be replying?

dstockman, 2023-06-16:
> Hey everyone, does anyone have an idea of the progress on this? I was looking to set up a livestream of the pan v3 for a local nonprofit using rtsp

Hey @KyloRilo, can't speak to the progress on getting support in wz_mini_hacks but as a stop gap, I've had good luck using mrlt8/docker-wyze-bridge to expose RTSP for these cams. HTH.

TheCableGuy99, 2023-06-16:
> mrlt8/docker-wyze-bridge

I looked at this a while ago but can't think now why I didn't use it. Does it give you an RTSP stream so I can still record it on my NAS?

dstockman, 2023-06-16:
Yep!

TheCableGuy99, 2023-06-16:
Excellent, i'll take a re-look at it then. I hope it also allows me to control the PTZ manually. I don't use the Wyze cam app because I put my cameras on a vLAN with no internet access after they have been setup as I don't want my feed going through their servers for anyone to peek in on my private life. I've heard stories of companies doing that. My cameras are for my local or secure remote access only.

TheCableGuy99, 2023-06-16:
> Yep!

ahh I just thought, does it grab the image from the cloud? That might be why I decided didn't use it as I don't stream them.

TheCableGuy99, 2023-07-12:
oh great....I have 2 cameras just sat there that I cannot use until there's an RTSP stream :)

gjmv, 2023-09-13:
Hi, i have a local vendor camera, that is based on ATOM Cam Swing / Wyze Pan v3, with provider custom firmware.
I try to load this mod but is failing on startup (i open the camera and solded console pins and connected to PC via COM)
Can you give me some help to start the bootfs and at least load initial script (then, i want to analize why is not working, but now i haven't booted yet)
If it helps, mnakada (atomcam_tools) boot works, but i want to try your mod.

I think the error is in
`
[    0.954786] mmc0: new high speed SDHC card at address 59b4
[    0.960663] mmcblk0: mmc0:59b4 LX32G 29.5 GiB
[    0.966647]  mmcblk0: p1
[    2.254784] jzmmc_v1.2 jzmmc_v1.2.0: timeout 500ms op:17 r sz:512 state:2 STAT:0x1F000140 DMALEN:0x00000200 blks:0/1 clk:enable clk_gate:enable
[    2.459359] mmcblk0: unknown error -1 sending read/write command, card status 0xb00
[    2.467315] end_request: I/O error, dev mmcblk0, sector 2048
[    2.473158] FAT-fs (mmcblk0p1): unable to read boot sector
`

This is the full boot log
`
U-Boot SPL 2013.07 (Sep 13 2021 - 17:24:15)
Timer init
CLK stop
PLL init
pll_init:366
pll_cfg.pdiv = 10, pll_cfg.h2div = 5, pll_cfg.h0div = 5, pll_cfg.cdiv = 1, pll_cfg.l2div = 2
nf=116 nr = 1 od0 = 1 od1 = 2
cppcr is 07405100
CPM_CPAPCR 0740510d
nf=100 nr = 1 od0 = 1 od1 = 2
cppcr is 06405100
CPM_CPMPCR 0640510d
nf=100 nr = 1 od0 = 1 od1 = 2
cppcr is 06405100
CPM_CPVPCR 0640510d
cppcr 0x9a7b5510
apll_freq 1392000000
mpll_freq 1200000000
vpll_freq = 1200000000
ddr sel mpll, cpu sel apll
ddrfreq 600000000
cclk  1392000000
l2clk 696000000
h0clk 240000000
h2clk 240000000
pclk  120000000
CLK init
SDRAM init
sdram init start
ddr_inno_phy_init ..!
phy reg = 0x00000007, CL = 0x00000007
ddr_inno_phy_init ..! 11:  00000004
ddr_inno_phy_init ..! 22:  00000006
ddr_inno_phy_init ..! 33:  00000006
REG_DDR_LMR: 00000210
REG_DDR_LMR: 00000310
REG_DDR_LMR: 00000110
REG_DDR_LMR, MR0: 00f73011
T31_0x5: 00000007
T31_0x15: 0000000c
T31_0x4: 00000000
T31_0x14: 00000002
INNO_TRAINING_CTRL 1: 00000000
INNO_TRAINING_CTRL 2: 000000a1
T31_cc: 00000003
INNO_TRAINING_CTRL 3: 000000a0
T31_118: 0000003c
T31_158: 0000003c
T31_190: 0000001e
T31_194: 0000001c
jz-04 :  0x00000051
jz-08 :  0x000000a0
[... 850 lines cut ...]
[   18.488145] Modules linked in:
[   18.491113]  atbm603x_wifi_sdio(O+) speaker_ctl(O)
[   18.496061]  mmc_detect_test(O) sample_pwm_hal(O)
[   18.500911]  sample_pwm_core(O) sinfo(O)
[   18.504951]  avpu(O) audio(O)
[   18.508003]  exfat(O) tx_isp_t31(O)
[   18.511786] Process  (pid: 0, threadinfo=84042000, task=804e4fb8, tls=84042150)

[   18.519329] Stack : 00000000 00000000 80500000 80482de4 ffff91fd 0000000b 6c0a0100 ffff0006
          8417e200 00000000 0000000b 8008eeb8 80593621 0000000a 00000000 801ff9e8
          84042368 80592da9 80592da8 804f728c 804e634c 805821e0 0000000b 00000028
          00000000 00000000 00030000 8402d5cc 8402d550 8008efc0 00000004 804f728c
          00001000 80096984 80593620 804e634c 8047cc10 800920e8 804e20f4 00001000
          ...
[   18.556189] Call Trace:
[   18.558713] [<802082cc>] add_interrupt_randomness+0x30/0x1a4
[   18.564549]
[   18.566074]
Code:
[   18.568046]  afa30010  10400004  afa40014
[   18.572273] <8c4200ac> afa0001c  afa20018  3c06804f  24c4757c  94880014
`

Sorry about my english
Thanks in advance

gtxaspec, 2023-09-20:
As it stands now, the pan v3 won't be supported unless the user re-flashes, or replaces the flash memory chip containing the stock bootloader.

We will keep trying to find an alternate solution.

gtxaspec, 2023-10-01:
this camera has secure-boot enabled by the factory, and can not be disabled.  replacing the flash module, SPL, or u-boot results in non-boot.
```
U-Boot SPL 2013.07 (Jun 28 2022 - 07:05:47)
Timer init
CLK stop
PLL init
pll_init:366
pll_cfg.pdiv = 10, pll_cfg.h2div = 5, pll_cfg.h0div = 5, pll_cfg.cdiv = 1, pll_cfg.l2div = 2
nf=116 nr = 1 od0 = 1 od1 = 2
cppcr is 07405100
CPM_CPAPCR 0740510d
nf=100 nr = 1 od0 = 1 od1 = 2
cppcr is 06405100
CPM_CPMPCR 0640510d
nf=100 nr = 1 od0 = 1 od1 = 2
cppcr is 06405100
CPM_CPVPCR 0640510d
cppcr 0x9a7b5510
apll_freq 1392000000 
mpll_freq 1200000000 
vpll_freq = 1200000000
ddr sel mpll, cpu sel apll
ddrfreq 600000000
cclk  1392000000
l2clk 696000000
h0clk 240000000
h2clk 240000000
pclk  120000000
CLK init
SDRAM init
sdram init start
ddr_inno_phy_init ..!
phy reg = 0x00000007, CL = 0x00000007
ddr_inno_phy_init ..! 11:  00000004
ddr_inno_phy_init ..! 22:  00000006
ddr_inno_phy_init ..! 33:  00000006
REG_DDR_LMR: 00000210
REG_DDR_LMR: 00000310
REG_DDR_LMR: 00000110
REG_DDR_LMR, MR0: 00f73011
T31_0x5: 00000007
T31_0x15: 0000000c
T31_0x4: 00000000
T31_0x14: 00000002
INNO_TRAINING_CTRL 1: 00000000
INNO_TRAINING_CTRL 2: 000000a1
T31_cc: 00000003
INNO_TRAINING_CTRL 3: 000000a0
T31_118: 0000003c
T31_158: 0000003c
T31_190: 0000001e
T31_194: 0000001c
jz-04 :  0x00000051 
jz-08 :  0x000000a0 
jz-28 :  0x00000024 
DDR PHY init OK
INNO_DQ_WIDTH   :00000003
INNO_PLL_FBDIV  :00000014
INNO_PLL_PDIV   :00000005
INNO_MEM_CFG    :00000051
INNO_PLL_CTRL   :00000018
INNO_CHANNEL_EN :0000000d
INNO_CWL        :00000006
INNO_CL         :00000007
DDR Controller init
DDRC_STATUS         0x80000001
DDRC_CFG            0x0aa88a42
DDRC_CTRL           0x0000011c
DDRC_LMR            0x00400008
DDRC_DLP            0x00000000
DDRC_TIMING1        0x050f0a06
DDRC_TIMING2        0x021c0a07
DDRC_TIMING3        0x200a0722
DDRC_TIMING4        0x26240031
DDRC_TIMING5        0xff060405
DDRC_TIMING6        0x321c0505
DDRC_REFCNT         0x00910503
DDRC_MMAP0          0x000020f8
DDRC_MMAP1          0x00002800
DDRC_REMAP1         0x030e0d0c
DDRC_REMAP2         0x07060504
DDRC_REMAP3         0x0b0a0908
DDRC_REMAP4         0x0f020100
DDRC_REMAP5         0x13121110
DDRC_AUTOSR_EN      0x00000000
sdram init finished
SDRAM init ok
board_init_r
verify_buf_equal!
verify signature uboot error!
```

endertable, 2023-10-01:
This means can’t use openipc, either?

gtxaspec, 2023-10-01:
Unfortunately at the moment, no.

sasdallas, 2023-11-26:
Oh, unfortunate. Just got one of these and hoped it would be supported.
That's not on devs though, it's on me for not checking - thankfully I can return it and hopefully get a better one.
I doubt there's gonna be an easy way to do this, especially with that sort of chip-level protection, but good luck to you anyways!

gtxaspec, 2024-01-02:
closing for now until a solution is found.  feedback and additional testing is welcome

closed 2024-01-02 by gtxaspec

pzich, 2024-01-19:
Has anyone managed to get their hands on one of the atom cams? Do they have the same limitations? Has anyone managed to figure out the white label vendor?

virmaior, 2024-01-19:
both myself and @gtxaspec have atomcam 2s (https://www.atomtech.co.jp/products/atomcam2) in addition to wyze cams ... but do you mean a specific model?

I'm using atomcam_tools on that. It probably can run wz_mini but for my purposes I don't need it to.

gtxaspec, 2024-01-20:
@virmaior I think he's referring to the atom cam swing.  The wyze version in the US has a locked SOC, and there is significant demand for a model that can be unlocked, like the atomcam swing or personalcam of Latin America which all share the same body.

pzich, 2024-01-23:
Whoops sorry for the delay here. @gtxaspec has it right, I was referring to the atom cam swing and it hopefully being unlocked (or in some way more hackable) than the wyze variant.

Sparpet, 2024-05-19:
Only way to bypass secure boot on thi camera is performing an SoC swap, i have performed one myself so i can actually test code with it. Unfortunately I dont have a dump of the original flash and flashed thingino instead so i cannot boot wyze firmware any more, but I can confirm that after SoC swap Wyze firmware boots normally.

Here are some photos of the process:
[image]
[image]
[image]
[image]
[image]