gtxaspec/wz_mini_hacks #466: v2 bricked?

ryny24, 2023-01-25:
I'm trying to recover a v2 that was running wz_mini but I accidentally formatted the SD card within the app.  It wouldn't come back online, so I tried the Wyze setup again without an SD card. It keeps saying Unable to connect to network but I just did this with a few OG camers which worked fine.

I've also tried the Wyze recovery demo.bin, but I can never get the blue light to come on while holding the setup button.

Every time it boots with a blank SD card, it creates a ".wz_mini_initramfs" file. And in the file, it says it's booting stock but 
+ echo 'wz_mini not found, booting stock'
wz_mini not found, booting stock
+ mount --move /dev /wz/dev
+ mount --move /sys /wz/sys
+ mount --move /proc /wz/proc
+ cp /tmp/wz_mini_initramfs.log /sdcard/.wz_mini_initramfs.log

I also tried to perform the wz_mini setup again, by cloning the repo but when I run ./compile_image.sh it fails about a missing URL.

build for v2
wget: missing URL
Usage: wget [OPTION]... [URL]...

Try `wget --help' for more options.
md5sum: ./v2_ro/demo.zip: No such file or directory
md5sum: ./v2_ro/demo.zip: No such file or directory
md5sum failed check, please manually supply file

Any thoughts on what to try next?

Thank you!

bjs-pdx, 2023-02-01:
I need to look back at the V2 setup.  I recall having to copy the Wyze firmware file locally so that the installer could find it since it doesn't seem to find the URL.

gtxaspec, 2023-07-20:
@ryny24 did you ever figure this out?