gtxaspec/wz_mini_hacks #470: Change config

jritsema, 2023-01-29:
Hello.  I didn't see it mentioned in the docs, but what is the intended way to change config (`SD_ROOT/wz_mini/wz_mini.conf`) options (e.g., change host name or RTSP creds)?

Is the intent to remove the sd card, delete all files, and replace it with a fresh config?

xconverge, 2023-01-31:
Removing the SD card is one valid way although most inconvenient as you have found, I edited my config quite often over the ssh connection (/media/mmc/* with nano and reboot command), and there is also the webserver (https://github.com/gtxaspec/wz_mini_hacks#web-server) that lets you edit/reboot from the website for the camera too with some checkboxes and text boxes on the website.

You should not need to delete files and simply changing the config file and restarting should apply I think

jritsema, 2023-02-05:
Awesome, that's what I was looking for.  Thanks @xconverge!

closed 2023-02-05 by jritsema