gtxaspec/wz_mini_hacks #477: PAN V2 Vertical Tilt

bjs-pdx, 2023-02-10:


endertable, 2023-02-11:
If cam is within reach, unplug cam, remove SD card, start cam normal and adjust view with app.  I believe the camera remembers its PT position between boots.  Then unplug cam, reinsert card, start cam.

bjs-pdx, 2023-02-11:
Yes, thought about that.  However, it is a temporary solution.  I might want to change the view on the camera and don't want to go through that each time.

Is there any plan to fix the issue or is it going to just be that way?  If there isn't any plan to fix it I'd probably be better off using a fixed position V3.

bjs-pdx, 2023-03-04:
The vertical tilt seems to work, sort of, if you set the motor speed to 9.  It is glitchy, but it does move.  It doesn't always go to the end, until you reboot the camera.

closed 2023-03-04 by bjs-pdx

wokawoka, 2024-04-02:
I see this issue as closed and already completed about one year ago, but I'm experiencing the problem on my PAN V2 running a recent verion of wz_mini_hacks.
Am I missing something? Do you have a suggestion?
Thank you!

wokawoka, 2024-04-02:
apparently it looks still open here https://github.com/gtxaspec/wz_mini_hacks/issues/19