gtxaspec/wz_mini_hacks #478: Video blurring/artifacts

pdobrien3, 2023-02-11:
I have a Wyze V3 on Wifi.  Using frigate for recording.  I have tried everything to stop the blurring / artifacts on the recorded video but cant figure it out.  I have heard that using RTSP over TCP fixes the issue.  Is this possible?  here are my settings:

```
RTSP_HI_RES_ENABLED="true"
RTSP_HI_RES_ENABLE_AUDIO="true"
RTSP_HI_RES_FPS="10"
RTSP_HI_RES_MAX_BITRATE="4096"
RTSP_HI_RES_TARGET_BITRATE="2048"
RTSP_HI_RES_ENC_PARAMETER="2"

RTSP_LOW_RES_ENABLED="true"
RTSP_LOW_RES_ENABLE_AUDIO="false"
RTSP_LOW_RES_FPS="5"
RTSP_LOW_RES_MAX_BITRATE="2048"
RTSP_LOW_RES_TARGET_BITRATE="1024"
RTSP_LOW_RES_ENC_PARAMETER="2"

```

```
  garage_camera: 
    ffmpeg:
      input_args: preset-rtsp-udp
      output_args:
        detect: -dn -an -f rawvideo -pix_fmt yuv420p
        record: -dn -f segment -segment_time 10 -segment_format h.264 -reset_timestamps 1 -strftime 1 -codec:v copy -codec:a aac -b:a 24k -ac 1 
      inputs:
        - path: rtsp://user:pass@192.168.xx.xxx:8554/video2_unicast
          hwaccel_args: preset-intel-qsv-h264
          roles:
            - detect
        - path: rtsp://user:pass@192.168.xx.xxx:8554/video1_unicast
          hwaccel_args: preset-intel-qsv-h264
          roles:
            - record
```

bailout00, 2023-02-16:
+1 I'm seeing the same issue, only seems to happen on the v3, random glitches of 1/3 of the screen, or blocks across the whole thing for many frames. Seems like a compression/keyframe issue.

pdobrien3, 2023-02-16:
> +1 I'm seeing the same issue, only seems to happen on the v3, random glitches of 1/3 of the screen, or blocks across the whole thing for many frames. Seems like a compression/keyframe issue.

I pretty much have just accepted it. Any chance you are using wz_mini with a wyze pan v1?  I have had one heck of a time with getting recordings on my pan to work in frigate.

mach7, 2023-02-22:
I have also been having the same issues. But it doesn't happen on all of my V3s. The setup is the same on each camera too. So I'm not sure exactly what is causing it either.

sjmellen, 2023-02-23:
Same issue here on V3's.  As @bailout00 noted, random glitches on lower 1/3 of the screen or whole screen for many frames.  Camera connected wirelessly.  Observed both streaming to VLC as well as streaming to a "picture glace card" in Home Assistant.
BTW - anyone know how to force TCP... don't see anything in the configuration files.

mach7, 2023-02-24:
Any chance that those of you who are seeing the green glitchy video are also using frigate?
I'm getting the green glitch when I am viewing my cameras in a picture glance card from Home assistant with the source being my frigate video feed, at the same time it happens I'm watching on the wyze app and on VLC the only place I get anything other then clean video is viewing the video feed from frigate.

IR2Lazy, 2023-02-24:
I have green, purple or just fuzzy (mostly bottom half) in Home Assistant using picture entity and Zoneminder. Saw it in MotionEye when testing also. Didn't see issue while using VLC.

Zoneminder wiki for WyzeCam mentions the use of LibVLC, but I never got that to work 

https://wiki.zoneminder.com/Wyze_Cam

Enable RTSP via Wyze app.

Note: RTSP is only available with special, non-standard firmware. See instructions here: https://support.wyzecam.com/hc/en-us/articles/360026245231-Wyze-Cam-RTSP

Use RTSP path provided by app in this format:

rtsp://[user]:[password]@[IP]/live

Regardless of app settings, capture settings are 1920x1080 at 32 bit colour. Do not use maximum FPS.

FFMPEG results in significant artifacts; use of LibVLC is recommended instead.

clee604, 2023-02-25:
> Same issue here on V3's. As @bailout00 noted, random glitches on lower 1/3 of the screen or whole screen for many frames. Camera connected wirelessly. Observed both streaming to VLC as well as [...]

I believe TCP can be set on the client side; for me I 'force' TCP by using simple-rtsp-server to proxy the wyze RTSP feed using 'sourceProtocol: tcp' in the simple-rtsp-server config. This eliminated the artifacting for me.

pdobrien3, 2023-02-26:
Ok, so this is not one of the default configuration settings for wz_mini_hacks. So, I you wouldn’t mind, could you please tell us how you are using tcp for wz_mini_hack video. Everything I have read has said that doing video over tcp fixes the issue. I just can’t figure out how to do it. Thanks in advance for any assistance.

clee604, 2023-02-26:
I have some wyze cameras running wz_mini_hacks with RTSP enabled. Instead of connecting to cameras directly, I run a rtsp proxy using simple-rtsp-server on a linux host. In the simple-rtsp-server config where I reference the wyze RTSP feeds, I set the sourceProtocol to TCP. The end clients (ie. VLC, TinyCam), point to the rtsp proxy instead of the cameras directly.

I don't know about VLC, but I know TinyCam allows you to specify the protocol, and you can set it to be TCP there.

pdobrien3, 2023-02-26:
Nice. I just spun up a container version. Was able to proxy my streams. Need to play with settings still. Thank you.

pdobrien3, 2023-02-26:
@clee604 do you get a lot of packet loss in your simple-rtsp-server logs?  Still seeing the artifacts when going through simple-rtsp-server. I wonder if it is just packet loss due to a Wi-Fi connection.  Although like everyone else says, the stream is clear in VLC.

clee604, 2023-02-26:
> @clee604 do you get a lot of packet loss in your simple-rtsp-server logs? Still seeing the artifacts when going through simple-rtsp-server. I wonder if it is just packet loss due to a Wi-Fi connection. Although like everyone else says, the stream is clear in VLC.

I got some RTP packet loss errors when I left it on the default (ie. using UDP) but after forcing TCP in the simple-rtsp-server yaml I haven't noticed any errors in the logs, though I haven't been monitoring them carefully since I haven't gotten the artifacts anymore

pdobrien3, 2023-02-26:
So I think I finally fixed it by increasing the bitrate figures and lowering the frame rate to 15 fps.  Not using anything but wz_mini_hacks. 

Max bitrate=8192
Target bitrate=4096
Enc parameter=2
FPS=15

still early on but looks promising.

mach7, 2023-02-26:
Can you tell me what your work flow is from the camera to when you view it?

pdobrien3, 2023-02-27:
Wyze V3 with wz_mini to Frigate. After about 12 hours, here is where I am.  It is not fixed with the wz_mini config changes. I am all but positive it has to do with frame drops and where you are viewing the feed. I have a windows 10 desktop and a windows 11 desktop. When viewing frigate recording on the windows 11 desktop, I see the issue manifest in video artifacts. When I watch the frigate recorded video on the windows 10 desktop, I see the issue manifest with glitching video but no artifacts. The time stamp either hesitates or goes backwards sometimes. On both computers, when watching the live video with VLC, no artifacts but the live video pauses.  All along, simple-rtsp-server still reports frames being dropped. Both my wyze v3s are connecting via Wi-Fi.

mach7, 2023-02-27:
> Wyze V3 with wz_mini to Frigate. After about 12 hours, here is where I am. It is not fixed with the wz_mini config changes. I am all but positive it has to do with frame drops and where you are viewing the feed. I have a windows 10 desktop and a windows 11 desktop. When viewing frigate recording on the windows 11 desktop, I see the issue manifest in video artifacts. When I watch the frigate recorded video on the windows 10 desktop, I see the issue manifest with glitching video but no artifacts. The time stamp either hesitates or goes backwards sometimes. On both computers, when watching the live video with VLC, no artifacts but the live video pauses. When being used to convert to TCP, simple-rtsp-server still reports frames being dropped. Both my wyze v3s are connecting via Wi-Fi.

That last comment you made about the wifi is one of the things I've been wondering about too. I have a mesh router system showing up tonight in hopes to clear up some wifi issues that I'm having in our home. When I was ordering it I was thinking about this thread and also wondered, what type of wifi system you had setup with your environment. Is it a standard router or are you also running a mesh system?

As for me it seems like I have a similar setup. I'll watch mine more closely and report on it when I start taking some notes on it as well. 

@gtxaspec - Do you have any input on this thread?

pdobrien3, 2023-02-28:
I have a DIY mesh with asus routers and sd-wrt. I am 99% sure the issue is caused by frame drops. I have been able to get the frame drops to an acceptable level on one of my Wyze cameras but haven’t had a chance to verify the results.

shehzman, 2023-05-02:
Just setup wz_mini_hacks on my v3 and am running Frigate and am also having these issues. Were you able to figure out a solution?

pdobrien3, 2023-05-02:
Pretty sure it is a bandwidth issue. I gave up. Are you running them wireless?  I think I was able to improve it a little by keeping them on separate routers in my house.

shehzman, 2023-05-02:
I am running it wireless, but then I switched to the official RTSP firmware
and it worked near perfectly so I don’t think it’s a bandwidth issue for me

pdobrien3, 2023-05-16:
So, I have 100% fixed the blurring/artifacts in frigate by using the restream:

```
go2rtc:
  streams:
    your_camera: 
      - "ffmpeg:rtsp://your_user:your_password@192.168.xx.xxx:8554/video1_unicast#video=h264#audio=aac"

```
Here is my camera setup in frigate:

```
  your_camera:
    ffmpeg:
      hwaccel_args: preset-vaapi
      input_args: preset-rtsp-udp
      output_args:
        detect: -dn -an -sn -f rawvideo -pix_fmt yuv420p
        record: -sn -dn -f segment -segment_time 10 -segment_format mp4 -reset_timestamps 1 -strftime 1 -codec:v copy -codec:a aac -b:a 16k -ac 0
      inputs:
        - path: rtsp://your_user:your_password@192.168.xx.xxx:8554/video2_unicast
          roles:
            - detect
        - path: rtsp://localhost:8554/your_camera
          input_args: preset-rtsp-restream
          roles:
            - record
```
Here is my docker-compose

 ```
 frigate:
    container_name: frigate
    privileged: true
    restart: unless-stopped
    image: ghcr.io/blakeblackshear/frigate:stable
    shm_size: "256M"
    devices:
      - /dev/bus/usb:/dev/bus/usb
      - /dev/dri/renderD128:/dev/dri/renderD128
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - /your/path/docker/frigate/config/config.yml:/config/config.yml:ro
      - /your/path/docker/frigate/data:/media/frigate
      - type: tmpfs
        target: /tmp/cache
        tmpfs:
          size: 1000000000
    ports:
      - "5000:5000"
      - "8554:8554"
    environment:
      FRIGATE_RTSP_PASSWORD: "your_password"
      PLUS_API_KEY: your_key
      LIBVA_DRIVER_NAME: i965

```

Still working through ffmpeg crashes in Frigate though.  Funny, I have two of the same Wyze cameras and only one seems to be crashing, although I have only been testing this setup today.  It is rock solid in record and mse playback though.

pdobrien3, 2023-05-17:
keep getting this:

```
2023-05-16 20:32:18.808069842  [2023-05-16 20:32:18] watchdog.your_camera    ERROR   : No new recording segments were created for your_camera in the last 120s. restarting the ffmpeg record process...
2023-05-16 20:32:18.808452006  [2023-05-16 20:32:18] watchdog.your_camera    INFO    : Terminating the existing ffmpeg process...
2023-05-16 20:32:18.809152291  [2023-05-16 20:32:18] watchdog.your_camera    INFO    : Waiting for ffmpeg to exit gracefully...
```

gtxaspec, 2023-07-20:
Try the new release with go2rtc.