gtxaspec/wz_mini_hacks #485: Question: How to monitor SDK call made by iCamera?

etiennedub, 2023-02-19:
Hi, I'm trying to reverse some call made by iCamera to `liblocalsdk.so`. 

My end goal would be to support at least a new command "waitForAudio" trigger on audio detection. The wyze app support this but I'm not sure which call to hook into. For that, I was thinking about monitoring calls made to liblocalsdk.so and poke around the app.

Do you have any way to monitor all library call made to `liblocalsdk.so`? So far, I tried to hook iCamera with frida-gadget but their static mipsel build doesn't seem to work on my wyze camera. I haven't try to build it personally. I also try to build `ltrace` with the buildroot toolchain from [mnakada/atomcam_tools](https://github.com/mnakada/atomcam_tools), but it's building with glib dependency instead of uclibc.

Any idea how to achieve that, or do you use another toolchain to build binary for this repo?

gtxaspec, 2023-07-12:
ude ghidra or ida to decompile the library, then you can use LD_PRELOAD on the iCamera app to load a custom .so as we do in libcallback.  

https://github.com/gtxaspec/wz_mini_sources/tree/master/libcallback
https://github.com/gtxaspec/wz_mini_hacks/blob/master/SD_ROOT/wz_mini/usr/bin/iCamera

gtxaspec, 2023-09-19:
Converting to discussion for visibility

closed 2023-09-19 by gtxaspec