gtxaspec/wz_mini_hacks #487: Webcam on V3 does not work !

abastola0, 2023-02-26:
Tested the webcam functionality on v3 didn't work for me.

akivacp, 2023-03-03:
I noticed that some programs won't accept webcam, but some do. YouCam seems to work for me.

abastola0, 2023-03-03:
Oh, is it? I even tried capturing the frame with python opencv, but that didn't work. The RTSP that it streams don't seem to be coming direct from the camera else that would suffice. I get around 1 second delay at first which ruins the whole purpose for me as i wanted something to be processed in realtime. Not sure if it's encoding latency or it's just rerouting the frames after being sent to the internet to the local endpoint.

SolAZDev, 2024-04-13:
Can Confirm. Just downgraded my v3 to the last supported firmware, and my linux systems just show black when trying to see the camera. In **SOME** ocassions there is a still frame. So this could be a v4l2 config issue _in_ the camera? -- Are there any suggestions to test?

Tahlwyn, 2024-05-03:
Seems to work for me on Linux (tested OBS and Discord) but Windows (10) almost fully fails to recognize the device, simply showing up as "USB Composite Device". Not sure if Windows support is within scope but it would certainly be convenient.

SolAZDev, 2024-05-08:
> Seems to work for me on Linux (tested OBS and Discord) but Windows (10) almost fully fails to recognize the device, simply showing up as "Unknown USB Device". Not sure if Windows support is within scope but it would certainly be convenient.

For me it shows on devices and is recognized as a camera, but it shows a black screen.

Tahlwyn, 2024-05-11:
It doesn't show up as a camera at all for me. In Devices and Printers it shows up literally as "Unknown USB Device". Sometimes (but not always) it will say "Unknown USB Device (Device Descriptor Request Failed)". It appears the same in Device Manager.

Rahzadan, 2024-05-19:
Works fine on my V3 & Win11. Shows up as "HD USB Camera", and functions properly.

SolAZDev, 2024-05-20:
@Rahzadan  do you mind sharing with us if you did any configs? And what firmware you camera's on?

Rahzadan, 2024-05-21:
@SolAZDev - very simple actually. All I did was change **WEB_CAM_ENABLE="false"** to true - and changed nothing else. I only put wz_mini_hacks on my V3 the other day. I discovered it was on a very recent firmware, so I downgraded it to the last supported one (4.36.9.139). After doing all that, webcam mode is working perfectly.

SolAZDev, 2024-05-26:
@Rahzadan then your setup is mostly the same as mine. Except I'm running Arch Linux, and Debian, neither system read anything from the camera. So I'll try to redoing the whole setup.

Rahzadan, 2024-05-27:
@SolAZDev I assume you've also tried the cam on Windows? I also have one of my V3's in Webcam mode attached to an OpenWRT router via USB (not ethernet). Installed the video-uvc kernel module, along with mjpg-streamer modules, and it outputs 1080p @ 25fps.

SolAZDev, 2024-05-27:
@Rahzadan No, I don't use windows, at all. It is irrelevant for me, and my servers use Linux. However, a USB device such as a webcam should work regardless of platform. (Drivers are usually a Windows-only thing)