gtxaspec/wz_mini_hacks #490: One camera freezes after 2 minutes of uptime

lachesis, 2023-03-05:
I have an isolated camera network with no internet access. I have two Wyzecam v3s, both runing wz_mini_hacks, both on version 4.36.9.139. I use RTSP streaming. One camera works excellently, with no drops and 5 days uptime since my last power outage. The other is new today, and seems to freeze after 2-3 minutes of runtime, even if I do not have the RTSP stream connected. They share the same wz_mini.conf and .user_config. I don't really know how to troubleshoot further, but I'm happy to capture whatever logs might be helpful. When the camera restarts, it does not recover on its own. My SSH connection freezes and I have to power-cycle it. However, if I kill iCamera.patched, it still restarts after 2 minutes, but it self-recovers - perhaps because some watchdog is restarting it. This camera has never been connected to the internet at all.

lachesis, 2023-03-05:
One note: If I disable RTSP streaming, the camera CPU usage is much lower, and it does not crash after 2-3 minutes. FWIW I am using the original power adapter and cable, and it is cool here today so overheating should not be an issue.

Here are some files from the camera, in case they are helpful:

/configs/.product_config
```
[XIAOMI]
CONFIG_INFO=D03F271ACFF6|||0000

[TYPE]
PRODUCT_MODEL=WYZE_CAKP2JFUS
PRODUCT_TYPE=camera

[HUALAI]
FACTORY_TEST=0000
BOARD_TEST=0FFF
KEY_CHECKSUM=00003f00
HARDWARE_VER=F00A0000
AGING_TIME=250
KEY=u/V6yNpet+hr0qJyqeacnI4WJbF6zFlTboN4s80hEeP1dARmvqreXDpICx8O28oEx4bfK/nMf94P836Q3G0c427VWWbr/YoGmO1JmYQh0/gcwv/zJVBD3y8PjOf4tPFhKFHlqXEoUAlZOC5JCZZ2jqBWLwM59nHLhkUASa1dqyU=AQAB

[TUYA]
CONFIG_INFO_UUID=KU55H8U86DWDV6KD111A
```

/configs/.user_config
```
[NET]
bindOk=1
p2pid=
r_encr=

[SETTING]
timezone=-4
drawBoxSwitch=2
MASwitch=2
AASwitch=2
nightVision=2
night_led_ex=2
NIGHT_LED_flag=2
indicator=2
watermark_flag=2
osdSwitch=1

[CAMERA_INFO]
TZ_STR=
```

/media/mmc/wz_mini/wz_mini.conf
```
#wz_mini configuration

#### W E B CAMERA###
##THIS MODE DISABLES EVERYTHING AND IT WILL
## WORK AS A WEB CAMERA FOR YOUR PC ***ONLY***
WEB_CAM_ENABLE="false"
WEB_CAM_BIT_RATE="8000"
WEB_CAM_FPS_RATE="25"

#####NETWORKING#####
CUSTOM_HOSTNAME="wyzecam2"

ENABLE_USB_ETH="false"
ENABLE_USB_ETH_MODULE_AUTODETECT="true"
ENABLE_USB_ETH_MODULE_MANUAL=""

ENABLE_USB_DIRECT="false"
USB_DIRECT_MAC_ADDR="02:01:02:03:04:08"

ENABLE_USB_RNDIS="false"

ENABLE_IPV6="false"

ENABLE_WIREGUARD="false"
WIREGUARD_IPV4=""
WIREGUARD_PEER_ENDPOINT=""
WIREGUARD_PEER_PUBLIC_KEY=""
WIREGUARD_PEER_ALLOWED_IPS=""
WIREGUARD_PEER_KEEP_ALIVE=""

ENABLE_IPTABLES="false"

ENABLE_NFSv4="false"

ENABLE_RTL8189FS_DRIVER="true"
ENABLE_ATBM603X_DRIVER="true"

NTP_SERVER="192.168.61.1"

##### NETWORK INTERFACE BONDING #####
BONDING_ENABLED="false"
BONDING_PRIMARY_INTERFACE="eth0"
BONDING_SECONDARY_INTERFACE="wlan0"
BONDING_LINK_MONITORING_FREQ_MS="100"
BONDING_DOWN_DELAY_MS="5000"
BONDING_UP_DELAY_MS="5000"
BONDING_FAIL_OVER_MAC="0"

#####ACCESSORIES#####
REMOTE_SPOTLIGHT="false"
REMOTE_SPOTLIGHT_HOST="0.0.0.0"

#####VIDEO STREAM#####
RTSP_LOGIN="admin"
RTSP_PASSWORD="SecretPasswordHere"
RTSP_PORT="8554"

RTSP_HI_RES_ENABLED="true"
RTSP_HI_RES_ENABLE_AUDIO="false"
RTSP_HI_RES_FPS=""
RTSP_HI_RES_MAX_BITRATE="2048"
RTSP_HI_RES_TARGET_BITRATE="1024"
RTSP_HI_RES_ENC_PARAMETER="2"

RTSP_LOW_RES_ENABLED="false"
RTSP_LOW_RES_ENABLE_AUDIO="false"
RTSP_LOW_RES_FPS=""
RTSP_LOW_RES_MAX_BITRATE=""
RTSP_LOW_RES_TARGET_BITRATE=""
RTSP_LOW_RES_ENC_PARAMETER=""

RTSP_AUTH_DISABLE="false"

#####RTMP STREAMING#####
#### REQUIRES RTSP ####
### AND RTSP AUDIO ###
RTMP_STREAM_ENABLED="false"
RTMP_STREAM_FEED="video1_unicast"
RTMP_STREAM_SERVICE="youtube"
RTMP_STREAM_DISABLE_AUDIO="false"
RTMP_STREAM_YOUTUBE_KEY=""
RTMP_STREAM_TWITCH_KEY=""
RTMP_STREAM_FACEBOOK_KEY=""

#####GENERAL#####
LIBCALLBACK_ENABLE="true"
ENABLE_SWAP="true"
ENABLE_USB_STORAGE="false"
ENABLE_EXT4="false"
ENABLE_CIFS="false"
DISABLE_FW_UPGRADE="false"
AUDIO_PROMPT_VOLUME="50"
ENABLE_MP4_WRITE="false"
NIGHT_DROP_DISABLE="false"
DISABLE_MOTOR="false"
ENABLE_FSCK_ON_BOOT="false"
ENABLE_CAR_DRIVER="false"
ENABLE_LOCAL_DNS="false"
ENABLE_CRONTAB="false"
ENABLE_SELFHOSTED_MODE="true"

#####DEBUG#####
#drops you to a shell via serial, doesn't load app_init.sh
DEBUG_ENABLED="false"
DEBUG_INITRAMFS_ENABLED="false"
DEBUG_PASSWORD="false"
ENABLE_SYSLOG_SAVE="false"

####WEB####
WEB_SERVER_ENABLED="true"

#####SCRIPTING#####
CUSTOM_SCRIPT_PATH=""
```

lachesis, 2023-03-05:
LOW_RES rtsp seems to work, but HI_RES does not...

lachesis, 2023-03-07:
Another oddity: when the issue occurs, the following things quit working:

* Any open SSH connections hang
* Pinging the device fails
* RTSP stream freezes
* Opening new connections fails

However, if the device is pinging another client, it continues to do so indefinitely. The power usage of the device also remains relatively constant between 230 and 270mA throughout the process, regardless of whether it is exhibiting the issue.

gtxaspec, 2023-07-12:
have you tried swapping power adapters? unstable power does weird things on the units.

lachesis, 2023-07-18:
I tried using the original power adapter that came with the Wyze Cam, as well as a USB power bank and my computer. A second WyzeCam running the same software works perfectly with all 3 of these adapters.