gtxaspec/wz_mini_hacks #492: Noise - cmd aplay

baudev, 2023-03-07:
I notice that the `cmd aplay` command (with a .wav audio file present on the camera) makes a strange noise at the beginning and end of the audio file.  
This noise is also present when the camera says "WZ initializing" when installing wz_mini.

However, the camera does not make this strange noise if you use only the Wyze firmware. Does anyone know where it comes from?

Thanks

gtxaspec, 2023-07-11:
i've noticed this varies by device, some do, and some don't.  could it be power supply related?

gtxaspec, 2023-09-19:
@baudev Try a file, not in wav format, but raw pcm s16le 16000hz mono,  and no noise should happen.

gtxaspec, 2023-09-25:
confirmed, playing raw pcm files without the wav header eliminated the click

gtxaspec, 2023-09-26:
converted all sounds to raw pcm https://github.com/gtxaspec/wz_mini_hacks/commit/6cda362ee84350e8b3df62e64f22b485be80c16f

closed 2023-09-26 by gtxaspec

baudev, 2024-01-05:
Thanks, I tried today and indeed it works perfectly 🎉