gtxaspec/wz_mini_hacks #507: I can connect with RTSP on VLC but no video?

pjuhl2313, 2023-04-09:
Hello,

I set up the config so I can access the RTSP stream. I enter the stream w/ password/username into VLC. It picks up the stream and I have audio, but no video?

#####VIDEO STREAM#####
RTSP_LOGIN="admin"
RTSP_PASSWORD="admin"
RTSP_PORT="8555"

RTSP_HI_RES_ENABLED="true"
RTSP_HI_RES_ENABLE_AUDIO="true"
RTSP_HI_RES_FPS="30"
RTSP_HI_RES_MAX_BITRATE="2048"
RTSP_HI_RES_TARGET_BITRATE="2048"
RTSP_HI_RES_ENC_PARAMETER="2"

RTSP_LOW_RES_ENABLED="false"
RTSP_LOW_RES_ENABLE_AUDIO="false"
RTSP_LOW_RES_FPS="5"
RTSP_LOW_RES_MAX_BITRATE="512"
RTSP_LOW_RES_TARGET_BITRATE="256"
RTSP_LOW_RES_ENC_PARAMETER="2"

RTSP_AUTH_DISABLE="false"

#####RTMP STREAMING#####
#### REQUIRES RTSP ####
### AND RTSP AUDIO ###
RTMP_STREAM_ENABLED="false"
RTMP_STREAM_FEED="video1_unicast"
RTMP_STREAM_SERVICE="youtube"
RTMP_STREAM_DISABLE_AUDIO="false"
RTMP_STREAM_YOUTUBE_KEY=""
RTMP_STREAM_TWITCH_KEY=""
RTMP_STREAM_FACEBOOK_KEY=""

akivacp, 2023-04-10:
I'm not too skilled in this, I've experienced a similar issue, sometimes it takes a few moments for the video to pop up.

pjuhl2313, 2023-04-10:
So update, The issue was the firmware on the cameras. Cameras with 4.36.9.139 worked fine, audio and visual showed up. Cameras with newer updates did not work. I rolled all the cameras back and now they work fine. For those using Home Assistant I also recommend the rtsptoweb addon for incredibly fast uploading of live RTSP video