gtxaspec/wz_mini_hacks #512: Installs on V3 but won't connect via ethernet port

pjuhl2313, 2023-04-23:
I have several V3's I'm trying to switch to ethernet. I've gotten two of them to work. I reset the camera, add it to the Wyze app. I then install the mini hacks on a SD card, it boots up and I can RTSP over wifi fine. As soon as I change the setting of Ethernet to True and connect it to an ethernet to micro USB adapter and my POE network switch, the blue light just flashes slowly like its trying to connect to the network. Any help?

Bob-AOL, 2023-04-25:
If you haven't run across this elsewhere yet, it pretty much covers the whole UCTRONICS subject along with a guide for initializing  an adapter once it's determined to be a good one. 

[UCTRONICS - Realtek Guide rev_A.pdf](https://github.com/gtxaspec/wz_mini_hacks/files/11326237/UCTRONICS.-.Realtek.Guide.rev_A.pdf)

pjuhl2313, 2023-04-25:
I ended up stumbling on this and got them to work. Thanks for the follow up

Bob-AOL, 2023-04-25:
Good deal. I'm glad it's helping some folks.

You may want to close this issue now that you have it sorted out.

closed 2023-04-25 by pjuhl2313