gtxaspec/wz_mini_hacks #513: RTSP Stream Requirements

rlarranaga, 2023-04-24:
Hello,
Do i need an rtsp enabled firmware to enable rtsp server, or does wz_mini_hacks provides the rtsp functionality?

I have to Wyze V3 cameras on 4.36.9.139, and i installed wz_mini_hacks. I can access the web server, but cannot open the rtsp streams in vlc.

Thanks!

pjuhl2313, 2023-04-24:
Do you have the correct IP address format? 
It should go:
rtsp://username:password@192.168.xx.xx:port/unicast

rlarranaga, 2023-04-24:
Yeah, i copied it from the interface, currently i have auth disabled, so it
is like that but with no user:pass@

Thanks

Bob-AOL, 2023-04-25:
Did you enable rtsp?

Bob-AOL, 2023-05-05:
What same issue? 

And if you are referring to the Wyze Beta RTSP firmware being required, the answer is no, though I personally prefer it and use wz_mini primarily for the USB/POE feature myself.

You are missing the FPS entry above which should be "20", but may not be a necessary entry for the stream to function. I don't actually know.

rlarranaga, 2023-05-05:
> I am currently running into the same issue. Is the RTSP firmware required
> for the mini_hacks to do RTSP?
>
> I have flashed firmware version4.36.9.139 non-RTSP (
> https://download.wyzecam.com/firmware/v3/demo_wcv3_4.36.9.139.zip). The
> device is connected to the Wyze app.
>
> My RTSP config looks like the following:
>
> #####VIDEO STREAM#####
> RTSP_LOGIN="admin"
> RTSP_PASSWORD="redacted"
> RTSP_PORT="8554"
>
> RTSP_HI_RES_ENABLED="true"
> RTSP_HI_RES_ENABLE_AUDIO="true"
> RTSP_HI_RES_MAX_BITRATE="2048"
> RTSP_HI_RES_TARGET_BITRATE="1024"
> RTSP_HI_RES_ENC_PARAMETER="2"
> RTSP_HI_RES_FPS=""
>
> RTSP_LOW_RES_ENABLED="false"
> RTSP_LOW_RES_ENABLE_AUDIO="false"
> RTSP_LOW_RES_FPS=""
> RTSP_LOW_RES_MAX_BITRATE=""
> RTSP_LOW_RES_TARGET_BITRATE=""
> RTSP_LOW_RES_ENC_PARAMETER=""
>
> RTSP_AUTH_DISABLE="false"

Let me know if you find a solution for this. I will do the same.

Thanks!

Bob-AOL, 2023-05-05:
Are you sure wz_mini is even loading?

rlarranaga, 2023-05-05:
It is definetely serving the web interface

dudududodododedede, 2023-05-05:
RTSP firmware is not a requirement to have RTSP working on wz_mini. I have RTSP working on v3 firmware version 4.36.9.139 (non-RTSP)

The only things I customized in the config file was:
RTSP_HI_RES_ENABLED
RTSP_HI_RES_ENABLE_AUDIO
RTSP_PASSWORD
(left everything else as defaults)

I also added the SSH key of course to authorized_keys

You should check to make sure you're using the right URL. Easiest way to test is by opening the stream in VLC

Bob-AOL, 2023-05-05:
SSH is not a requirement and can initially be problematic if set up incorrectly.

The MRL must be correct and the login credentials must be the same as for the camera itself if memory serves me correctly.

VLC viewer can be very picky and downright non-functional at times. I would not rely on it for troubleshooting until knowing the login credentials actually work first.

rlarranaga, 2023-05-05:
What would be the best way to test rtsp without VLC?

Thanks

Bob-AOL, 2023-05-05:
It _should_ work in VLC. I use Blue Iris. Another way to sort it out is with tinyCam if you have an Android device handy. It has an entry for Wyze/RTSP and will basically tell you what the problem may be if the stream won't start. A bit of a learning curve, but a handy tool and viewer nonetheless. Works with a Firestick as well.

When using the config editor in the web interface it may not correctly save/apply the file changes during the reboot. Double check for the edits afterward to be sure they were actually implemented.

dudududodododedede, 2023-05-05:
> What would be the best way to test rtsp without VLC? Thanks

You could try telnetting to the camera on the rtsp port

```
~ user$ telnet 192.168.0.225 8554
Trying 192.168.0.225...
Connected to 192.168.0.225.
Escape character is '^]'.
```

That last line means a connection is successful on the RTSP port 8554

Or if you SSH into the camera you can check `top` or `netstat` to ensure the rtsp process is running and that it's listening on the right port

```
[root@WCV3:~]# netstat -tulnap| grep rtsp
tcp        0      0 0.0.0.0:8554            0.0.0.0:*               LISTEN      695/v4l2rtspserver
tcp        0      0 192.168.0.225:8554      192.168.0.243:55471     ESTABLISHED 695/v4l2rtspserver
tcp        0      0 :::8554                 :::*                    LISTEN      695/v4l2rtspserver
udp        0      0 0.0.0.0:6970            0.0.0.0:*                           695/v4l2rtspserver
udp        0      0 0.0.0.0:6971            0.0.0.0:*                           695/v4l2rtspserver
```
most important is line 1 there

Bob-AOL, 2023-05-05:
Then there's Analiti -- or a simple port scanner. I would also consider starting over with a new SD card if all else fails.

[image]