gtxaspec/wz_mini_hacks #515: NTP Not working?

TheCableGuy99, 2023-04-26:
Hi,

I installed this on 4 Wyze cams yesterday and all was working well. I've looked at them today and all the dates and times are wrong.

They are displaying 2022-05-13 and the time is way out also. I have set the ntp server and also tested it works so I'm unsure why it's not working?

Also can the date be set in UK format? dd/mm/yyyy?

Thanks.

sideup66, 2023-05-01:
NTP on these devices is in UTC. You might be able to change it but I wouldn't given the stock wyze software can also sync the clock. UTC is easier as it doesn't deal with locale or timezones. You can simply convert for your needs.

TheCableGuy99, 2023-05-01:
How do you suggest changing it? It's not good having the wring time set :/

sideup66, 2023-05-01:
I mean, in the wyze app you can go into settings and have it sync time from there. Otherwise no way to keep it.

sideup66, 2023-05-01:
From command line you can trigger a sync from running /system/bin/timesync, see what comes of that...

TheCableGuy99, 2023-05-01:
So the script was kind enough to output the NTP servers:
[root@Back-Garden-Cam:~]# /system/bin/timesync
[timesync_utility.c:ip_validate::69] ip address obtained

[main.c:time_sync_process::276] time sync process running!
[main.c:time_sync_ntp::65] ntp_worker_handler time_zone:0
[main.c:time_sync_ntp::106] ntp ip: 66.220.9.122
[main.c:time_sync_ntp::134] select timeout
[main.c:time_sync_ntp::106] ntp ip: 132.236.56.250
[main.c:time_sync_ntp::134] select timeout
[main.c:time_sync_ntp::106] ntp ip: 129.6.15.28
[main.c:time_sync_ntp::134] select timeout
[main.c:time_sync_ntp::106] ntp ip: 51.145.123.29
[main.c:time_sync_ntp::134] select timeout
[main.c:time_sync_ntp::106] ntp ip: 129.6.15.29
[main.c:time_sync_ntp::134] select timeout

So I added them to the white list to allow out. My question now is, does this script run peridocically? Or does it need to be set in a cron?

Thanks.

sideup66, 2023-05-01:
If you look at the processes running (ps) it does already run as a daemon.
You could also just allow traffic on the NTP ports it uses while blocking
everything else probably. That may be your best bet

TheCableGuy99, 2023-05-01:
[root@Back-Garden-Cam:~]# ps aux |grep timesync
  370 root      0:00 dmon --stderr-redir --max-respawns -1 --environ LD_PRELOAD=libsetunbuf.so /opt/wz_mini/tmp/.bin/ntpd -ndp 192.168.5.254 -- dslog --priority DEBUG --facility USER timesync
  375 root      0:00 dslog --priority DEBUG --facility USER timesync
11933 root      0:00 grep timesync

So this is where it gets confusing... It is running as a daemon but as you can see above it's set to use the routers NTP server (this is a verified working NTP server). I set the IP (192.168.5.254) in the web admin panel but it doesn't work and hence this thread.

When running the script manually as you suggested it connects to different servers and works fine.

Thanks.

sideup66, 2023-05-01:
That's interesting. Almost like you have two different NTP clients running....what happens if you kill the second timesync thing? That could be cronjobbed at start too, and maybe solely on NTP? What does NTP tell you when it tries

TheCableGuy99, 2023-05-01:
I've popped out now so will have to check if anything else is running when I get back a in a bit.

I can tell you that I've not installed anything additional.

I ran the command you gave above and it works connecting to 3rd party external servers. 

When I grep 'timesync' it shows the output in my precious post and nothing else, showing the routers IP.

Unless there's an NTP server running under another name I'm fairly sure there's only the default daemon running and not working, and the manual version working when not specifying any servers which is working.

I could try a workaround by specifying one of the servers in the web interface that it tries to connect to when running it manually.

If that doesn't work it's something with the command the daemon is running. As said, I can't test this until I'm back though.

How often is the daemon set to run? If I set the time to something incorrect and reboot the camera will it force the daemon to run on boot so i can check if the new IP works?

Thanks again for your support.

TheCableGuy99, 2023-05-01:
So I've had a little play and when I look in the web interface there's a blue bar at the bottom of the page. This time and data is completely wrong and different to the time on the picture of the video recording.

It's currently set to: Thu Oct 13 12:37:18 AM PDT 2022

I can change the time/data on the cameras video display either manually using the time or data command, or by setting the NTP server to one of those the script uses and allowing that through the firewall so there is a workaround but something is definitely not right due to me not being able to use a local NTP server and the time/date in the web interface being well out.

It's also worth noting that I have this setup on 4 Wyze Cam v3's and it's the same on all of them.

Cheers.

gtxaspec, 2023-07-11:
you can use `rdate` to set the time in a script manually or in cron as well.