gtxaspec/wz_mini_hacks #522: Attempting v2_install bad checksum

Triplehtown, 2023-05-09:
Attempting to install v2 in ubuntu and I come up to this error:

2023-05-09 18:42:05 (15.1 MB/s) - ‘./v2_ro/demo.zip’ saved [9133357/9133357]

md5sum failed check, please manually supply file

Please advice on how to proceed.

Grntrenchman, 2023-05-25:
This was caused by Wyze pulling the firmwares needed by this script from their download page (but still available directly lol) under some downgrade trouble pretense, possibly on the sly just to break this (and keep people in the walled garden of services). 

Thanks Wyze. I read the script, saw it's looking for a specific firmware version, and got the link in there.

You can change, or add a second "DL_URL" line after the first, and directly supply the proper link:
`DL_URL="https://download.wyzecam.com/firmware/v2/demo_4.9.8.1002.zip"`

Alternatively, you "manually supply file" as it says, by downloading the firmware above, stripping the version (leaving it `demo.zip`), and place it in the dir with the script before you run it, according to what the script says. Then you don't need to edit anything.

All is well after this.

irsl, 2023-05-31:
Have you got a URL for Pan v1 4.10.8.1002?
https://download.wyzecam.com/firmware/pan/demo_4.10.8.1002.bin.zip returns 403.
I prefer not experimenting with the latest firmware, but rather use the same one you have been testing with.

irsl, 2023-05-31:
I added an echo after this line of `compile_image.sh`:

```
PAN_DL_URL=$(wget --header="Accept: text/html" --user-agent="$UA_STRING" -qO- https://support.wyze.com/hc/en-us/articles/360024852172-Release-Notes-Firmware | grep -Eo "(http|https)://[a-zA-Z0-9./?=_-]*" | grep /pan/ |   sort -t . -k2r,5 | grep demo | head -1)
```
At the time of writing these lines, it evaluates to `https://download.wyzecam.com/firmware/pan/demo_4.10.9.1433.bin.zip`. I replaced the version number in there.

Thanks!

dasb00ter, 2023-08-17:
Hmm I am attempting to prepare an install for v1 pan cam or v2 cam and I am also getting a checksum problem. I have downloaded the firmware from the link directly and placed it as demo.bin in the directory but it appears to just download whatever is available and ignores whatever I placed there? Can somebody help?