gtxaspec/wz_mini_hacks #536: Is Wyzecam V3 Pro supported?

SomebodySysop, 2023-06-13:
Does wz_mini_hacks support the new V3 Pro cameras?  I've tried to search for this but can't find anything.

DAMAGEDatheCORE, 2023-06-13:
It's literally the first thing written at the top of the readme:

"v2/car/PANv1/v3/PANv2 devices ONLY"

SomebodySysop, 2023-06-13:
> No. It's literally the first thing written at the top of the readme:
> 
> "v2/car/PANv1/v3/PANv2 devices ONLY"
> 
> Also, your question isn't an issue and should instead have been posted under the Discussion section.

Oops, my bad.

But, semantically speaking, since V3 Pro is technically a V3, one could reasonably assume that it was supported from this first thing written.  Anyway, doesn't hurt to ask.

gtxaspec, 2023-06-29:
not until we get the GPL sources for the v3 pro released by wyze. 

https://support.wyze.com/hc/en-us/articles/360012546832-Open-Source-Software

they are missing sources for a bunch of new devices.