gtxaspec/wz_mini_hacks #551: feature request: webserver: don't modify anything in /configs.

gtxaspec, 2023-07-12:
@virmaior 

I believe that allowing users to edit the configurations could introduce significant risks. If the configuration were to become corrupted or lost, it seems that the only recourse would be to restore from an unmodified backup of the configurations. In the worst-case scenario where a user doesn't have these backups, the camera could be permanently disconnected from the cloud due to the loss of the encryption keys.  Perhaps we should disable this feature and have it be "read-only"...

What are your thoughts on this potential issue?

virmaior, 2023-07-12:
very strongly agree.  I actually did lose the config file on one of my V3s irrecoverably (not sure if it was from this code or something else I was trying).

I think basically read-only should be the default. I'm not even sure if it makes sense to allow any edits at all. Too much of a risk on that front.

I'll do this and the RTSP link for V2s soon... Juggling lots of other things right now!

virmaior, 2023-07-12:
resolved in https://github.com/gtxaspec/wz_mini_hacks/pull/558

gtxaspec, 2023-07-12:
updated in master.

closed 2023-07-12 by gtxaspec