gtxaspec/wz_mini_hacks #559: Test latest PanV1/V2 firmware, should work.

gtxaspec, 2023-07-12:
fix compile_image script as well.

Grntrenchman, 2023-08-07:
Tried installing on OOB refurb pan v2. updated to latest firmware manually (4.49.3.4054), installed wz_mini_hacks set up in standalone. 
Scanned qr code from FANG config generator, "timeout" played twice, but it instantly reconnected to wifi (and stayed). ssh works, rtsp is enabled but fails to run properly. 

process is running, but errors in log:
```
log level:500

[NOTICE] /root/CLEAN/v4l2rtspserver/main.cpp:297
        Version: 0.3.3-4-ga43e094-dirty live555 version:2022.04.26
[NOTICE] /root/CLEAN/v4l2rtspserver/src/V4l2RTSPServer.cpp:37
        Create V4L2 Source.../dev/video1
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:133
        driver:v4l2 loopback capabilities:85008003 mandatory:4000001
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:135
        /dev/video1 support output
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:136
        /dev/video1 support capture
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:138
        /dev/video1 support read/write
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:139
        /dev/video1 support streaming
[ERROR] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:192
        /dev/video1: Cannot get format Invalid argument
[ERROR] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:192
        /dev/video1: Cannot get format Invalid argument
[ERROR] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:192
        /dev/video1: Cannot get format Invalid argument
[ERROR] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:83
        Cannot init device:/dev/video1
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:141
        Device /dev/video1VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor

```

Also, manually running "cmd move" does nothing, and does not exit. Ctrl+c closes fine.

Here is wz_mini.conf

```
#wz_mini configuration

#### W E B CAMERA###
##THIS MODE DISABLES EVERYTHING AND IT WILL
## WORK AS A WEB CAMERA FOR YOUR PC ***ONLY***
WEB_CAM_ENABLE="false"
WEB_CAM_BIT_RATE="8000"
WEB_CAM_FPS_RATE="25"

#####NETWORKING#####
CUSTOM_HOSTNAME="OpenMiko3"

ENABLE_USB_ETH="false"
ENABLE_USB_ETH_MODULE_AUTODETECT="true"
ENABLE_USB_ETH_MODULE_MANUAL=""
USB_ETH_MAC_ADDR=""

ENABLE_USB_DIRECT="false"
USB_DIRECT_MAC_ADDR="02:01:02:03:04:08"

ENABLE_USB_RNDIS="false"

ENABLE_IPV6="false"

ENABLE_WIREGUARD="false"
WIREGUARD_IPV4=""
WIREGUARD_PEER_ENDPOINT=""
WIREGUARD_PEER_PUBLIC_KEY=""
WIREGUARD_PEER_ALLOWED_IPS=""
WIREGUARD_PEER_KEEP_ALIVE=""

ENABLE_IPTABLES="false"

ENABLE_NFSv4="false"

ENABLE_RTL8189FS_DRIVER="true"
ENABLE_ATBM603X_DRIVER="true"

NTP_SERVER=""

##### NETWORK INTERFACE BONDING #####
BONDING_ENABLED="false"
BONDING_PRIMARY_INTERFACE="eth0"
BONDING_SECONDARY_INTERFACE="wlan0"
BONDING_LINK_MONITORING_FREQ_MS="100"
BONDING_DOWN_DELAY_MS="5000"
BONDING_UP_DELAY_MS="5000"
BONDING_FAIL_OVER_MAC="0"

#####ACCESSORIES#####
REMOTE_SPOTLIGHT="false"

#####VIDEO STREAM#####
RTSP_LOGIN="openmiko"
RTSP_PASSWORD="openmiko"
RTSP_PORT="8554"

RTSP_HI_RES_ENABLED="true"
RTSP_HI_RES_ENABLE_AUDIO="false"

RTSP_LOW_RES_ENABLED="false"
RTSP_LOW_RES_ENABLE_AUDIO="false"

RTSP_AUTH_DISABLE="false"

VIDEO_0_MAX_BITRATE=""
VIDEO_0_TARGET_BITRATE=""
VIDEO_0_ENC_PARAMETER="2"
VIDEO_0_FPS="20"

VIDEO_1_MAX_BITRATE=""
VIDEO_1_TARGET_BITRATE=""
VIDEO_1_ENC_PARAMETER=""
VIDEO_1_FPS=""

GO2RTC_SERVER_ENABLED="false"
GO2RTC_HI_RES_ENABLED="false"
GO2RTC_HI_RES_ENABLE_AUDIO="false"
GO2RTC_LOW_RES_ENABLED="false"
GO2RTC_LOW_RES_ENABLE_AUDIO="false"

#####RTMP STREAMING#####
#### REQUIRES RTSP ####
### AND RTSP AUDIO ###
RTMP_STREAM_ENABLED="false"
RTMP_STREAM_FEED="video1_unicast"
RTMP_STREAM_SERVICE="youtube"
RTMP_STREAM_DISABLE_AUDIO="false"
RTMP_STREAM_YOUTUBE_KEY=""
RTMP_STREAM_TWITCH_KEY=""
RTMP_STREAM_FACEBOOK_KEY=""

#####GENERAL#####
LIBCALLBACK_ENABLE="true"
ENABLE_SWAP="true"
ENABLE_USB_STORAGE="false"
ENABLE_EXT4="false"
ENABLE_CIFS="false"
DISABLE_FW_UPGRADE="true"
AUDIO_PROMPT_VOLUME="50"
ENABLE_MP4_WRITE="true"
NIGHT_DROP_DISABLE="false"
DISABLE_MOTOR="false"
ENABLE_FSCK_ON_BOOT="true"
ENABLE_CAR_DRIVER="false"
ENABLE_LOCAL_DNS="false"
ENABLE_CRONTAB="false"
ENABLE_SELFHOSTED_MODE="true"

#####DEBUG#####
#drops you to a shell via serial, doesn't load app_init.sh
DEBUG_ENABLED="false"
DEBUG_INITRAMFS_ENABLED="false"
DEBUG_PASSWORD="false"
ENABLE_SYSLOG_SAVE="false"

####WEB####
WEB_SERVER_ENABLED="false"
WEB_SERVER_AUTH="true"
WEB_SERVER_LOGIN="admin"
WEB_SERVER_PASSWORD=""

#####SCRIPTING#####
CUSTOM_SCRIPT_PATH=""
```

Ignore "openmiko" artifacts, using naming matching other cameras already installed for uniformity. Trying with sound enabled produces same results.
Blocking internet access to camera causes the expected flapping, as there's no patch to iCamera for this model, let alone firmware version.

I'm in no rush to get this camera installed, so let me know if there's anything I can send/run/install to help.

gtxaspec, 2023-08-08:
See https://github.com/gtxaspec/wz_mini_hacks/wiki/Firmware-Support

Grntrenchman, 2023-08-08:
Ah. I see you've now edited the page and issue to clarify.