gtxaspec/wz_mini_hacks #564: v2 cam unable to reboot/flash/setup

87435, 2023-07-15:
Hi,  
I need help flashing/restoring my v2 cam that looks bricked after a reboot.

### Context
I installed the hacks yesterday (July 12 version, with firmware `4.9.8.1002`) and managed to make it work. I had webserver enabled and RTSP but then I edited the config to try `VIDEO_0_MAX_BITRATE` and `VIDEO_0_TARGET_BITRATE` to some values, see if I could increase the quality of the streams. I believe I used "4192" for both, then rebooted.

### The issue
After that reboot, the cam had a solid orange light and I could not connect to it, either through the IP or the wyze app. My router does not show it as connected to the wifi. 

### What I tried
Tried to flash with the hacks, flashing to the same original firmware, flashing down to `4.9.2.42` which lead me to a blinking orange light but still no way to connect/setup, flashing to `4.9.9.1851` without success. I am not even sure what firmware is installed at this point since the cam doesn't finish booting either way.   

FYI I use the SD Card Formatter app which works great and has allowed me to have the hack working on my other v2 cam.

I would appreciate any ideas to try and get the cam back to a working state.

gtxaspec, 2023-07-15:
would you describe what method you used to install wz_mini?

87435, 2023-07-15:
> would you describe what method you used to install wz_mini?  

to install wz_mini: I manually put the `4.9.8.1002` firmware in the `v2_install` folder, then ran the script, copied to SD card and flashed. It worked fine including RTSP etc, until I tried to set a bitrate and rebooted.   

> have you tried to reset the camera and flash the factory firmware, and see if you can press the setup button to begin the factory setup process again?  
I would love to be able to do that! Unable to get back to the factory setup process. I have held the setup button for a while in different states and it just doesn't seem to work.

gtxaspec, 2023-07-15:
when you try to flash the firmware, do you get the solid blue/yellow led on the back of the v2?

if you pull the sd card there are some log files there:

`wz_mini_initramfs.log`
`wz_init.log`

could you share those if they are present?

87435, 2023-07-15:
I am not even able to get a blue/yellow light now when trying to flash. It goes to solid yellow right away with a couple of clicking noises, and that's it. Holding the setup button doesn't do anything, whether it's when plugging it or after giving it time to boot.  
No wz_mini logs on the card - only whatever demo.bin file I put there to try and restore to factory. I believe one of the firmware flashes worked at some point, I don't believe wz_mini is running anymore

gtxaspec, 2023-07-15:
Very strange, I've not had a V2 act this way.  

The way wz_mini works on the v2, it flashes a new kernel to the device, but leaves everything else intact, including the bootloader.  

So you should still be able to re-flash the stock firmware no matter what. 

If you have one, try a different SD card, format as FAT again, and keep trying to re-flash.

87435, 2023-07-15:
Tried a different SD, tried on Windows and make a 512MB partition (since I only have 32/64GB SD cards) ... nothing works.  
The only sign of life I can see is that if I put the `4.9.2.42` firmware as demo.bin and hold the setup button on boot, it will start blinking yellow after 10s despite not showing any blue at any moment during the process. So it seems _something_ is happening inside...
Any other ideas? I could try other firmwares but I have only found `4.9.2.42`, `4.9.8.1002` and `4.9.9.1851` from trustworthy sources.

87435, 2023-07-16:
is there a way to install the hacks on an uninitialized cam? it seems that I am able to boot it with the mini_hacks but since there is no wifi configuration it cannot fully initalize. The Setup button does not trigger the normal config assistant ("Ready to connect" etc). I guess if there is a simple `wpa_supplicant.conf` I could write to somewhere via script?

bjs-pdx, 2023-07-17:
You have to setup WIFI in the app first.  Then you can move on with wz_mini.

bjs-pdx, 2023-07-17:
**"...but then I edited the config..."**

Did you use the Windows editor?  Did you tell it to use linux/unix format for CR/LF?  IF not the file is probably FUBAR.

87435, 2023-07-17:
> Did you use the Windows editor? Did you tell it to use linux/unix format for CR/LF? IF not the file is probably FUBAR.

I edited the config directly from the device using `vi` through ssh

gtxaspec, 2023-07-19:
```
# cat /tmp/wpa_supplicant.conf 
ctrl_interface=/var/run/wpa_supplicant
update_config=1
network={
	ssid="ssid_here"
	key_mgmt=WPA-PSK
	pairwise=CCMP TKIP
	group=CCMP TKIP WEP104 WEP40
	psk="passphrase_here"
	scan_ssid=1
```

87435, 2023-07-19:
Here is how I tried to implement it:

Let's say the SD Card is mounted on a freshly formatted `/sd_card/`
- copied the mini_hacks files (demo.bin, factory_t31....., wz_mini/) on `/sd_card/` 
- created `/sd_card/wz_mini/set_wpa_supplicant.sh` containing the following:
```bash
#!/opt/wz_mini/bin/bash

echo "set_wpa_supplicant | Sleeping 10s"
sleep 10
echo "set_wpa_supplicant | Starting"
wpa_supplicant -D nl80211 -i wlan0 -c /opt/wz_mini/tmp/wpa_supplicant.conf -B -s
echo "set_wpa_supplicant | Finished"
```  
- `chmod a+x /sd_card/wz_mini/set_wpa_supplicant.sh` to make it executable
- created `/sd_card/wz_mini/tmp/wpa_supplicant.conf` containing the following (replacing $SSID and $PASSPHRASE with the actual values):
```bash
ctrl_interface=/var/run/wpa_supplicant
update_config=1
network={
	ssid="$SSID"
	key_mgmt=WPA-PSK
	pairwise=CCMP TKIP
	group=CCMP TKIP WEP104 WEP40
	psk="$PASSPHRASE"
	scan_ssid=1
}
```
- edited `/sd_card/wz_mini/wz_mini.conf` to add custom script: `CUSTOM_SCRIPT_PATH="/opt/wz_mini/set_wpa_supplicant.sh"` 

This resulted in the camera booting, wz_mini audio prompts (first boot, initializing swap) but nothing else. 

[Here are the logs](https://pastebin.com/3sSfSPES). No mention of the custom script - could you please confirm the `echo` from my script should appear in the logs? Is there anything else you can see in the logs that could help troubleshoot?

Thanks!

PS: the closing curly bracket is missing from your wpa_supplicant.conf code above - please add it in case someone else goes through these steps in the future

gtxaspec, 2023-07-20:
try
```
#!/opt/wz_mini/bin/bash

. /opt/wz_mini/etc/rc.common
. /opt/wz_mini/wz_mini.conf

wait_for_icamera

sleep 15

echo "set_wpa_supplicant | Sleeping 10s"
sleep 10
echo "set_wpa_supplicant | Starting"
wpa_supplicant -D nl80211 -i wlan0 -c /opt/wz_mini/tmp/wpa_supplicant.conf -B -s
echo "set_wpa_supplicant | Finished"
```

87435, 2023-07-20:
> name it test.sh, chmod +x it, and stick it in /opt/wz_mini/etc/rc.local/, don't use the custom script path

do you mean in `wz_mini/etc/rc.local.d/`? `wz_mini/etc/rc.local` exists but it's not a directory.

87435, 2023-07-21:
My other cam who was working fine has gone unresponsive as well now. I am afraid it's the same issue.  
Both of them became unresponsive right after I rebooted them through ssh by entering `reboot` in the terminal. Could it be the cause of the issue somehow?  

This cam was rebooted multiple times in the past (whether by unplugging it or by using the `reboot` command) without a problem. It has been working well for a week, serving RTSP video and through the Wyze app.  

If you have a moment to help me troubleshoot, I am going to leave it as is and not try to format/flash it for now in case you need to check some specific files. If you don't have time just let me know and I'll go ahead and flash and try stuff to revive it.  
Thank you!

87435, 2023-07-22:
Made some progress - I am able to run scripts, pass commands to `wpa_supplicant` (using `/system/bin/wpa_supplicant`) and `wpa_cli`, see `wlan0` and use `iwconfig` HOWEVER the wifi scans come up empty so far. Unable to see or join a network.

87435, 2023-07-26:
Do you have any other ideas by any chance @gtxaspec? I feel like trying to manually connect to wifi is a dead end. Even if I managed to successfully connect AND ssh there, I still need to figure out what to change to make it work again.  

FYI I was able to use the `demo_uboot.bin.gz` you provided here https://github.com/gtxaspec/wz_mini_hacks/issues/241#issuecomment-1214471826 and got the purple light, a log that seemed to show it work successfully. Is there any way to use this to reformat to the default wyze bootloader/config/files/whatever this updates? (after the sucessful uboot, step 7. didn't work when I tried to flash stock firmware)

87435, 2023-07-27:
Another possible route to recovery: i was able to get the wifi connection working with ssh access using openmiko! Still unsure how to reset everything to stock though.

stephenmetzger, 2023-08-01:
I am having the same problem as you. I had everything running fine, but was tweaking the bitrate settings when my camera fell offline. I have not been able to get it working by stock firmware (pulling SD card and powering/pressing the setup button) or trying to load the hacked firmware again. 

I also can get to the wz_hack voice prompt, but then the device hangs and never connects to my network. 

With the SD card ejected, plugging in triggers a yellow light for about 5-10 seconds, the device shutter clicks, and then the light turns off an it seems dead.

87435, 2023-08-01:
I am able to get both Dafang hacks and Openmiko to work. But no stock firmware which means the cams don't show in the app, and both these hacks look pretty unstable. Still trying to find a good config for Openmiko but that might be the best of the two.  
Until someone more knowledgeable is willing to help I feel like I'm stuck.

stephenmetzger, 2023-08-02:
> 

Thanks. If I’m able to get RTSP streams working, I’m okay without the Wyze app. Right now I’m not able to get anything running.

gtxaspec, 2023-08-02:
@87435 on any of the memory cards, do you have a folder in the root of the card, named .configs or config?  It sounds like for some reason the camera configs were corrupted somehow

87435, 2023-08-02:
@gtxaspec no there is no `config` or .`config` folder at the root of the card

87435, 2023-08-02:
@gtxaspec ok there is `.wz_backup/configs`

virmaior, 2023-08-06:
> Made some progress - I am able to run scripts, pass commands to `wpa_supplicant` (using `/system/bin/wpa_supplicant`) and `wpa_cli`, see `wlan0` and use `iwconfig` HOWEVER the wifi scans come up empty so far. Unable to see or join a network.

I think you mean `/bin/wpa_supplicant` ?

virmaior, 2023-08-06:
that's odd. on my firmware version there didn't seem to be a wpa_supplicant in that directory.

I was able to debrick my camera by copying `/configs` from a different camera. Without the content in `/configs`  I couldn't flash the default firmware successfully or get the camera to run because it was dying do iCamera not running if wlan0 is not found.

87435, 2023-08-07:
> that's odd. on my firmware version there didn't seem to be a wpa_supplicant in that directory.
>
> I was able to debrick my camera by copying `/configs` from a different camera. Without the content in [...]

YES!!! My camera is back thanks to you!  
In my case it was the `params` folder (mounted from `/dev/mtdblock9`) thas was empty. I copied the values from the SD card (`.wz_backup/params/*`) back there and the camera was back online after a reboot.  

A clue was actually right at the beginning of the logs that [I posted weeks ago](https://pastebin.com/3sSfSPES): 
`cat: can't open '/opt/wz_mini/tmp/params/config/.product_config': No such file or directory` which probably prevented other steps from executing correctly.

stephenmetzger, 2023-08-07:
> YES!!! My camera is back thanks to you! In my case it was the `params` folder (mounted from `/dev/mtdblock9`) thas was empty. I copied the values from the SD card (`.wz_backup/params/*`) back there [...]

Can you provide some additional info??

I’m currently running the dafang hack, but would like to get wz_mini working. 

Did you have to grab the configs folder from a different cam? One that was running wz_mini? 

Seems like you and I were experiencing the same issue, I’d like to trace the steps you took to see if I can get mine working.

virmaior, 2023-08-08:
> Did you have to grab the configs folder from a different cam? One that was running wz_mini?

On my version, I just copied it from a working a camera. I don't use WyzeAPP at all so I don't care if that means it can't connect. 

Specifically, one of the config files gives each camera a unique id in wyze's system.
Also on more recent versions the wifi info is not stored as plain text (I'm using an old firmware where it still is).

87435, 2023-08-08:
Of course - I was planning on posting a summary here. Although keep in mind for now I have only tried on my 2nd cam, which was left almost untouched since it got bricked i.e it still had wz_mini, I never tried to flash it etc. More importantly, that means the `.wz_backup` folder was still on the SD card.
For anyone in that situation, you should be able to fix it by simply creating a script, naming it something like `S04restore_configs_and_params` and placing it on your SD card in `/wz_mini/etc/init.d/`:
```bash
#!/bin/sh

mount -t jffs2 /dev/mtdblock8 /configs
cp -R /opt/.wz_backup/configs/* /configs/
umount /configs

mount -t jffs2 /dev/mtdblock9 /params
cp -R /opt/.wz_backup/params/* /params/
umount /params
```

Plug in, give it a few minutes to boot up to that point, then unplug, go remove that script from the SD Card, put the sd back in the camera and plug it - you should be all set.  

I will try to get my other cam back up - that one is a similar situation as yours since it has been flashed etc

Real-Time-Kodi, 2023-08-12:
> Of course - I was planning on posting a summary here. Although keep in mind for now I have only tried on my 2nd cam, which was left almost untouched since it got bricked i.e it still had wz_mini, I [...]

It looks like my params backup is empty  
This script doesn't seem to be bringing the camera back to life  

I think what caused the bricking was turning on the self hosted mode on the wrong firmware.  
I might try to reflash the chip today if I can find a flash dump somewhere. I also have one with a bad bootloader to flash

e46lux, 2023-10-02:
> ```
> # cat /tmp/wpa_supplicant.conf 
> ctrl_interface=/var/run/wpa_supplicant
> update_config=1
> network={
> 	ssid="ssid_here"
> 	key_mgmt=WPA-PSK
> 	pairwise=CCMP TKIP
> 	group=CCMP TKIP WEP104 WEP40
> 	psk="passphrase_here"
> 	scan_ssid=1
> }
> ```
> 
> run it with: ` wpa_supplicant -D nl80211 -i wlan0 -c /tmp/wpa_supplicant.conf -B -s`

Hi, not sure if I should create a separate issue but I was having the same problem - possibly strung along from an SD corruption (I wasn't able to update settings via web server and they would boot back to old settings). I tried to 'refresh' the SD properly via linux and ended up being unable to boot back into wz_mini. In troubleshooting process, I attempted loading newer / older firmwares to no avail - my V2 would just not connect to wifi and would come up to flashing yellow light. I ended up adding the scripts proposed above and still wasn't able to connect - "Network not found". At some point, I was able to use a prior backup folder but didn't realize that it actually fixed things because of the following:

My wlan0 MAC address changed to what is the mac address of the V2 I got the backup from. So while the original 'backup restored from' cam is online, my 'restoring V2' was unable to join the wifi network LOL. I ended up adding 

ifconfig wlan0 down
ifconfig wlan0 hw ether XX:XX:XX:XX:XX:XX
ifconfig wlan0 192.168.1.25 netmask 255.255.255.0
ifconfig wlan0 up

to opt/wz_mini/etc/rc.common after while loops in wait_for_wlan and wait_for_wlan_ip

Camera now boots up and joins the network, gets an ip under the 'backup restored from' mac and then resets itself to it's own MAC with my script additions and gets on the network with proper MAC. I ended up having to specify the ifconfig IP because it just would not pull a new dhcp lease even with 'non-permanent' MAC address changed and wlan0 interface being brought back up after being down. 'non-permanent' because 'ifconfig wlan0 hw ether XX:XX:XX:XX:XX:XX' does not persist a boot. I cannot get this camera to join the network via Wyze setup, likely due to soft-overwritten MAC somewhere. This works fine for me now - just wanted to report. If you have a suggestion on how I can get the MAC address back to it's original - that would be great.
I was able to find that the MAC is overwritten in /sys/class/net/wlan0/address but I'm unable to edit this file even after chmod 777 - something to do with linux permissions on /sys/ directory and since my script above was working I left it alone and didn't try setting permissions on each folder to be able to edit the 'address' file.

alienjon, 2025-02-13:
Jumping on this thread [looks at watch] a year and a half late.  I had this issue come up (problem with flashing a v2 and light would only turn solid yellow even after restarting (After a brief period off yellow on, then off, then on solid until unplugged).  I suspected either a problem with the flashing itself or a network issue (I realized after the fact that I had reset the camera but hadn't connected it to the network first).  I was getting an error about wpa_supplicant not being found an wlan0 failing.  I was able to fix the issue by flashing back to stock firmware.  I've since reconnected to the network properly and going to try reflashing wz_mini_hacks.  In the meantime I wanted to share this experience for anyone this came up for.

To flash stock firmware you first need to [download the firmware for your camera](https://support.wyze.com/hc/en-us/articles/360024852172-Release-Notes-Firmware) (again, I'm v2 - I selected the most recent, which may well be different by the time you read this).  Unplug the camera and take the sdcard out (if you haven't already).  Empty the card of all files (I backed mine up just in case, but I don't think you want any other files in there, though I honestly didn't test that aspect so maybe it'll work).  Unzip the downloaded firmware and rename the .bin file to **demo.bin**.  Copy this file to the SD card and insert back into the camera.  While holding the 'setup' button, plug it into a power source.  Continue holding the 'setup' button until the light turns blue and then release.  Then let it sit for a while (maybe 3-4 minutes).  Now you can press and release 'setup' and it should be working again (at least it did for me and I was able to connect to the Wyze app and connect to my network at this point).

bjgrenke, 2025-02-25:
I've ran into this same issue every time I've attempted to put wz_mini on my V2. After using the script to successfully generate a demo.bin for the V2, I flash it and am left with a boot looping camera. Intermittent solid yellow light and a periodic clicking sound.

Luckily I can flash the latest FW and resume using on the Wyze app, but would love to get this one running RTSP.