gtxaspec/wz_mini_hacks #565: ICamera cpu usage increased

vic4news, 2023-07-15:
Hi,
I installed the latest version of wz_mini_hacks on some of my v3 cams and noticed that the cpu usage of the ICamera process almost doubled. Is this expected? 

Btw, been using this code for over a year and it really makes the cameras usable

gtxaspec, 2023-07-16:
for me: 
old release 60-80 cpu icamera usage
new release: same

what is your expected cpu usage?

vic4news, 2023-07-16:
The cameras with the 09-27-22 version normally have cpu usage of 34% - 42%.
The cameras running the new version have cpu usage of 57% - 62%
The cameras record continuously to the sd card. Enable mp4 write is on.

clee604, 2023-07-16:
Probably dumb question hut what are you using to monitor the usage?

I have a bunch of cameras I'm about to update so I can also do some comparisons.

vic4news, 2023-07-16:
I’m just using the ‘top’ command

dudududodododedede, 2023-07-16:
What time of day are you checking? For me the cpu usage is much higher during the day, and lower at night. I assume because there is less activity/visible detail at night

stncttr908, 2023-07-20:
My CPU usage is the same after updating on all of my V3s

gtxaspec, 2023-09-25:
closing for now, expected cpu usage for iCamera is  50-80 (it's a huge, cobbled together app that has tons of code)
reopen issue if further issues persist.

closed 2023-09-25 by gtxaspec