gtxaspec/wz_mini_hacks #578: Unable to SSH into V3 cam, RTSP stream is blank

sth144, 2023-07-24:
I am able to view my camera in the app and everything appears fine, but I'm unable to SSH into my V3 cam with wz_mini_hacks. Every time I try, I get `Permission denied (publickey)`. I have added my SSH .pub keys (both RSA and ED25519) to `/wz_mini/etc/ssh/authorized_keys` on the SD card, so as far as I know, I should be able to SSH in, but I'm not. Do I need to use a specific username for SSH?

Also, when I try to play the RTSP streams, I just get a blank black screen.

Wyze firmware version: 4.36.11.4679

Here is my wz_mini.conf configuration:
```
#wz_mini configuration

#### W E B CAMERA###
##THIS MODE DISABLES EVERYTHING AND IT WILL
## WORK AS A WEB CAMERA FOR YOUR PC ***ONLY***
WEB_CAM_ENABLE="false"
WEB_CAM_BIT_RATE="8000"
WEB_CAM_FPS_RATE="25"

#####NETWORKING#####
CUSTOM_HOSTNAME="WCV3"

ENABLE_USB_ETH="false"
ENABLE_USB_ETH_MODULE_AUTODETECT="true"
ENABLE_USB_ETH_MODULE_MANUAL=""
USB_ETH_MAC_ADDR=""

ENABLE_USB_DIRECT="false"
USB_DIRECT_MAC_ADDR="02:01:02:03:04:08"

ENABLE_USB_RNDIS="false"

ENABLE_IPV6="false"

ENABLE_WIREGUARD="false"
WIREGUARD_IPV4=""
WIREGUARD_PEER_ENDPOINT=""
WIREGUARD_PEER_PUBLIC_KEY=""
WIREGUARD_PEER_ALLOWED_IPS=""
WIREGUARD_PEER_KEEP_ALIVE=""

ENABLE_IPTABLES="false"

ENABLE_NFSv4="false"

ENABLE_RTL8189FS_DRIVER="true"
ENABLE_ATBM603X_DRIVER="true"

NTP_SERVER=""

##### NETWORK INTERFACE BONDING #####
BONDING_ENABLED="false"
BONDING_PRIMARY_INTERFACE="eth0"
BONDING_SECONDARY_INTERFACE="wlan0"
BONDING_LINK_MONITORING_FREQ_MS="100"
BONDING_DOWN_DELAY_MS="5000"
BONDING_UP_DELAY_MS="5000"
BONDING_FAIL_OVER_MAC="0"

#####ACCESSORIES#####
REMOTE_SPOTLIGHT="false"

#####VIDEO STREAM#####
RTSP_LOGIN=""
RTSP_PASSWORD=""
RTSP_PORT="8554"

RTSP_HI_RES_ENABLED="true"
RTSP_HI_RES_ENABLE_AUDIO="true"

RTSP_LOW_RES_ENABLED="true"
RTSP_LOW_RES_ENABLE_AUDIO="true"

RTSP_AUTH_DISABLE="true"

VIDEO_0_MAX_BITRATE=""
VIDEO_0_TARGET_BITRATE=""
VIDEO_0_ENC_PARAMETER=""
VIDEO_0_FPS=""

VIDEO_1_MAX_BITRATE=""
VIDEO_1_TARGET_BITRATE=""
VIDEO_1_ENC_PARAMETER=""
VIDEO_1_FPS=""

GO2RTC_SERVER_ENABLED="false"
GO2RTC_HI_RES_ENABLED="false"
GO2RTC_HI_RES_ENABLE_AUDIO="false"
GO2RTC_LOW_RES_ENABLED="false"
GO2RTC_LOW_RES_ENABLE_AUDIO="false"

#####RTMP STREAMING#####
#### REQUIRES RTSP ####
### AND RTSP AUDIO ###
RTMP_STREAM_ENABLED="true"
RTMP_STREAM_FEED="video1_unicast"
RTMP_STREAM_SERVICE="youtube"
RTMP_STREAM_DISABLE_AUDIO="false"
RTMP_STREAM_YOUTUBE_KEY=""
RTMP_STREAM_TWITCH_KEY=""
RTMP_STREAM_FACEBOOK_KEY=""

#####GENERAL#####
LIBCALLBACK_ENABLE="true"
ENABLE_SWAP="true"
ENABLE_USB_STORAGE="false"
ENABLE_EXT4="false"
ENABLE_CIFS="false"
DISABLE_FW_UPGRADE="false"
AUDIO_PROMPT_VOLUME="50"
ENABLE_MP4_WRITE="false"
NIGHT_DROP_DISABLE="false"
DISABLE_MOTOR="false"
ENABLE_FSCK_ON_BOOT="false"
ENABLE_CAR_DRIVER="false"
ENABLE_LOCAL_DNS="false"
ENABLE_CRONTAB="false"
ENABLE_SELFHOSTED_MODE="false"

#####DEBUG#####
#drops you to a shell via serial, doesn't load app_init.sh
DEBUG_ENABLED="false"
DEBUG_INITRAMFS_ENABLED="false"
DEBUG_PASSWORD="false"
ENABLE_SYSLOG_SAVE="false"

####WEB####
WEB_SERVER_ENABLED="true"
WEB_SERVER_AUTH="true"
WEB_SERVER_LOGIN="admin"
WEB_SERVER_PASSWORD=""

#####SCRIPTING#####
CUSTOM_SCRIPT_PATH=""
```

vic4news, 2023-07-24:
I think your wyze firmware version is too high. You should flash the firmware to  4.36.1.139 and then install the mini_hacks. 

https://github.com/gtxaspec/wz_mini_hacks/wiki/Firmware-Support

clee604, 2023-07-24:
the ssh username should be 'root' - at least that's how mine are set up

sth144, 2023-07-24:
> I think your wyze firmware version is too high. You should flash the firmware to 4.36.9.139 and then install the mini_hacks. I think ssh should have worked but definitely not RTSP.
> 
> https://github.com/gtxaspec/wz_mini_hacks/wiki/Firmware-Support This page says that versions up to 4.36.1.139 are supported but I think that may be a typo. Most people run 4.36.9.139
> 
> Also see #567

thank you. Downgrading firmware seems to have fixed it

gtxaspec, 2023-07-24:
Converting to discussion for visibility

closed 2023-07-24 by gtxaspec