gtxaspec/wz_mini_hacks #587: Unable to connect to WiFi Network - Unable to roll back to stock firmware

stephenmetzger, 2023-08-02:
Hi all, 

I had wz_mini_hacks running okay a few days ago, then my camera stopped responding after making some config changes in the web ui. I have since not been able to flash the firmware back to a Wyze stock firmware (attempted a few different versions). However, I have been able to run dafang and openmiko - I believe these have worked because they have a step to manually configure a wpa_supplicant.conf file within /etc. 

My wz-mini logs show: 

#####S21syslog#####
: IP Address not acquired yet, try again in 5 seconds
S14storemac: wpa_supplicant not initialized yet, try again in 5 seconds
S12remoteaccessory: IP Address not acquired yet, try again in 5 seconds
S09coredump: wpa_supplicant not initialized yet, try again in 5 seconds
iCamera is Running
rm: can't remove '/configs/upgrade.log': No such file or directory
: IP Address not acquired yet, try again in 5 seconds
S14storemac: wpa_supplicant not initialized yet, try again in 5 seconds
S12remoteaccessory: IP Address not acquired yet, try again in 5 seconds
S09coredump: wpa_supplicant not initialized yet, try again in 5 seconds
```

which indicates to me that the device is not able to connect to a WiFi network. However, I am not able to find a wpa_supplicant file anywhere. 

Any way to manually configure my wifi credentials?

Thanks, 

Steve

DAMAGEDatheCORE, 2023-08-03:
Mini Hacks doesn't flash/modify the stock firmware. You simply remove the
MicroSD card and restart or power cycle the cam and you're back to stock
state.

bjs-pdx, 2023-08-04:
Did you follow the Wyze instructions for flashing factory firmware?  Which file did you download?  What happens when you try to manually flash older firmware versions from the Wyze downloads?

87435, 2023-08-04:
> Mini Hacks doesn't flash/modify the stock firmware. You simply remove the MicroSD card and restart or power cycle the cam and you're back to stock state.

Then it's hard to explain how I got the same issue than @stephenmetzger even though I went straight from stock firmware/bootloader to wz_mini and my cams have gone unresponsive the same way. My cameras haven't been able to boot to stock despite repeated attempts to remove the SD card, flash different versions etc. I never get to a point where I can get it to the"Ready to connect" prompt so something is modified that corrupts the stock firmware boot in some scenarios.

virmaior, 2023-08-06:
> 
> Yes, unfortunately something else must have occurred while I was testing different alt firmwares. Removing the SD card and restarting/powercycing/attempting a button-hold reset doesn't get me back to stock. Refreshing with a stock firmware .bin also doesn't allow me to enter the setup mode and get my V2 back into the Waze app.

Can you check the camera to see if the config info is clobbered?

I did so by modify S04mount and adding 
`ls -alh /configs`

should have about 10 files including camera config files and wifi config

good output looks like:
```
-rwxr-xr-x    1 root     root        4.0K Aug  6 15:21 ._user_config
-rwxr-xr-x    1 root     root        1.9K Aug  6 15:21 .grid_config
-rwxr-xr-x    1 root     root         441 Aug  6 15:21 .product_config
-rwxr-xr-x    1 root     root          11 Aug  6 15:21 .time_checkpoint
-rwxr-xr-x    1 root     root          18 Aug  6 15:21 .timezone_config
-rwxr-xr-x    1 root     root         197 Aug  6 15:21 .user_config
-rwxr-xr-x    1 root     root          11 Aug  6 15:21 .wifipasswd
-rwxr-xr-x    1 root     root           6 Aug  6 15:21 .wifissid
-rwxr-xr-x    1 root     root          10 Aug  6 15:21 TZ
-rwxr-xr-x    1 root     root          22 Aug  6 15:21 app.ver
-rwxr-xr-x    1 root     root          22 Aug  6 15:21 os-release
-rwxr-xr-x    1 root     root           0 Aug  6 15:21 slapped_config
-rwxr-xr-x    1 root     root         192 Aug  6 15:21 user_config
```

bad output is empty.

I think the key killer is no .wifipasswd and .wifissid

87435, 2023-08-10:
@stephenmetzger any updates? I have been able to restore the cam that was still intouched since becoming unresponsive, however the one I flashed several times since I have not been able to bring back to life yet. Restoring /configs and /params does allow me to get the "Ready to connect" prompt, however the wifi connection immediatly times out after scanning the QR code.

tavicu, 2023-11-25:
I have the same problem with one of my cameras but I don't think it's related to wz_mini.
I think the wifi fried on a power surge or something.