gtxaspec/wz_mini_hacks #588: Have the firmware requirements changed?  Anyone try the latest firmware 4.36.11+ with V3 cams?

gleep52, 2023-08-02:
Just curious if a V3 camera will still work with 4.36.11 firmware instead of the much older and security prone .9 range of firmware?  Is the .9 build of the firmware required for the mini hack to work? 

I'd rather not keep my cameras on a vulnerable firmware.

sideup66, 2023-08-02:
The newer firmware will work for a v3 with the hack however rtsp and cmd functions will not. 

May I ask what security issue you are referring to? This hack already loads in a custom kernel and not the stock one shipped with firmware.

Grntrenchman, 2023-08-09:
I'd like to hear this out too. What is "security prone", I think you wanted "vulnerability prone"? AFAIK there was 2 things so far, authentication logic error, and serving SD on www. Both are local only, so unless you're assigning public IPs... 

These CVE dropped (and were already patched when it did) in 3/2022. Only the Pan V2 firmware recommended is from that time (3/2022), and the CVE explicitly mentions this version as fixed. All others are newer, recommended 4.36.9.139 released 5/2022. It's specifically the ".9 range" that fixes this vuln.

I'd love to hear of these new vulnerabilities for the V3, past what the hack itself opens up... otherwise this is hyperbole.

If we're doing a "they had a vuln, and didn't fix it for a while, so they're lax and have others" being on newest firmware really won't help with that.

gtxaspec, 2023-08-09:
There's no new vulnerabilities.  

Any potential issues would be server side anyway, and I personally don't consider a www server running locally to be a CVE anyway.   

It makes for great headlines and enables non-technical people to worry over nothing.

You are at greater risk from undisclosed vulnerabilities in your average consumer home wifi router... lol.

biggt, 2023-09-14:
I'm on the 4.36.10.x firmware with 2x V3 cameras and struggling with getting them running. They get stuck at the red flashing light and that's it. It doesn't boot, doesn't say anything, no purple light, nothing.

mikecirioli, 2023-09-20:
> I am on firmware vesrion `4.36.11.5859` and WyzeCam v3 and am unable to get RTSP via go2rtc to work as well. I am able to access the camera at `http://<ip>:1984` but get the following error in the logs when trying to view the stream from there:
> 
> ```
> 10:37:03.009 DBG [exec] run url="exec:ffmpeg -hide_banner -f alsa -ac 1 -i dsnooper0 -c:a libopus -compression_level 7 -frame_duration 10 -muxdelay 0 -muxpreload 0 -ac:a 1 -af adelay=0|0 -f rtsp {output}"
> [alsa @ 0x770f2030] cannot set channel count to 1 (Invalid argument)
> ```
> 
> Any tips on how to debug this? I feel like i need to be adding some additional parameters to the stream url (ie. more than just http://192.168.1.32:1984/stream.html?src=1080p)

ok, it seems that RTSP is not supported on the firmware i have installed.  I will attempt to downgrade and see if that resolves the issue.

gtxaspec, 2023-09-20:
see https://github.com/gtxaspec/wz_mini_hacks/wiki/Firmware-Support
converting to discussion for visibility

closed 2023-09-20 by gtxaspec