gtxaspec/wz_mini_hacks #593: Option to set pan camera angle at boot

Grntrenchman, 2023-08-08:
This is a feature request for a new script, and options to be added to wz_mini.conf.

For my panv2 (not sure if applies to panv1), it goes back to center position every boot, which makes keeping it at a certain angle annoying. I want a feature where you can define set angles for the camera, and it will return to them after a reset/reboot, effectively  keeping it there.

I'd make a pull request (?) but I'm bad at git/understanding the flow. Hopefully you can work with what I post, but either way, it Works On My System (TM).

possibly a section in config as such

```
#### PAN ADJUSTMENT ####
PAN_DEFAULT_X=""
PAN_DEFAULT_Y=""
```

and a startup script (/opt/wz_mini/etc/rc.local.d/S80rotate in my case)

```
#!/bin/sh
### BEGIN INIT INFO
# Provides:
# Short-Description: Rotate Motor
# Description:       Rotate Motor on specified devices
### END INIT INFO

. /opt/wz_mini/etc/rc.common
. /opt/wz_mini/wz_mini.conf

rotate_default() {

	wait_for_icamera

	echo "#####$(basename "$0")#####"

	if [[ "$DISABLE_MOTOR" == "false" ]]; then
		if [ -n "$PAN_DEFAULT_X" ]; then
			if [ -n "$PAN_DEFAULT_Y" ]; then
				echo "Moving to user-defined default position"
				cmd move $PAN_DEFAULT_X $PAN_DEFAULT_Y
			else
				echo "Default rotation not defined - Y"
			fi
		else
			echo "Default rotation not defined - X"
		fi
	fi
}	
	
case "$1" in
	start)
		rotate_default &
		;;
	*)
		echo "Usage: $0 {start}"
		exit 1
		;;
esac

```

This setup is working for me, moving the camera to specified angles on boot.

gtxaspec, 2023-09-19:
I'll add this to the next release, thank you!

gtxaspec, 2023-09-20:
available in latest master.

closed 2023-09-25 by gtxaspec